gnuradio/volk #421: Static build fails for v2.4.0

mkacher, 2020-12-01:
The static build of v2.4.0 fails. I confirmed that a static build of v2.3.0 works OK for me.

```
$ git clone https://github.com/gnuradio/volk.git
$ cd volk/
$ git submodule update --init --recursive
$ git checkout v2.4.0
$ mkdir build && cd build
$ cmake -DENABLE_STATIC_LIBS=True ..
$ make

...
Scanning dependencies of target volk-config-info
[ 85%] Building CXX object apps/CMakeFiles/volk-config-info.dir/volk-config-info.cc.o
[ 86%] Building CXX object apps/CMakeFiles/volk-config-info.dir/volk_option_helpers.cc.o
[ 88%] Linking CXX executable volk-config-info
../lib/libvolk.a(volk_cpu.c.o): In function `i_can_has_avx512cd':
volk_cpu.c:(.text+0x28): undefined reference to `GetX86Info'
../lib/libvolk.a(volk_cpu.c.o): In function `i_can_has_avx512f':
volk_cpu.c:(.text+0x78): undefined reference to `GetX86Info'
../lib/libvolk.a(volk_cpu.c.o): In function `i_can_has_avx2':
volk_cpu.c:(.text+0xc8): undefined reference to `GetX86Info'
../lib/libvolk.a(volk_cpu.c.o): In function `i_can_has_avx':
volk_cpu.c:(.text+0x118): undefined reference to `GetX86Info'
../lib/libvolk.a(volk_cpu.c.o): In function `i_can_has_sse4_2':
volk_cpu.c:(.text+0x168): undefined reference to `GetX86Info'
../lib/libvolk.a(volk_cpu.c.o):volk_cpu.c:(.text+0x1a8): more undefined references to `GetX86Info' follow
collect2: error: ld returned 1 exit status
apps/CMakeFiles/volk-config-info.dir/build.make:121: recipe for target 'apps/volk-config-info' failed
make[2]: *** [apps/volk-config-info] Error 1
CMakeFiles/Makefile2:464: recipe for target 'apps/CMakeFiles/volk-config-info.dir/all' failed
make[1]: *** [apps/CMakeFiles/volk-config-info.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
```

drmpeg, 2020-12-01:
If you add ```cpu_features``` to this line, it will build.

https://github.com/gnuradio/volk/blob/master/lib/CMakeLists.txt#L572

However, there is a secondary issue. If liborc-0.4 is installed, the static build will fail with a different error.

I'd submit a PR, but it will conflict with https://github.com/gnuradio/volk/pull/420, so I'm going to wait until that PR is merged.

jdemel, 2020-12-01:
@drmpeg it would be great to have your PR. Would you mind to add a CI test with a static lib to our CI system? This would be great and probably prevent future issues like this one. I didn't check static builds before.

closed 2020-12-02 by jdemel