gnuradio/volk #423: auto format cmake files

jdemel, 2020-12-01:
I suggest to use [cmake_format](https://github.com/cheshirekow/cmake_format) to auto-format all our CMake files. cmake is often hard to read and having it at least formatted would ease this task. Of course, we'd need to decide on any particular style guide. etc.

drmpeg, 2020-12-24:
I'm not sure how well supported this tool is. I tried to install it and was not successful. There's more than one complaint about this in the cmake_format Github issues, and no response.

https://github.com/cheshirekow/cmake_format/issues/238

https://github.com/cheshirekow/cmake_format/issues/234

jdemel, 2020-12-25:
I'm aware that this is not perfect. Unfortunately, it is the only tool I found that does CMake formatting. Of course, I'd be happy to use a better tool. Do you happen to know one?

jdemel, 2023-01-15:
This is essentially the same Issue as #600 . Since GR used it, we might follow.

PR #615 (2024-02-20): cmake: Apply cmake-format
This commit adds `.cmake-format.py`. It is mostly inspired by the GNU Radio version of it.

Steps to reproduce these results:
- `pip3 install cmakelang`
- `cmake-format --dump-config python > .cmake-format.py`
- Modify .cmake-format.py to be close to the GNU Radio version.
- `cmake-format -c .cmake-format.py -i $(git ls-files | grep CMakeLists.txt) $(git ls-files | grep .cmake)`

Fixes #600
Fixes #423
closed 2024-02-20 by jdemel