gnuradio/volk #426: Tarballs do not contain the cpu_features subdirectory

mbr0wn, 2020-12-18:


jdemel, 2020-12-19:
Oh. Where did this happen? I updated the 2.4.0 and 2.4.1 tarballs to contain this subfolder. The error message should be helpful for people that try to use VOLK via git clone and forgot to do so recursively.

marcusmueller, 2020-12-19:
They do contain the subdirectory, but the detection fails when you build with `cmake -S <path to source> -B <path to build>` 

Which, sadly, our buildbot containers do ---

marcusmueller, 2020-12-19:
no, that's not even the issue: it doesn't work when the build directory isn't a subdirectory of the source directory for some reason?

marcusmueller, 2020-12-19:
what. not even that was the problem. (don't worry, I'll condense this flow of conscience down later)

marcusmueller, 2020-12-19:
@mbr0wn when using the github automatic archive URL, you get an unsigned tarball that github rolled automatically. Using the VOLK releases page, everything seems peachy.

jdemel, 2020-12-21:
So, my manually created tarball:  https://github.com/gnuradio/volk/releases/download/v2.4.1/volk-2.4.1.tar.gz  works correctly? But the GH auto generated one fails?

jdemel, 2021-01-02:
@mbr0wn I assume we can't fix GH auto-generated files. I close this issue now. If my assumption is incorrect feel free to re-open this issue.

closed 2021-01-02 by jdemel