gnuradio/volk #428: Check/add support for Apple M1 processor

dkozel, 2020-12-21:
With Apple's switch to ARM there may be some adjustments needed to detect and use the available NEON instructions.

@sbehnke is interested in working on/testing this feature.

jdemel, 2020-12-21:
I'd hope we can just use an updated cpu_features version if this seems necessary. Besides publicly available docs regarding that chip would be helpful as well.

sbehnke, 2020-12-21:
If I change the cpu_features to recognize that the M1 chip is actually ARM with CPU_FEATURES_ARCH_ARM defined, volk compiles and runs without issue. I'm not sure if this the appropriate solution or if there is more to do. While clang 12 has __aarch64__ define, it does not appear to define __arm__ so the cpu_feature submodule fails to build.

jdemel, 2020-12-21:
So VOLK builds and uses NEON on M1? That's good.

And we might face 2 issues in the future:
1. `cpu_features` seems to fail at automatically detecting M1 as an ARM CPU with NEON. In that case I suggest to open an issue against `cpu_features` and we can update our submodule pointer as soon as this is fixed.
2. clang12 has some issue. Are you refering to a `__aarch64__` define as opposed to a `__arm__` define?

sbehnke, 2020-12-21:
With regards to 2 first, clang12 has `__aarch64__` defined but does not have `__arm__` defined so the cpu_features emits an error about trying to compile cpuinfo_arm.c on a non-arm cpu.

In the email from @dkozel I'm not sure if it is using neon or not yet. I have attached the make test results here as I only downloaded / built it for the first time anywhere yesterday. 

```
Running tests...
Test project /Users/sbehnke/Documents/GitHub/volk/build
        Start   1: qa_volk_16i_32fc_dot_prod_32fc
  1/134 Test   #1: qa_volk_16i_32fc_dot_prod_32fc .......................   Passed    0.19 sec
        Start   2: qa_volk_16i_branch_4_state_8
  2/134 Test   #2: qa_volk_16i_branch_4_state_8 .........................   Passed    0.01 sec
        Start   3: qa_volk_16i_convert_8i
  3/134 Test   #3: qa_volk_16i_convert_8i ...............................   Passed    0.01 sec
        Start   4: qa_volk_16i_max_star_16i
  4/134 Test   #4: qa_volk_16i_max_star_16i .............................   Passed    0.01 sec
        Start   5: qa_volk_16i_max_star_horizontal_16i
  5/134 Test   #5: qa_volk_16i_max_star_horizontal_16i ..................   Passed    0.01 sec
        Start   6: qa_volk_16i_permute_and_scalar_add
  6/134 Test   #6: qa_volk_16i_permute_and_scalar_add ...................   Passed    0.01 sec
        Start   7: qa_volk_16i_s32f_convert_32f
  7/134 Test   #7: qa_volk_16i_s32f_convert_32f .........................   Passed    0.01 sec
        Start   8: qa_volk_16i_x4_quad_max_star_16i
  8/134 Test   #8: qa_volk_16i_x4_quad_max_star_16i .....................   Passed    0.01 sec
        Start   9: qa_volk_16i_x5_add_quad_16i_x4
  9/134 Test   #9: qa_volk_16i_x5_add_quad_16i_x4 .......................   Passed    0.01 sec
        Start  10: qa_volk_16ic_convert_32fc
 10/134 Test  #10: qa_volk_16ic_convert_32fc ............................   Passed    0.01 sec
        Start  11: qa_volk_16ic_deinterleave_16i_x2
 11/134 Test  #11: qa_volk_16ic_deinterleave_16i_x2 .....................   Passed    0.01 sec
        Start  12: qa_volk_16ic_deinterleave_real_16i
 12/134 Test  #12: qa_volk_16ic_deinterleave_real_16i ...................   Passed    0.01 sec
        Start  13: qa_volk_16ic_deinterleave_real_8i
 13/134 Test  #13: qa_volk_16ic_deinterleave_real_8i ....................   Passed    0.01 sec
        Start  14: qa_volk_16ic_magnitude_16i
 14/134 Test  #14: qa_volk_16ic_magnitude_16i ...........................   Passed    0.01 sec
        Start  15: qa_volk_16ic_s32f_deinterleave_32f_x2
 15/134 Test  #15: qa_volk_16ic_s32f_deinterleave_32f_x2 ................   Passed    0.01 sec
        Start  16: qa_volk_16ic_s32f_deinterleave_real_32f
 16/134 Test  #16: qa_volk_16ic_s32f_deinterleave_real_32f ..............   Passed    0.01 sec
        Start  17: qa_volk_16ic_s32f_magnitude_32f
 17/134 Test  #17: qa_volk_16ic_s32f_magnitude_32f ......................   Passed    0.01 sec
        Start  18: qa_volk_16ic_x2_dot_prod_16ic
 18/134 Test  #18: qa_volk_16ic_x2_dot_prod_16ic ........................   Passed    0.01 sec
        Start  19: qa_volk_16ic_x2_multiply_16ic
 19/134 Test  #19: qa_volk_16ic_x2_multiply_16ic ........................   Passed    0.01 sec
        Start  20: qa_volk_16u_byteswap
 20/134 Test  #20: qa_volk_16u_byteswap .................................   Passed    0.00 sec
        Start  21: qa_volk_16u_byteswappuppet_16u
 21/134 Test  #21: qa_volk_16u_byteswappuppet_16u .......................   Passed    0.01 sec
        Start  22: qa_volk_32f_64f_add_64f
 22/134 Test  #22: qa_volk_32f_64f_add_64f ..............................   Passed    0.01 sec
        Start  23: qa_volk_32f_64f_multiply_64f
 23/134 Test  #23: qa_volk_32f_64f_multiply_64f .........................   Passed    0.01 sec
        Start  24: qa_volk_32f_8u_polarbutterfly_32f
 24/134 Test  #24: qa_volk_32f_8u_polarbutterfly_32f ....................   Passed    0.01 sec
        Start  25: qa_volk_32f_8u_polarbutterflypuppet_32f
 25/134 Test  #25: qa_volk_32f_8u_polarbutterflypuppet_32f ..............   Passed    0.02 sec
        Start  26: qa_volk_32f_accumulator_s32f
 26/134 Test  #26: qa_volk_32f_accumulator_s32f .........................   Passed    0.01 sec
        Start  27: qa_volk_32f_acos_32f
 27/134 Test  #27: qa_volk_32f_acos_32f .................................   Passed    0.01 sec
        Start  28: qa_volk_32f_asin_32f
 28/134 Test  #28: qa_volk_32f_asin_32f .................................   Passed    0.01 sec
        Start  29: qa_volk_32f_atan_32f
 29/134 Test  #29: qa_volk_32f_atan_32f .................................   Passed    0.01 sec
        Start  30: qa_volk_32f_binary_slicer_32i
 30/134 Test  #30: qa_volk_32f_binary_slicer_32i ........................   Passed    0.01 sec
        Start  31: qa_volk_32f_binary_slicer_8i
 31/134 Test  #31: qa_volk_32f_binary_slicer_8i .........................   Passed    0.01 sec
        Start  32: qa_volk_32f_convert_64f
[... 183 lines cut ...]
        Start 124: qa_volk_8ic_deinterleave_real_16i
124/134 Test #124: qa_volk_8ic_deinterleave_real_16i ....................   Passed    0.01 sec
        Start 125: qa_volk_8ic_deinterleave_real_8i
125/134 Test #125: qa_volk_8ic_deinterleave_real_8i .....................   Passed    0.01 sec
        Start 126: qa_volk_8ic_s32f_deinterleave_32f_x2
126/134 Test #126: qa_volk_8ic_s32f_deinterleave_32f_x2 .................   Passed    0.01 sec
        Start 127: qa_volk_8ic_s32f_deinterleave_real_32f
[... cut ...]
131/134 Test #131: qa_volk_8u_x2_encodeframepolar_8u ....................   Passed    0.00 sec
        Start 132: qa_volk_8u_x3_encodepolar_8u_x2
132/134 Test #132: qa_volk_8u_x3_encodepolar_8u_x2 ......................   Passed    0.00 sec
        Start 133: qa_volk_8u_x3_encodepolarpuppet_8u
133/134 Test #133: qa_volk_8u_x3_encodepolarpuppet_8u ...................   Passed    0.01 sec
        Start 134: qa_volk_8u_x4_conv_k7_r2_8u
134/134 Test #134: qa_volk_8u_x4_conv_k7_r2_8u ..........................   Passed    0.00 sec

100% tests passed, 0 tests failed out of 134
```

jdemel, 2020-12-21:
You have multiple options to check: `./build/apps/volk-config-info --avail-machines` and the `--machine` and `--all-machines` flags. The last one should tell you which machines are actually compiled in. The other are more host specific. If you run `ctest -V`, you should see a lot of kernels with output like `neon ...`. In that case, the NEON kernel was executed.
I saw tweets that confirmed M1 supports SVE as well. But I couldn't find any documentation that supports this. Do you have any more knowledge about that?

About the `__aarch64__` vs `__arm__` thing. I just read up on it and as far as I understand it, whenever there's an ARM target, `__arm__` should be present but `__aarch64__` only if a 64bit ARM is available. I assume this is smth that should be fixed in `cpu_features`.

sbehnke, 2020-12-21:
I don't think it does support SVE. Looking at the `sysctl -a` output:

```
hw.optional.floatingpoint: 1
hw.optional.watchpoint: 4
hw.optional.breakpoint: 6
hw.optional.neon: 1
hw.optional.neon_hpfp: 1
hw.optional.neon_fp16: 1
hw.optional.armv8_1_atomics: 1
hw.optional.armv8_crc32: 1
hw.optional.armv8_2_fhm: 1
hw.optional.armv8_2_sha512: 1
hw.optional.armv8_2_sha3: 1
hw.optional.amx_version: 2
hw.optional.ucnormal_mem: 1
hw.optional.arm64: 1
```

I don't think that even thought cpu_features is compiling it is detecting anything properly.

```
❯ list_cpu_features
arch            : ARM
implementer     :   0 (0x00)
architecture    :   0 (0x00)
variant         :   0 (0x00)
part            :   0 (0x00)
revision        :   0 (0x00)
flags           :

❯ volk-config-info --avail-machines
generic;
❯ volk-config-info --all-machines
generic;neon;neonv8
```

Looks like there's an open issue about adding support for M1 to cpu_feature already, so that much is good.

https://github.com/google/cpu_features/issues/121

sbehnke, 2020-12-22:
Getting closer to having something functional:

```
❯ ./list_cpu_features
arch            : aarch64
implementer     : 16777228 (0x100000C)
variant         :   2 (0x02)
part            : 458787763 (0x1B588BB3)
revision        :   2 (0x02)
flags           : asimdfhm,atomics,crc32,fp,fphp,sha3,sha512
```

jdemel, 2020-12-22:
@sbehnke thanks for working on it!

sbehnke, 2020-12-23:
Submitted https://github.com/google/cpu_features/pull/150 Now we wait for Google to tell me what I've done wrong.

michaelld, 2020-12-24:
Very cool! I have an ARM Mac I can do testing on ... not M1 but still ARM. I left a comment on you PR ... LGMT either way, but I think that change would help. Thx for your work!

sabrinastronomy, 2021-04-24:
Hi there! Just found this thread after my intel based MacBook died this week, and I chose (which has now become a painful process in reinstalling GNU radio) to get the M1 chip based MacBook. I currently can't successfully install VOLK in the Rosetta emulator or with the ARM cmake toolchain files. Are there any updates on this? I can also help with testing if needed.

jdemel, 2021-04-24:
We're waiting for the PR against cpu_features to get merged. Afterwards, we can probably add M1 support pretty quickly. It looks like M1 is ARM based but there are some subtle differences.
For now, I assume it would be really helpful to support this PR against cpu_features. Maybe test it, etc.

sabrinastronomy, 2021-04-27:
This (among other near impossible installations) prompted me to actually return the computer for an Intel chip. All will be much easier now! Thank you for your help and great development.

dkozel, 2021-06-14:
@sbehnke Did you successfully get GNU Radio running on an M1 mac?

sbehnke, 2021-06-14:
> @sbehnke Did you successfully get GNU Radio running on an M1 mac?

No, I haven't. I got pulled in a different direction and haven't had a chance to circle back yet.

dkozel, 2021-06-14:
Thanks for the fast response! I think no one has succeeded then.

sbehnke, 2021-06-14:
> Thanks for the fast response! I think no one has succeeded then.

I made a pull request for the upstream Volk, but it didn't get approved and pulled in, which I don't blame them for. I don't think we quite capture all of the capabilities of the processor and without that building in the tree I'm not sure what to do.

https://github.com/google/cpu_features/pull/150

jdemel, 2021-06-15:
Thanks for doing this PR! I still hope it'll get merged at some point. At this point 2 out of 3 PRs against cpu_features are directly relevant to us.

fllay, 2021-08-21:
Any news ? It seems that macports can compile gnuradio and volk  for m1  https://ports.macports.org/port/volk/details/  https://ports.macports.org/port/volk/builds/?page=1. I am using homebrew  but still no news for homebrew.

michaelld, 2021-08-21:
@fllay Yes I pushed [fixes for Volk's cpu_features into MacPorts](https://github.com/macports/macports-ports/commit/e83a55ef196d4283be438c052295b2fc44f3df5b). No movement upstream there; no idea what the timeframe is. I don't mix and match package managers, and I do MacPorts -- so I have no idea what homebrew is doing.

michaelld, 2021-08-21:
Having Volk patch cpu_features is a possible, I suppose ... that would help homebrew and other package managers and building from source on M1. Worth considering.

fllay, 2021-08-22:
Thank you for information, I guess I will move to macports. Actually, when I used intel Mac, I was using macports for package management and gnuradio.

jdemel, 2021-08-22:
On the plus side: It is a very simple patch to cpu_features to compile and use VOLK on M1.
On the minus side: This would report incomplete data for M1 with cpu_features.
I assume they don't want to report incomplete features and thus, they (cpu_features maintainers) didn't merge the available PRs yet.

We could patch cpu_features ourselves. However, that'll give quite a few package maintainers quite a headache. We'd rely on a custom cpu_features version that is not available through package managers. I'd say that is not a solution we should do here but leave to the package maintainers to decide for themselves. I just hope that there will be progress soon on the cpu_features side.

michaelld, 2021-09-25:
@jdemel discussion from GRCon21 summary: (1) don't use `cpu_features` since it is no longer in active development and clearly the maintainers are not engaged with supporting the M1 even though it's been out for "a long time"; (2) hence, revert much of the work that added support for `cpu_features` (e.g., 5e2193c69eb47a383e351079fa4c078381b9e6d6); (3) fixup the new code to work with M1 & other CPU / ARCH. This will require some work, but it would at least give us control over the CPU / ARCH determination, which is a serious problem right now.

michaelld, 2021-09-25:
The related discussion was to minimally patch `cpu_features` to return `UNKNOWN` (or whatever the correct term is) for the M1, and then handle that inside Volk CMake. This would be -more- consistent with current behavio[u]r, but still might be a pain for quite a few package managers (as noted a few comments above).

jdemel, 2021-09-26:
I fear that `cpu_features` is abandonware.
If we revert to the situation we had before the introduction of `cpu_features`, we add quite a few more regressions. Basically everything Windows related falls back to `generic`. Also, we would again fail to properly detect AVX/AVX2 features.

I'd really like to find a maintained project that supports CPU feature detection. We might extract the relevant parts from `cpu_features` for our purposes. Then, we'd extend them to support M1. 

In any case, we might run into issues e.g. in VMs. If there'd be an option to run e.g. AVX512 commands and catch `illegal instruction error`, we might just use that to detect the system we're on.

michaelld, 2021-09-26:
I just started looking at this project: https://github.com/simd-everywhere/simde.git ... it's pretty cool!

michaelld, 2021-09-26:
Maybe as a compromise at least right now, we tweak the CMake scripts to check for Apple M1 & if that succeeds then we don't do `cpu_features`. If not M1 then we do `cpu_features` & hope for the best. That would at least take care of the immediate need: proper Volk code to get it building on M1.

michaelld, 2021-09-26:
Thinking of SIMDe for the CPU detection more-so than their actual code ... though that's pretty cool too!

juanibuqt, 2021-09-30:
It seems that Volk without M1 support is making a headache to lot of us! Cant finishing installing SatDump here since Vosk is a must

jdemel, 2021-09-30:
If you compile from source, you can fix it by patching `cpu_features`. VOLK itself is ready for M1. But the library we use to detect CPU features is not. The version you obtain via conda receives this patch as well. At least as far as I know. You might want to double check.

jdemel, 2021-12-19:
It seems like the relevant patch got merged 
https://github.com/google/cpu_features/commit/69d39934e83be9a0133d2850f15f5debd5eba06f
I can't test it on an M1 at the moment. However, this commit claims to fix our issue as well.

PR #550 (2021-12-23): cpu_features: Update submodule pointer
We have 3 issues that should be fixed with this commit.

Fix #428
Should be fixed because cpu_features detects `arm64` now. Thus, VOLK builds with cpu_features on MacOS and reports M1 capabilities.

Fix #478
Fix #484
cpu_features received quite a bit of contributions for FreeBSD. All the issues we had should be fixed now. However, this might require further evaluation.
commit c8198263 (2021-12-23): cpu_features: Update submodule pointer

We have 3 issues that should be fixed with this commit.

Fix #428
Should be fixed because cpu_features detects `arm64` now. Thus, it
builds on MacOS and reports M1 capabilities.

Fix #478
Fix #484
cpu_features received quite a bit of contributions for FreeBSD. All the
issues we had should be fixed now. However, this might require further
evaluation.
closed 2021-12-23 by michaelld