gnuradio/volk #430: error: ‘isinf’ was not declared in this scope

cherchais, 2020-12-27:
When I installed gnuradio3.8 using pybombs, I encountered the following error in compiling libvolk:
```bash
 Install tree:
|
\- gnuradio
   |
   \- libvolk
[INFO] Phase 1 complete: All binary dependencies installed.
[INFO] Phase 2: Recursively installing source packages to prefix:
[INFO] Installing package: libvolk
[WARNING] Build dir already exists: /home/lsh/pybombs/bladeRF/src/libvolk/build
Building:    (100%) [=========================================================================================================]
[  5%] Built target utils
[  8%] Built target cpu_features
[ 10%] Built target unix_based_hardware_detection
[ 13%] Built target list_cpu_features
[ 76%] Built target volk_obj
[ 77%] Built target volk
[ 79%] Building CXX object lib/CMakeFiles/volk_test_all.dir/testqa.cc.o
In file included from /home/lsh/pybombs/bladeRF/src/libvolk/build/include/volk/volk.h:31:0,
                 from /home/lsh/pybombs/bladeRF/src/libvolk/lib/qa_utils.h:5,
                 from /home/lsh/pybombs/bladeRF/src/libvolk/lib/kernel_tests.h:1,
                 from /home/lsh/pybombs/bladeRF/src/libvolk/lib/testqa.cc:31:
/home/lsh/pybombs/bladeRF/src/libvolk/include/volk/volk_common.h: In function ‘float log2f_non_ieee(float)’:
/home/lsh/pybombs/bladeRF/src/libvolk/include/volk/volk_common.h:153:24: error: ‘isinf’ was not declared in this scope
     return isinf(result) ? copysignf(127.0f, result) : result;
                        ^
/home/lsh/pybombs/bladeRF/src/libvolk/include/volk/volk_common.h:153:24: note: suggested alternative:
In file included from /usr/include/c++/5/complex:44:0,
                 from /home/lsh/pybombs/bladeRF/src/libvolk/include/volk/volk_complex.h:23,
                 from /home/lsh/pybombs/bladeRF/src/libvolk/build/include/volk/volk_typedefs.h:30,
                 from /home/lsh/pybombs/bladeRF/src/libvolk/build/include/volk/volk.h:29,
                 from /home/lsh/pybombs/bladeRF/src/libvolk/lib/qa_utils.h:5,
                 from /home/lsh/pybombs/bladeRF/src/libvolk/lib/kernel_tests.h:1,
                 from /home/lsh/pybombs/bladeRF/src/libvolk/lib/testqa.cc:31:
/usr/include/c++/5/cmath:621:5: note:   ‘std::isinf’
     isinf(_Tp __x)
     ^
lib/CMakeFiles/volk_test_all.dir/build.make:62: recipe for target 'lib/CMakeFiles/volk_test_all.dir/testqa.cc.o' failed
make[2]: *** [lib/CMakeFiles/volk_test_all.dir/testqa.cc.o] Error 1
CMakeFiles/Makefile2:400: recipe for target 'lib/CMakeFiles/volk_test_all.dir/all' failed
make[1]: *** [lib/CMakeFiles/volk_test_all.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
[ERROR] Build failed. See output above for error messages.
[ERROR] Problem occurred while building package libvolk:
Build failed.
[ERROR] Error installing package libvolk. Aborting.
```
I checked the file volk_common.h and it imported the math.h library
I don`t known how to deal with this problems,Can anyone help me?

rear1019, 2020-12-27:
It looks like you use a somewhat old compiler (GCC 5?) and [1] might be the cause of the error. 

@jdemel I am not sure how to deal with this. Fixing this might require some error prone and ugly checking of compiler and C/C++ library versions. On the other hand, I don’t think that `log2f_non_ieee()` (and some other definitions) should be part of `volk_common.h` anyway. 

[1] https://developers.redhat.com/blog/2016/02/29/why-cstdlib-is-more-complicated-than-you-might-think

cherchais, 2020-12-28:
I use ubuntu16.04 and the default gcc version of it should be GCC5.4

And I suspect that the system version(or may be the old compiler) is too low, so I reconfigured it with ubuntu18, and it works.

Thank you very much!

cherchais, 2020-12-28:
> It looks like you use a somewhat old compiler (GCC 5?) and [1] might be the cause of the error.
>
> @jdemel I am not sure how to deal with this. Fixing this might require some error prone and ugly [...]

I use ubuntu16.04 and the default gcc version of it should be GCC5.4

And I suspect that the system version(or may be the old compiler) is too low, so I reconfigured it with ubuntu18, and it works.

Thank you very much!

rear1019, 2020-12-28:
GCC 5 is recent enough to support C++11 which is required for GNU Radio 3.8. Assuming that other dependencies are satisfied by Ubuntu 16.04 as well, GNU Radio (including Volk) should compile… Switching to more recent compiler and C/C++ standard libraries is the most simple solution. Some GCC versions have fixes for `math.h` and `cmath` headers, at least I remember reading that in release notes.

The error is related to #281 given that `volk_common.h` is used by C and C++ compilers.

PS: Found the release notes for GCC I talked about [1][2]. Search for “math”.

[1] https://gcc.gnu.org/gcc-6/porting_to.html
[2] https://gcc.gnu.org/gcc-7/porting_to.html

jdemel, 2020-12-28:
This error is known and basically a "wontfix". See #375 and #378 . Basically this specific Ubuntu version in combination with GCC5 seems to have a bug here that is fixed later. In essence this is a Ubuntu bug in 16.04 which is nearing EOL.

`log2f_no_ieee` ended up in `volk_common.h` because it is used in multiple kernels. At the time, this seemed like the best solution. Of course, I'd be glad to discuss a better solution.

cherchais, 2020-12-29:
> This error is known and basically a "wontfix". See #375 and #378 . Basically this specific Ubuntu version in combination with GCC5 seems to have a bug here that is fixed later. In essence this is a Ubuntu bug in 16.04 which is nearing EOL.
> 
> `log2f_no_ieee` ended up in because it is used in multiple kernels. At the time, this seemed like the best solution. Of course, I'd be glad to discuss a better solution.`volk_common.h`

I get it，thank you!

closed 2020-12-29 by cherchais