gnuradio/volk #431: Consistent function prototypes are not enforced

rear1019, 2020-12-28:
While working on #368 I ended up using an inconsistent function prototype:
```c
volk_32fc_index_max_16u_a_avx2(uint32_t* target, lv_32fc_t* src0, uint32_t num_points)
```
instead of the required
```c
volk_32fc_index_max_16u_a_avx2(uint16_t* target, lv_32fc_t* src0, uint32_t num_points)
```
Note the different data type of the first argument.

This error has been detected by the compiler which issued a warning. We should consider doing at least one of the following:
* Turn `-Wincompatible-pointer-types` and possibly `-Wpointer-sign` warnings into errors.
* Check the prototypes in the generator scripts. I have written some code for this.

jdemel, 2021-01-02:
I like that suggestion. Did you run a test how many kernels need to be fixed or which kernels need to be fixed if we activate that switch?

rear1019, 2021-01-03:
> I like that suggestion. Did you run a test how many kernels need to be fixed or which kernels need to be fixed if we activate that switch?

None. `-Wincompatible-pointer-types` is a warning enabled by default. `-Wpointer-sign` is enabled by `-Wall` (which we do enable). See PR #433 which makes these warnings errors.

PR #433 (2021-01-06): Enforce consistent function prototypes
See commit message/code changes.

This fixes #431.
closed 2021-01-06 by michaelld