gnuradio/volk #436: volk cross build on Xilinx e3xx fails due to incorrect processor flags

mgarrett1955, 2021-01-15:
In building gnuradio 3.8  and volk for the ettus e3xx sdk version 4.0 (and 3.15) I get the following error:

/home/mgarrett/rfnoc/src/gnuradio/volk/kernels/volk/asm/neon/volk_16i_max_star_horizontal_16i.s: Assembler messages:
/home/mgarrett/rfnoc/src/gnuradio/volk/kernels/volk/asm/neon/volk_16i_max_star_horizontal_16i.s:8: Error: selected processor does not support `pld [r1,#128]' in ARM mode
/home/mgarrett/rfnoc/src/gnuradio/volk/kernels/volk/asm/neon/volk_16i_max_star_horizontal_16i.s:19: Error: selected processor does not support `pld [r4,#128]' in ARM mode
/home/mgarrett/rfnoc/src/gnuradio/volk/kernels/volk/asm/neon/volk_16i_max_star_horizontal_16i.s:43: Error: selected processor does not support `sxth r6,r6' in ARM mode
make[2]: *** [volk/lib/CMakeFiles/volk_obj.dir/build.make:1755: volk/lib/CMakeFiles/volk_obj.dir/__/kernels/volk/asm/neon/volk_16i_max_star_horizontal_16i.s.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:265: volk/lib/CMakeFiles/volk_obj.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

If I read the gnuradio-config_info --cflags on the e310 (uhd 4.0, gnuradio 3.8), and see the following processor switches:
-mthumb -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a9 on the e310.

I really appreciate any help on this matter !!

mgarrett1955, 2021-01-15:
I am executing the following CMAKE command:

 cmake -Wno-dev -DCMAKE_TOOLCHAIN_FILE=~/rfnoc/src/gnuradio/cmake/Toolchains/oe-sdk_cross.cmake -DENABLE_GR_WXGUI=OFF -DENABLE_GR_VOCODER=OFF -DENABLE_GR_DTV=OFF -DENABLE_GR_ATSC=OFF -DENABLE_DOXYGEN=OFF -DCMAKE_INSTALL_PREFIX=/usr ../

Which point to the oe-sdk_cross.make file, in that same directory is a file called arm_cortexa8_native.cmake, Is thi supposed to be updated to an a9 processor directive?

balister, 2021-01-16:
Look at this commit: https://github.com/gnuradio/volk/commit/92c16d1fb4a08ba88441baf7f4a99fcbeb075618

balister, 2021-01-16:
The oe-sdk file is the one to use with the sdk. It may need the CMAKE_ASM... directive referenced above. Generally, I use an external volk, which should also work for you.

mgarrett1955, 2021-01-17:
Thank you for the reply!! I can see on the e310, that the arm directive it was built with is-mcpu=cortex-a9, which most likely came from the cmake/Toolchains/arm_cortex_a9_hardfp_native.cmake file during the build.

My gnuradio distribution G3.7, only contnains the file for the a8, and that file does not contain the assembler directive!!

So I have a missing file in the main-3.7 branch, and the one that is there, does not have the assembler directive.

Thank you for the insight and for solving my problem!

mgarrett1955, 2021-01-17:
There does not seem to be an arm_9 file in any gnuradio release, so have to add it as a cmake directive. In none of the e310 documentation I can find do they force that directive!!

jdemel, 2021-01-17:
I neither have the hardware nor the set up available to test any of this. Could you add all the steps to reproduce your issue please? This might help to discuss a solution and it would probably help others with the same issue if they come across this thread. 
Is this an issue with GNU Radio or VOLK? Or both? Which files are you referring to?

mgarrett1955, 2021-01-19:
Yes I will do that!! Currently my volk is in the gnuradio tree. I have not been able to figure out how to solve the issue. I added compiler switches to force the c, cpp and as flags, but recieved the same error. SOmehow Ettus created the SDK for the e31x with the correct switches but I can not figure it out. I get the same error compiling GR3.7.14 (maint-3.7) and the 4.0 Head. The files included in the new volk tree for the arm 9, do not appear in the gnuradio tree. I will update if I figure anything out. Ettus has the flags somewhere, but not in any supplied documentation for cross compiling for the e31x.

jdemel, 2021-01-19:
I'm still trying to understand the issue.
There's an `arm9` switch that seems to be missing in a CMake toolchain file. Which one exactly? Could you add a permalink to the line with the `arm9` switch in the VOLK repo?
Also, you mentioned that the `arm9` switch is present in the VOLK repository but missing in the GNU Radio repository? Obviously this needs to be fixed in the correct repository.
If the problem is that GNU Radio is currently missing the a toolchain file (or it needs a fix) then this needs to go to the gnuradio repo. We can't possibly fix issues beyond our scope. Still, if there's anything missing in VOLK that would help, let's discuss this.

mgarrett1955, 2021-01-20:
jdeml, I will try to compile volk separately. What are the cmak switches to call on the command line to force use of the arm_cortex_a9_hardfp_native.cmake compiler settings, or does it figure that out in the oe-sdk_cross.cmake script based on the cross complier configuration in the shell?

mgarrett1955, 2021-01-28:
Just a status update:

I was able to get uhd-3.15 / gnuradio maint-3.7 volk to cross compile by adding the following flag  
-DCMAKE_ASM_COMPILER_ARG1:STRING="-mcpu=cortex-a9" 
(Gnuradio failed but for a different reason)

I am in the process of building a maint-3.8, UHD 4.0 cross build environment and will report back findings.

mgarrett1955, 2021-01-28:
Ok... In order to cross build volk in the gnuradio maint-3.8 branch I have to add:
-DCMAKE_ASM_COMPILER_ARG1:STRING="-mcpu=cortex-a9"

To get through the missing pld instruction error, now I get a different error:

/home/mgarrett/rfnoc/src/gnuradio/volk/lib/volk_malloc.c: In function ‘volk_malloc’:
/home/mgarrett/rfnoc/src/gnuradio/volk/lib/volk_malloc.c:49:13: warning: implicit declaration of function ‘posix_memalign’ [-Wimplicit-function-declaration]
   49 |   int err = posix_memalign(&ptr, alignment, size);
      |             ^~~~~~~~~~~~~~
[  4%] Building C object volk/lib/CMakeFiles/volk_obj.dir/volk.c.o
[  5%] Building C object volk/lib/CMakeFiles/volk_obj.dir/volk_cpu.c.o
[  5%] Building C object volk/lib/CMakeFiles/volk_obj.dir/volk_machines.c.o
[  5%] Building C object volk/lib/CMakeFiles/volk_obj.dir/volk_machine_generic.c.o
[  5%] Building C object volk/lib/CMakeFiles/volk_obj.dir/volk_machine_neon.c.o
[  5%] Building C object volk/lib/CMakeFiles/volk_obj.dir/volk_machine_neonv7_hardfp.c.o
[  5%] Building C object volk/lib/CMakeFiles/volk_obj.dir/volk_machine_neonv8.c.o
In file included from /home/mgarrett/rfnoc/src/gnuradio/build-arm/volk/lib/volk_machine_neonv8.c:143:
/home/mgarrett/rfnoc/src/gnuradio/volk/kernels/volk/volk_32u_byteswap.h: In function ‘volk_32u_byteswap_neonv8’:
/home/mgarrett/rfnoc/src/gnuradio/volk/kernels/volk/volk_32u_byteswap.h:215:13: warning: implicit declaration of function ‘vqtbl1q_u8’; did you mean ‘vtbl1_u8’? [-Wimplicit-function-declaration]
  215 |     input = vqtbl1q_u8(input, idx);
      |             ^~~~~~~~~~
      |             vtbl1_u8
/home/mgarrett/rfnoc/src/gnuradio/volk/kernels/volk/volk_32u_byteswap.h:215:13: error: incompatible types when assigning to type ‘uint8x16_t’ from type ‘int’
/home/mgarrett/rfnoc/src/gnuradio/volk/kernels/volk/volk_32u_byteswap.h:220:13: error: incompatible types when assigning to type ‘uint8x16_t’ from type ‘int’
  220 |     input = vqtbl1q_u8(input, idx);
      |             ^~~~~~~~~~
In file included from /home/mgarrett/rfnoc/src/gnuradio/build-arm/volk/lib/volk_machine_neonv8.c:147:
/home/mgarrett/rfnoc/src/gnuradio/volk/kernels/volk/volk_32u_reverse_32u.h: In function ‘volk_32u_reverse_32u_neonv8’:
/home/mgarrett/rfnoc/src/gnuradio/volk/kernels/volk/volk_32u_reverse_32u.h:352:49: warning: implicit declaration of function ‘vrbitq_u8’; did you mean ‘vbicq_u8’? [-Wimplicit-function-declaration]
  352 |  uint32x4_t z = vreinterpretq_u32_u8(vqtbl1q_u8(vrbitq_u8(vreinterpretq_u8_u32 (x)),
      |                                                 ^~~~~~~~~
      |                                                 vbicq_u8
/home/mgarrett/rfnoc/src/gnuradio/volk/kernels/volk/volk_32u_reverse_32u.h:352:38: error: incompatible type for argument 1 of ‘vreinterpretq_u32_u8’
  352 |  uint32x4_t z = vreinterpretq_u32_u8(vqtbl1q_u8(vrbitq_u8(vreinterpretq_u8_u32 (x)),
      |                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                      |
      |                                      int
  353 |              idx));
      |              ~~~~                     
In file included from /home/mgarrett/rfnoc/src/gnuradio/volk/kernels/volk/volk_16i_32fc_dot_prod_32fc.h:96,
                 from /home/mgarrett/rfnoc/src/gnuradio/build-arm/volk/lib/volk_machine_neonv8.c:41:
/home/mgarrett/rfnoc/oe/sysroots/x86_64-oesdk-linux/usr/lib/arm-oe-linux-gnueabi/gcc/arm-oe-linux-gnueabi/9.2.0/include/arm_neon.h:16829:34: note: expected ‘uint8x16_t’ but argument is of type ‘int’
16829 | vreinterpretq_u32_u8 (uint8x16_t __a)
      |                       ~~~~~~~~~~~^~~
In file included from /home/mgarrett/rfnoc/src/gnuradio/build-arm/volk/lib/volk_machine_neonv8.c:153:
/home/mgarrett/rfnoc/src/gnuradio/volk/kernels/volk/volk_64u_byteswap.h: In function ‘volk_64u_byteswap_neonv8’:
/home/mgarrett/rfnoc/src/gnuradio/volk/kernels/volk/volk_64u_byteswap.h:254:20: error: incompatible types when assigning to type ‘uint8x16_t’ from type ‘int’
  254 |     input.val[0] = vqtbl1q_u8(input.val[0], idx);
      |                    ^~~~~~~~~~
/home/mgarrett/rfnoc/src/gnuradio/volk/kernels/volk/volk_64u_byteswap.h:255:20: error: incompatible types when assigning to type ‘uint8x16_t’ from type ‘int’
  255 |     input.val[1] = vqtbl1q_u8(input.val[1], idx);
      |                    ^~~~~~~~~~
volk/lib/CMakeFiles/volk_obj.dir/build.make:1949: recipe for target 'volk/lib/CMakeFiles/volk_obj.dir/volk_machine_neonv8.c.o' failed
make[2]: *** [volk/lib/CMakeFiles/volk_obj.dir/volk_machine_neonv8.c.o] Error 1
CMakeFiles/Makefile2:264: recipe for target 'volk/lib/CMakeFiles/volk_obj.dir/all' failed
make[1]: *** [volk/lib/CMakeFiles/volk_obj.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

mgarrett1955, 2021-01-29:
My SDCARD and SDK provided by Ettus, have information that I could give you is there anything I can provide you to assist on figuring this out? Let me know!

jdemel, 2021-01-29:
Your last error message sounds like something is missing on your host. `posix_memalign` should probably only ever be used on MacOS at this point. Unless you use an older VOLK release.

Could you give us more details on your host? Maybe CMake is not working correctly. 
Which version of VOLK do you use?
Which compiler do you use?
What where the exact steps that led to your error? Build commands are certainly useful.
Do you still try to build VOLK together with GNU Radio? In that case, could you try to build a stand-alone version of VOLK?
How did you get your copy of VOLK? Via `git clone --recursive <gnuradio_repo>`? In that case an update to the latest VOLK release would be helpful.

mgarrett1955, 2021-01-29:
I will start from scratch and give you every step. Im working in a Vmware Ubutnu 18.04LTS machine currently.
Im not sure If ill start with Ubuntu 10.04.1LTS or 18.04LTS Ill let you know.

Im going to build UHD 4.0 using the Ettus SDK, Ill post the revisions.

mgarrett1955, 2021-01-29:
A quick note, The build environment should be identical to the environment that the ETTUS sdk was built on. I'm not sure how to get the Volk version from that. I can read the gnuradio-config file. But the idea is  to recreate the environment required for the SDK, to build images that are compatible with the SDK on the SD Card.

mgarrett1955, 2021-01-30:
Attached is the full sequence starting with a fresh Ubuntu 20.04LTS.
At the end is the results of 2 attempts the first with:

cmake -Wno-dev -DCMAKE_TOOLCHAIN_FILE=~/rfnoc/src/volk/cmake/Toolchains/oe-sdk_cross.cmake DENABLE_DOXYGEN=OFF -DCMAKE_INSTALL_PREFIX=/usr ../

and the second with:

cmake -Wno-dev -DCMAKE_TOOLCHAIN_FILE=~/rfnoc/src/volk/cmake/Toolchains/arm_cortex_a9_hardfp_native.cmake DENABLE_DOXYGEN=OFF -DCMAKE_INSTALL_PREFIX=/usr ../

In the second, the incorrect gcc is called for the test as indicated and annotated in the document. 

Here is the installation sequence. (I misspelled Installation.. )

[UHD e31x Installtion Notes.txt](https://github.com/gnuradio/volk/files/5897480/UHD.e31x.Installtion.Notes.txt)

mgarrett1955, 2021-01-30:
The oe-sdk_cross.cmake does not seem to be setting the asm flags?

mgarrett1955, 2021-01-30:
I added this flga to the cmake: -DCMAKE_ASM_COMPILER_ARG1:STRING="-mcpu=cortex-a9", which eliminated the pld, and sxth instruction errors but left the vmov,vld2.16,,vsub,vcge,vcgt,vand,vadd and vst1 errors.

I called cmake with the following directives:

cmake -Wno-dev -DCMAKE_TOOLCHAIN_FILE=~/rfnoc/src/volk/cmake/Toolchains/oe-sdk_cross.cmake -DCMAKE_CXX_FLAGS:STRING="-march=armv7-a -mfloat-abi=hard -mfpu=neon -mtune=cortex-a9" -DCMAKE_C_FLAGS:STRING="-march=armv7-a -mfloat-abi=hard -mfpu=neon -mtune=cortex-a9" -DCMAKE_ASM_FLAGS:STRING="-march=armv7-a -mfloat-abi=hard -mfpu=neon -mtune=cortex-a9 -g" -DENABLE_DOXYGEN=OFF -DCMAKE_INSTALL_PREFIX=/usr ../

Which actually Built!!!

I will now attempt to cross compile gnuradio maint-3.8 with a separate volk...

mgarrett1955, 2021-01-30:
I made it through cross compiling Volk and gnuradio maint-3.8, having a separate issue with gr-ettus. I beieve there may be a python issue version issue against the python on the e310. 

Volk compiled with the above flags added to the cmake command. I will validate the operation and make sure gnuradio runs, and will post a detailed build process for the complete chain, when I solve the volk issue.

mgarrett1955, 2021-01-30:
Here is the volk-profile run on the e310 with the sdk mounted to the device.
[volk-profile-e310.txt](https://github.com/gnuradio/volk/files/5898916/volk-profile-e310.txt)

mgarrett1955, 2021-01-30:
Here is the complete process all the way up to the last step which is the gr-ettus cross compile which is having an issue. This includes the Ubuntu 20.04, UHD 4.0, volk, gnuradio all host and cross compiled into the e310 sdk. The only remaining issue os the cross compile of gr-ettus.

[VM using Ubuntu 20.04LTS-gr-ettus issue.txt](https://github.com/gnuradio/volk/files/5898943/VM.using.Ubuntu.20.04LTS-gr-ettus.issue.txt)

jdemel, 2021-02-06:
VOLK is intended to run on specialized hardware and abstract that away for users. But VMs are not exactly the most reliable in passing on CPU features. We had a rather recent issue with MacOS Docker (outside Linux, Docker runs in a VM).

I assume the critical section to cross-compile VOLK is:
```
Cross Compile Volk:

cd ~/rfnoc/oe
source ./environment-setup-cortexa9t2hf-neon-oe-linux-gnueabi
echo $CC

arm-oe-linux-gnueabi-gcc -mthumb -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a9 --sysroot=/home/mgarrett/rfnoc/oe/sysroots/cortexa9t2hf-neon-oe-linux-gnueabi

cd ~/rfnoc/src/volk$
mkdir build-arm

----> Critical step in compiling volk for the e3xx
cmake -Wno-dev -DCMAKE_TOOLCHAIN_FILE=~/rfnoc/src/volk/cmake/Toolchains/oe-sdk_cross.cmake -DCMAKE_CXX_FLAGS:STRING="-march=armv7-a -mfloat-abi=hard -mfpu=neon -mtune=cortex-a9" -DCMAKE_C_FLAGS:STRING="-march=armv7-a -mfloat-abi=hard -mfpu=neon -mtune=cortex-a9" -DCMAKE_ASM_FLAGS:STRING="-march=armv7-a -mfloat-abi=hard -mfpu=neon -mtune=cortex-a9 -g" -DENABLE_DOXYGEN=OFF -DCMAKE_INSTALL_PREFIX=/usr ../
```
I'd like to isolate the issue here. 
- Is there an issue with any of the `volk/cmake/Toolchain` files? 
- Do you need another toolchain file?
- Your CMake flags suggest you need to overwrite most of the CMake variables that should be set. Is there something that needs to be fixed?
- What's the exact target? I know, it's an E310 but what's the exact CPU?

@balister could you chime in?

mgarrett1955, 2021-02-06:
Here is data from the 7000 Zync manual:

The application processing unit (APU), located within the PS, contains one processor for single-core
devices or two processors for dual-core devices.

 These are ARM® Cortex™-A9 processors with NEON
co-processors connected in an MP configuration sharing a 512 KB L2 cache. Each processor is a
high-performance and low-power core that implements two separate 32 KB L1 caches for instruction
and data. 

The Cortex-A9 processor implements the ARM v7-A architecture with full virtual memory
support and can execute 32-bit ARM instructions, 16-bit and 32-bit Thumb instructions, and 8-bit
Java™ byte codes in the Jazelle state. 

The NEON™ coprocessor media and signal processing
architecture adds instructions that target audio, video, image and speech processing, and 3D
graphics. 

These advanced single instruction multiple data (SIMD) instructions are available in both
ARM and Thumb states.

mgarrett1955, 2021-02-06:
I do not believe the Assembler switch are getting set in the current cmake structure.

mgarrett1955, 2021-04-24:
That did not work. I just can not get volk to cross compile for the Xilinx Zynq in the e310.

I tried this:
cmake -Wno-dev -DCMAKE_TOOLCHAIN_FILE=~/rfnoc/src/gnuradio/cmake/Toolchains/oe-sdk_cross.cmake -DENABLE_GR_WXGUI=OFF -DENABLE_GR_VOCODER=OFF -DENABLE_GR_DTV=OFF -DENABLE_GR_ATSC=OFF -DENABLE_DOXYGEN=OFF DCMAKE_ASM_FLAGS:STRING="-march=armv7-a -mfloat-abi=hard -mfpu=neon -mtune=cortex-a9 -g" -DCMAKE_INSTALL_PREFIX=/usr ../

WHich sets just the compiler switches and got this error:

[  4%] Building C object volk/lib/CMakeFiles/volk_obj.dir/volk_machine_neonv8.c.o
In file included from /home/mgarrett/rfnoc/src/gnuradio/build-arm/volk/lib/volk_machine_neonv8.c:143:
/home/mgarrett/rfnoc/src/gnuradio/volk/kernels/volk/volk_32u_byteswap.h: In function ‘volk_32u_byteswap_neonv8’:
/home/mgarrett/rfnoc/src/gnuradio/volk/kernels/volk/volk_32u_byteswap.h:215:13: warning: implicit declaration of function ‘vqtbl1q_u8’; did you mean ‘vtbl1_u8’? [-Wimplicit-function-declaration]
  215 |     input = vqtbl1q_u8(input, idx);
      |             ^~~~~~~~~~
      |             vtbl1_u8
/home/mgarrett/rfnoc/src/gnuradio/volk/kernels/volk/volk_32u_byteswap.h:215:13: error: incompatible types when assigning to type ‘uint8x16_t’ from type ‘int’
/home/mgarrett/rfnoc/src/gnuradio/volk/kernels/volk/volk_32u_byteswap.h:220:13: error: incompatible types when assigning to type ‘uint8x16_t’ from type ‘int’
  220 |     input = vqtbl1q_u8(input, idx);

mgarrett1955, 2021-04-24:
I am trying this again in the cmake:

-DCMAKE_CXX_FLAGS:STRING="-march=armv7-a -mfloat-abi=hard -mfpu=neon -mtune=cortex-a9" -DCMAKE_C_FLAGS:STRING="-march=armv7-a -mfloat-abi=hard -mfpu=neon -mtune=cortex-a9" -DCMAKE_ASM_FLAGS:STRING="-march=armv7-a -mfloat-abi=hard -mfpu=neon -mtune=cortex-a9 -g"

mgarrett1955, 2021-04-24:
Same error.

[  4%] Building C object volk/lib/CMakeFiles/volk_obj.dir/volk_machine_neonv8.c.o
In file included from /home/mgarrett/rfnoc/src/gnuradio/build-arm/volk/lib/volk_machine_neonv8.c:143:
/home/mgarrett/rfnoc/src/gnuradio/volk/kernels/volk/volk_32u_byteswap.h: In function ‘volk_32u_byteswap_neonv8’:
/home/mgarrett/rfnoc/src/gnuradio/volk/kernels/volk/volk_32u_byteswap.h:215:13: warning: implicit declaration of function ‘vqtbl1q_u8’; did you mean ‘vtbl1_u8’? [-Wimplicit-function-declaration]
  215 |     input = vqtbl1q_u8(input, idx);
      |             ^~~~~~~~~~
      |             vtbl1_u8
/home/mgarrett/rfnoc/src/gnuradio/volk/kernels/volk/volk_32u_byteswap.h:215:13: error: incompatible types when assigning to type ‘uint8x16_t’ from type ‘int’
/home/mgarrett/rfnoc/src/gnuradio/volk/kernels/volk/volk_32u_byteswap.h:220:13: error: incompatible types when assigning to type ‘uint8x16_t’ from type ‘int’
  220 |     input = vqtbl1q_u8(input, idx);

mgarrett1955, 2021-04-24:
THere is a cross-compiled version in the oe/sysroots sdk, and a version on the sd card.... can I probe those for the correct build calls?

bkerler, 2022-07-18:
I was able to sort out the issue by patching the volk/lib/CMakeLists.txt as the neonv8 detection returns invalid results and thus neonv7 needs to be enforced on the zynq. Just copy the patch file to the package/gnuradio directory in buildroot. Then gnuradio just compiles fine.
[0005-force_neonv7_zynq.patch.txt](https://github.com/gnuradio/volk/files/9132139/0005-force_neonv7_zynq.patch.txt)