gnuradio/volk #441: volk_profile segmentation fault (master branch)

crroush, 2021-01-18:
I ran volk_profile (using a custom install directory) and manually setting my LD_LIBRARY_PATH on a ubuntu 20.04 and gcc 8
```
mkdir build
cd build
cmake .. -DCMAKE_INSTALL_PREFIX=../opt
make -j64
make test
make install

```
After running for a while it eventually segfaults, which is shown below and the back trace coming out of gdb:
I did try to rerun with the --path option to see if it would work, but it failed in the same manner. 
```
RUN_VOLK_TESTS: volk_32fc_s32f_power_spectral_densitypuppet_32f(131071,1987)
a_avx completed in 1169.55 ms
a_sse3 completed in 1170.15 ms
generic completed in 1096.39 ms
Best aligned arch: generic
Best unaligned arch: generic

Program received signal SIGSEGV, Segmentation fault.
0x000055e6edc938bb in std::filesystem::__cxx11::path::~path() ()
(gdb) bt
#0  0x000055e6edc938bb in std::filesystem::__cxx11::path::~path() ()
#1  0x000055e6edc938dc in std::filesystem::__cxx11::path::~path() ()
#2  0x000055e6edc938dc in std::filesystem::__cxx11::path::~path() ()
#3  0x000055e6edc9205c in write_results(std::vector<volk_test_results_t, std::allocator<volk_test_results_t> > const*, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) ()
#4  0x000055e6edc927ac in write_results(std::vector<volk_test_results_t, std::allocator<volk_test_results_t> > const*, bool) ()
#5  0x000055e6edc8ab00 in main ()

```

jdemel, 2021-01-18:
Could you share some more info about your system? Currently, I gathered Ubuntu 20.04 with GCC8. Which is actually surprising because I'd expect GCC 9 to be available there. Any reason to use an older GCC version?
I assume you work with a large CPU, given you do `-j64`?
Personally, I use custom install locations quite frequently and things work for me. So the thing you do should work. 

The backtrace indicates that the destructor `std::filesystem::path::~path` segfaults.

Do you have multiple VOLK installs on your system? e.g. did you install one via package manager? Can you check which `libvolk.so` is linked against `volk_profile`? `ldd volk_profile` should do the trick.

crroush, 2021-01-18:
@jdemel  sure thing.  I am using gcc8 because of other software I am working on targeted platform uses GCC8, so I want to ensure I test in the same environment that I will use the software.
I generally use nproc*2 for any of my building, that seems to work well.```

System:    Host: warlock Kernel: 5.8.0-7630-generic x86_64 bits: 64 Desktop: Gnome 3.36.4 Distro: Pop!_OS 20.04 LTS 
Machine:   Type: Desktop System: Alienware product: Alienware Aurora Ryzen Edition v: 1.0.8 serial: <superuser/root required> 
           Mobo: Alienware model: 0NWN7M v: A00 serial: <superuser/root required> UEFI: Alienware v: 1.0.8 date: 08/11/2020 
CPU:       Topology: 16-Core (2-Die) model: AMD Ryzen 9 3950X bits: 64 type: MT MCP MCM L2 cache: 8192 KiB 
           Speed: 2196 MHz min/max: 2200/3500 MHz Core speeds (MHz): 1: 2192 2: 2195 3: 2196 4: 2192 5: 2195 6: 2202 7: 2195 
           8: 2192 9: 2192 10: 2195 11: 2188 12: 2194 13: 2196 14: 2196 15: 2198 16: 2196 17: 2195 18: 2191 19: 2192 20: 2188 
           21: 2195 22: 2195 23: 2196 24: 2196 25: 2196 26: 2196 27: 2194 28: 2191 29: 2195 30: 2203 31: 2195 32: 2193 

```
ldd points to this:
libvolk.so.2.4 => /opt/projects/volk/opt/lib/libvolk.so.2.4 (0x00007f63e1a13000)

I reran in a docker container and got the same results of a segfault.  I probably need to rebuild with debug flags so that we are 100% sure the segfault is pointing to the right place.  
I verified volk is not installed at the system level.

jdemel, 2021-01-18:
I assume you git cloned into `/opt/projects/volk`?
 My next guess would be to check how `filesystem` is found and included. It might interfere with your GCC9 install. And I suppose `std::filesystem` changed between 8 and 9. Do you have a special filesystem that might cause the issue?
It's really hard to tell what's going on. After all, it segfaults in an STL destructor.

crroush, 2021-01-18:
@jdemel up that is exactly what I did and I built off of master.  ```
   update-alternatives --install /usr/bin/gcc gcc  /usr/bin/gcc-8 80 \
                          --slave /usr/bin/g++ g++ /usr/bin/g++-8  \
                          --slave /usr/bin/gfortran gfortran /usr/bin/gfortran-8  \
                          --slave /usr/bin/gcov gcov /usr/bin/gcov-8       
```
I rebuilt with  cmake .. -DCMAKE_INSTALL_PREFIX=../opt -DCMAKE_BUILD_TYPE=Debug

```
(gdb) bt
#0  0x00005601e60b85bd in std::vector<std::filesystem::__cxx11::path::_Cmpt, std::allocator<std::filesystem::__cxx11::path::_Cmpt> >::~vector (
    this=0x6672657474756292, __in_chrg=<optimized out>) at /usr/include/c++/8/bits/stl_vector.h:567
#1  0x00005601e60b753e in std::filesystem::__cxx11::path::~path (this=0x6672657474756272, __in_chrg=<optimized out>)
    at /usr/include/c++/8/bits/fs_path.h:209
#2  0x00005601e60be3ce in std::filesystem::__cxx11::path::_Cmpt::~_Cmpt (this=0x6672657474756272, __in_chrg=<optimized out>)
    at /usr/include/c++/8/bits/fs_path.h:644
#3  0x00005601e60be3ed in std::_Destroy<std::filesystem::__cxx11::path::_Cmpt> (__pointer=0x6672657474756272)
    at /usr/include/c++/8/bits/stl_construct.h:98
#4  0x00005601e60bd1d9 in std::_Destroy_aux<false>::__destroy<std::filesystem::__cxx11::path::_Cmpt*> (__first=0x6672657474756272, __last=0x41)
    at /usr/include/c++/8/bits/stl_construct.h:108
#5  0x00005601e60bb6a0 in std::_Destroy<std::filesystem::__cxx11::path::_Cmpt*> (__first=0x6672657474756272, __last=0x41)
    at /usr/include/c++/8/bits/stl_construct.h:137
#6  0x00005601e60b9e9b in std::_Destroy<std::filesystem::__cxx11::path::_Cmpt*, std::filesystem::__cxx11::path::_Cmpt> (__first=0x6672657474756272,
    __last=0x41) at /usr/include/c++/8/bits/stl_construct.h:206
#7  0x00005601e60b85d3 in std::vector<std::filesystem::__cxx11::path::_Cmpt, std::allocator<std::filesystem::__cxx11::path::_Cmpt> >::~vector (
    this=0x5601e72c8e80, __in_chrg=<optimized out>) at /usr/include/c++/8/bits/stl_vector.h:567
#8  0x00005601e60b753e in std::filesystem::__cxx11::path::~path (this=0x5601e72c8e60, __in_chrg=<optimized out>) at /usr/include/c++/8/bits/fs_path.h:209
#9  0x00005601e60be3ce in std::filesystem::__cxx11::path::_Cmpt::~_Cmpt (this=0x5601e72c8e60, __in_chrg=<optimized out>)
    at /usr/include/c++/8/bits/fs_path.h:644
#10 0x00005601e60be3ed in std::_Destroy<std::filesystem::__cxx11::path::_Cmpt> (__pointer=0x5601e72c8e60) at /usr/include/c++/8/bits/stl_construct.h:98
#11 0x00005601e60bd1d9 in std::_Destroy_aux<false>::__destroy<std::filesystem::__cxx11::path::_Cmpt*> (__first=0x5601e72c8e60, __last=0x2)
    at /usr/include/c++/8/bits/stl_construct.h:108
#12 0x00005601e60bb6a0 in std::_Destroy<std::filesystem::__cxx11::path::_Cmpt*> (__first=0x5601e72c8e60, __last=0x2)
    at /usr/include/c++/8/bits/stl_construct.h:137
#13 0x00005601e60b9e9b in std::_Destroy<std::filesystem::__cxx11::path::_Cmpt*, std::filesystem::__cxx11::path::_Cmpt> (__first=0x5601e72c8e60,
    __last=0x2) at /usr/include/c++/8/bits/stl_construct.h:206
#14 0x00005601e60b85d3 in std::vector<std::filesystem::__cxx11::path::_Cmpt, std::allocator<std::filesystem::__cxx11::path::_Cmpt> >::~vector (
    this=0x5601e71c4a60, __in_chrg=<optimized out>) at /usr/include/c++/8/bits/stl_vector.h:567
#15 0x00005601e60b753e in std::filesystem::__cxx11::path::~path (this=0x5601e71c4a40, __in_chrg=<optimized out>) at /usr/include/c++/8/bits/fs_path.h:209
#16 0x00005601e60be3ce in std::filesystem::__cxx11::path::_Cmpt::~_Cmpt (this=0x5601e71c4a40, __in_chrg=<optimized out>)
    at /usr/include/c++/8/bits/fs_path.h:644
#17 0x00005601e60be3ed in std::_Destroy<std::filesystem::__cxx11::path::_Cmpt> (__pointer=0x5601e71c4a40) at /usr/include/c++/8/bits/stl_construct.h:98
#18 0x00005601e60bd1d9 in std::_Destroy_aux<false>::__destroy<std::filesystem::__cxx11::path::_Cmpt*> (__first=0x5601e71c4a40, __last=0x1)
    at /usr/include/c++/8/bits/stl_construct.h:108
#19 0x00005601e60bb6a0 in std::_Destroy<std::filesystem::__cxx11::path::_Cmpt*> (__first=0x5601e71c4a40, __last=0x1)
    at /usr/include/c++/8/bits/stl_construct.h:137
#20 0x00005601e60b9e9b in std::_Destroy<std::filesystem::__cxx11::path::_Cmpt*, std::filesystem::__cxx11::path::_Cmpt> (__first=0x5601e71c4a40,
    __last=0x1) at /usr/include/c++/8/bits/stl_construct.h:206
#21 0x00005601e60b85d3 in std::vector<std::filesystem::__cxx11::path::_Cmpt, std::allocator<std::filesystem::__cxx11::path::_Cmpt> >::~vector (
    this=0x7ffffc54cf90, __in_chrg=<optimized out>) at /usr/include/c++/8/bits/stl_vector.h:567
#22 0x00005601e60b753e in std::filesystem::__cxx11::path::~path (this=0x7ffffc54cf70, __in_chrg=<optimized out>) at /usr/include/c++/8/bits/fs_path.h:209
#23 0x00005601e60b65e6 in write_results (results=0x7ffffc54d640, update_result=false, path="/home/xmmgr/.volk/volk_config")
    at /opt/projects/volk/apps/volk_profile.cc:276
#24 0x00005601e60b64e0 in write_results (results=0x7ffffc54d640, update_result=false) at /opt/projects/volk/apps/volk_profile.cc:262
--Type <RET> for more, q to quit, c to continue without paging--
#25 0x00005601e60b54fa in main (argc=1, argv=0x7ffffc54dbd8) at /opt/projects/volk/apps/volk_profile.cc:186
```
This stack trace + a little googling showed that with gcc8.x you have to include -lstdc++fs or it will not work.  After including that (forcefully in the CMAKE) it built without a segfault.  So probably the check_cxx_source_compiles isn't doing what is needed for this check.  
https://stackoverflow.com/questions/56845734/c17-filesystem-call-generates-segmentation-fault

jdemel, 2021-01-19:
I know that GCC moved filesystem from `std::experimental` to `std` and finally included the `-lstd++fs` switch by default. I assume our cmake code to detect how to include `std::filesystem`does not do everything it needs to do.

Just to verify that we're on the correct path: Can you add the `-lstd++fs` switch to `volk_profile` and check if it fixes your issue?

jdemel, 2021-01-29:
The current solution to check if the linker flag is needed, is certainly insufficient. Checking for GCC versions only would probably be insufficient as well. But I don't know the details for clang. This needs further investigation.

jdemel, 2021-01-31:
@crroush Can you check PR #443 and report if it fixes your issue? That would be great.

crroush, 2021-02-01:
@jdemel  I checked out the PR and it built and ran correctly for me.

jdemel, 2021-02-02:
That's great news. I added Actions to our CI that run GCC8 but couldn't see your issue. So I suppose this PR is a great fix.

closed 2021-02-14 by michaelld