gnuradio/volk #445: Build fails on Ubuntu for Raspberry Pi

hjf, 2021-02-02:
I'm using Ubuntu Server on a Raspberry Pi 4. I'm not using raspberry pi OS because it includes an older GCC version, and I need C++17 for another program I need to compile. 

I tried the instructions (```cmake -DCMAKE_TOOLCHAIN_FILE=../cmake/Toolchains/arm_cortex_a72_hardfp_native.cmake ```) but I got the error that gcc did not recognze -mfpu=neon (I think that was the error).
So I installed arm-linux-gnueabihf-gcc-10 and this time cmake works:

```
ubuntu@ubuntu:~/volk/build$ cmake ..
-- Build type not specified: defaulting to release.
-- Build type set to Release.
-- Extracting version information from git describe...
-- Building Volk with cpu_features
--
-- Python checking for distutils
-- Python checking for distutils - found
--
-- Python checking for python >= 3.4
-- Python checking for python >= 3.4 - found
--
-- Python checking for mako >= 0.4.2
-- Python checking for mako >= 0.4.2 - found
-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
-- orc files (missing: ORC_LIBRARY ORC_INCLUDE_DIR ORCC_EXECUTABLE)
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
-- QA Testing is enabled.
--   Modify using: -DENABLE_TESTING=ON/OFF
-- System profiling is disabled.
--   Modify using: -DENABLE_PROFILING=ON/OFF
-- Compiler name: GNU
-- Architecture is not x86 or x86_64, Overruled arch 3dnow
-- Architecture is not x86 or x86_64, Overruled arch mmx
-- Architecture is not x86 or x86_64, Overruled arch sse
-- Architecture is not x86 or x86_64, Overruled arch sse2
-- Architecture is not x86 or x86_64, Overruled arch sse3
-- Architecture is not x86 or x86_64, Overruled arch ssse3
-- Architecture is not x86 or x86_64, Overruled arch sse4_a
-- Architecture is not x86 or x86_64, Overruled arch sse4_1
-- Architecture is not x86 or x86_64, Overruled arch sse4_2
-- Architecture is not x86 or x86_64, Overruled arch avx
-- Architecture is not x86 or x86_64, Overruled arch avx512f
-- Architecture is not x86 or x86_64, Overruled arch avx512cd
-- Compiler doesn't support neonv8, Overruled arch neonv8
-- ORC support not found, Overruled arch orc
-- Available architectures: generic;hardfp;neon;neonv7;norc
-- Available machines: generic;neon;neonv7_hardfp
-- BUILD TYPE = RELEASE
-- Base cflags = -O3 -DNDEBUG -march=armv8-a -mtune=cortex-a72 -mfpu=neon-fp-armv8 -mfloat-abi=hard -fcx-limited-range -Wall -Werror=incompatible-pointer-types -Werror=pointer-sign
-- BUILD INFO ::: generic ::: GNU ::: -O3 -DNDEBUG -march=armv8-a -mtune=cortex-a72 -mfpu=neon-fp-armv8 -mfloat-abi=hard -fcx-limited-range -Wall -Werror=incompatible-pointer-types -Werror=pointer-sign
-- BUILD INFO ::: neon ::: GNU ::: -O3 -DNDEBUG -march=armv8-a -mtune=cortex-a72 -mfpu=neon-fp-armv8 -mfloat-abi=hard -fcx-limited-range -Wall -Werror=incompatible-pointer-types -Werror=pointer-sign -funsafe-math-optimizations
-- BUILD INFO ::: neonv7_hardfp ::: GNU ::: -O3 -DNDEBUG -march=armv8-a -mtune=cortex-a72 -mfpu=neon-fp-armv8 -mfloat-abi=hard -fcx-limited-range -Wall -Werror=incompatible-pointer-types -Werror=pointer-sign -funsafe-math-optimizations -mfpu=neon -funsafe-math-optimizations -mfloat-abi=hard
-- Compiler Version: gcc (Ubuntu 10.2.0-5ubuntu1~20.04) 10.2.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
-- ---- Adding ASM files
-- -- Detected neon architecture; enabling ASM
-- Adding source file: /home/ubuntu/volk/kernels/volk/asm/neon/volk_16i_max_star_horizontal_16i.s
-- Adding source file: /home/ubuntu/volk/kernels/volk/asm/neon/volk_32f_s32f_multiply_32f_a_neonasm.s
-- Adding source file: /home/ubuntu/volk/kernels/volk/asm/neon/volk_32f_x2_add_32f_a_neonasm.s
-- Adding source file: /home/ubuntu/volk/kernels/volk/asm/neon/volk_32f_x2_add_32f_a_neonpipeline.s
-- Adding source file: /home/ubuntu/volk/kernels/volk/asm/neon/volk_32f_x2_dot_prod_32f_a_neonasm.s
-- Adding source file: /home/ubuntu/volk/kernels/volk/asm/neon/volk_32f_x2_dot_prod_32f_a_neonasm_opts.s
-- Adding source file: /home/ubuntu/volk/kernels/volk/asm/neon/volk_32fc_32f_dot_prod_32fc_a_neonasm.s
-- Adding source file: /home/ubuntu/volk/kernels/volk/asm/neon/volk_32fc_32f_dot_prod_32fc_a_neonasmvmla.s
-- Adding source file: /home/ubuntu/volk/kernels/volk/asm/neon/volk_32fc_32f_dot_prod_32fc_a_neonpipeline.s
-- Adding source file: /home/ubuntu/volk/kernels/volk/asm/neon/volk_32fc_32f_dot_prod_32fc_a_unrollasm.s
-- Adding source file: /home/ubuntu/volk/kernels/volk/asm/neon/volk_32fc_x2_dot_prod_32fc_a_neonasm.s
-- Adding source file: /home/ubuntu/volk/kernels/volk/asm/neon/volk_32fc_x2_dot_prod_32fc_a_neonasm_opttests.s
-- Adding source file: /home/ubuntu/volk/kernels/volk/asm/neon/volk_32fc_x2_multiply_32fc_a_neonasm.s
-- c flags: -march=armv8-a -mtune=cortex-a72 -mfpu=neon-fp-armv8 -mfloat-abi=hard -fcx-limited-range -Wall -Werror=incompatible-pointer-types -Werror=pointer-sign;
-- asm flags: -march=armv8-a -mtune=cortex-a72 -mfpu=neon-fp-armv8 -mfloat-abi=hard -mthumb -g
-- c flags: -march=armv8-a -mtune=cortex-a72 -mfpu=neon-fp-armv8 -mfloat-abi=hard -fcx-limited-range -Wall -Werror=incompatible-pointer-types -Werror=pointer-sign;
-- asm flags: -march=armv8-a -mtune=cortex-a72 -mfpu=neon-fp-armv8 -mfloat-abi=hard -mthumb -g
-- Did not find liborc and orcc, disabling orc support...
-- Loading version 2.4 into constants...
-- Using install prefix: /usr/local
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ubuntu/volk/build
```

 but make fails with the following error:

```
In file included from /home/ubuntu/volk/cpu_features/src/cpuinfo_aarch64.c:15:
/home/ubuntu/volk/cpu_features/include/cpuinfo_aarch64.h:153:2: error: #error "Including cpuinfo_aarch64.h from a non-aarch64 target."
  153 | #error "Including cpuinfo_aarch64.h from a non-aarch64 target."
      |  ^~~~~
make[2]: *** [cpu_features/CMakeFiles/cpu_features.dir/build.make:63: cpu_features/CMakeFiles/cpu_features.dir/src/cpuinfo_aarch64.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:227: cpu_features/CMakeFiles/cpu_features.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

uname -a, though, shows: ```Linux ubuntu 5.4.0-1015-raspi #15-Ubuntu SMP Fri Jul 10 05:34:24 UTC 2020 aarch64 aarch64 aarch64 GNU/Linux```

I'm not exactly sure what could be going on here.

balister, 2021-02-02:
I have a bad feeling that toolchain file was tested for a 32 bit system. https://lists.linaro.org/pipermail/linaro-toolchain/2016-July/005815.html

jdemel, 2021-02-06:
Is there anything we can do in VOLK to fix it? Provide separate toolchain files?

hjf, 2021-02-06:
> Is there anything we can do in VOLK to fix it? Provide separate toolchain files?

Well, according to the posted link:

> Meanwhile, the aarch64 spec requires FP and ASIMD instruction support
in the linux ABI, so there are no options to enable them, they are on
by default.

So there is not really a need for the flags. The build instructions should be updated to reflect that, I suppose. Officially Raspberry Pi OS is 32 bit only, still. They were dragging their feet but they were forced to go 64-bit once they released the 8GB Raspberry Pi 4. So there is a 64-bit Raspberry Pi OS but it's "beta". Ubuntu for Pi is 64 or 32-bit. 

So basically the instructions should be: if using a 32-bit OS, use this toolchain. Otherwise, the default toolchain will provide optimized binaries.

jdemel, 2021-02-13:
We should clearly separate our toolchain files into 64bit and 32bit. I assume that would solve the issue.

jdemel, 2021-04-30:
I just discovered this CI error:
https://travis-ci.com/github/gnuradio/volk/jobs/502050291
I assume, we need to fix this one too for RPi.

jdemel, 2024-01-12:
I wonder if this is still an issue. In this case, time is on our side and users potentially migrated to a 64bit Pi OS.