gnuradio/volk #448: "volk_32fc_x2_conjugate_dot_prod_32fc()" returns wrong results

ludwisc, 2021-02-15:
When I insert:
test[0] = 3+2i;
test[1] = 2+3i;
in volk_32fc_x2_conjugate_dot_prod_32fc(&d_power, &test[0], &test[0],2);
the output I get is 12-11i. As I understand the doc, the calculation should be: (3+2i)(3-2i)+(2+3i)(2-3i). This would get a result of 26+0i.

When I calculate
volk_32fc_x2_conjugate_dot_prod_32fc(&d_power, &test[0], &test[1],1);
I should get (3+2i)(2-3i) = 12-5i, but I get 10-2i.

test and d_power are of type gr_complex.
Is this a bug or do I understand something wrong?

jdemel, 2021-02-15:
I agree, I'd expect `26+0j` too. First off, GNU Radio 3.7 is ancient. I recommend to upgrade.

Besides, GR 3.7 uses VOLK as a submodule. It might be the case that you use a very old VOLK version that has a bug. How did you install VOLK (I assume with GNU Radio)? Which VOLK version do you use? You might be able to fix things if you check out the latest VOLK release. Things might be more difficult to update in case you used apt to install GNU Radio.

You `volk_config.txt` indicates that you use AVX kernels. Though, your 2 element example will never call any SIMD intrinsics but go straight to the tail handling part. Thus, I'd consider the generic part to be relevant.

https://github.com/gnuradio/volk/blob/cf446931fde39fdc01b7882b7d8799371f37d2d4/kernels/volk/volk_32fc_x2_conjugate_dot_prod_32fc.h#L155-L161

ludwisc, 2021-02-15:
Thank you for your answer.
I know GNU Radio 3.7 is old, but I didn't want to update all python files to python 3 and all the other changes.
Because I have to do some test in the next weeks I don't want to update anything or install some new versions. I don't want to take the risk that something stops working after the update.
In the part where the volk_32fc_x2_conjugate_dot_prod_32fc() was used, I use only one complex number. To fix the problem I use now the std::norm to calculate the norm of the complex number. It is not a nice solution because when the filter is configured to have more than one tab, the calculation won't be correct, but it works for me now.

I think I installed VOLK with GNU Radio 3.7. If I remember correctly I did a build on my one system and installed it.

jdemel, 2021-02-15:
The easiest way to obtain the installed VOLK version would probably be `volk-config-info --version`. In order to analyse your issue we'd need more info. Your last answer seems to imply that you're happy with your work around though.

ludwisc, 2021-02-15:
I have VOLK version 1.4 and GNU Radio version 3.7.13.5 installed.
As for now I'm happy with my workaround.
I thought if this is a bug I should report it, even that I found another solution for me.

jdemel, 2021-02-15:
Thanks for reporting this issue. It'd be great if you could test the latest VOLK release and report if your bug still exists.

ludwisc, 2021-02-15:
I will test it in a few weeks.
As I mentioned, I don't want to mess up my test equipment with new installations.

rear1019, 2021-02-19:
This is a bug fixed in commit cf5d05e96297. The fix is _not_ contained in the `v1.4` branch.

The numbers work out for the seconds example. The first example has undefined behaviour because the real part of `test[2]` is accessed.

jdemel, 2021-02-21:
@rear1019 Thanks for clarifying the issue. I assume we can close it then.

The issue was fixed in new VOLK versions years ago.

closed 2021-02-23 by ludwisc