gnuradio/volk #459: release 2.4.1 BAD gz checksum and signature

farlies, 2021-03-14:
I observed the following problem verifying the 2.4.1 release file **volk-2.4.1.tar.gz**  (the **.xz** file seems fine). 

```
sh@medusa:volk$ sha256sum -c 2.4.1.sha256.sig
volk-2.4.1.tar.gz: FAILED
volk-2.4.1.tar.xz: OK

sh@medusa:volk$ gpg --verify volk-2.4.1.tar.gz.asc
gpg: assuming signed data in 'volk-2.4.1.tar.gz'
gpg: Signature made Thu 17 Dec 2020 09:49:17 AM CST
gpg:                using RSA key 60FD9F5FDCDC9C8CE87BA5150579D69772CD9B22
gpg: BAD signature from "Johannes Demel <jdemel@gnuradio.org>" [full]

sh@medusa:volk$ gpg --verify volk-2.4.1.tar.xz.asc
gpg: assuming signed data in 'volk-2.4.1.tar.xz'
gpg: Signature made Thu 17 Dec 2020 09:49:29 AM CST
gpg:                using RSA key 60FD9F5FDCDC9C8CE87BA5150579D69772CD9B22
gpg: Good signature from "Johannes Demel <jdemel@gnuradio.org>" [full]
Primary key fingerprint: 60FD 9F5F DCDC 9C8C E87B  A515 0579 D697 72CD 9B22
```
Repeated downloads of .gz and signature and saw the same results...

willcode, 2021-03-14:
Looks OK here. Of course, the public key _is_ in the same place as the files ...
```
$ gpg --import gpg_volk_release_key.asc

$ sha256sum -c 2.4.1.sha256.sig 
volk-2.4.1.tar.gz: OK
volk-2.4.1.tar.xz: OK
sha256sum: WARNING: 2 lines are improperly formatted

$ gpg --verify volk-2.4.1.tar.gz.asc
gpg: assuming signed data in 'volk-2.4.1.tar.gz'
gpg: Signature made Thu 17 Dec 2020 10:49:17 AM EST
gpg:                using RSA key 60FD9F5FDCDC9C8CE87BA5150579D69772CD9B22
gpg: Good signature from "Johannes Demel <jdemel@gnuradio.org>" [unknown]
gpg: WARNING: This key is not certified with a trusted signature!
gpg:          There is no indication that the signature belongs to the owner.
Primary key fingerprint: 60FD 9F5F DCDC 9C8C E87B  A515 0579 D697 72CD 9B22

$ gpg --verify volk-2.4.1.tar.xz.asc
gpg: assuming signed data in 'volk-2.4.1.tar.xz'
gpg: Signature made Thu 17 Dec 2020 10:49:29 AM EST
gpg:                using RSA key 60FD9F5FDCDC9C8CE87BA5150579D69772CD9B22
gpg: Good signature from "Johannes Demel <jdemel@gnuradio.org>" [unknown]
gpg: WARNING: This key is not certified with a trusted signature!
gpg:          There is no indication that the signature belongs to the owner.
Primary key fingerprint: 60FD 9F5F DCDC 9C8C E87B  A515 0579 D697 72CD 9B22
```

farlies, 2021-03-14:
That was a useful data point, @willcode . On attempt-4 (an hour later) I was able to get a copy of the .gz that passed verify.  Not clear why, or why everything else seemed intact, but if integrity looks good to others I will assume the issue is somewhere close to my end of the wire, and close the issue.  The SHA256 sum did its job: the defective .gz untarred without any warnings, but was missing certain files. Yes, it would be nice to have the signing key on keys.openpgp.org.

closed 2021-03-14 by farlies