gnuradio/volk #460: volk_32fc_x2_conjugate_dot_prod_32fc

Ka-zam, 2021-03-15:
This seems to not give the expected result:

```
int
main(int argc, char *argv[]){

  int a = volk_get_alignment();

  cxf_t *c1;
  c1 = (cxf_t*) volk_malloc(5*sizeof(cxf_t), a);
  cxf_t *c2;
  c2 = (cxf_t*) volk_malloc(5*sizeof(cxf_t), a);

  c1[0] =  2.f + 1j*2.f;
  c1[1] =  1.f + 1j*2.f;
  c1[2] =  0.f + 1j*2.f;
  c1[3] = -1.f + 1j*2.f;
  c1[4] = -2.f + 1j*2.f;

  c2[0] = -2.f + 1j*2.f;
  c2[1] = -1.f + 1j*1.f;
  c2[2] =  0.f + 1j*0.f;
  c2[3] =  1.f - 1j*1.f;
  c2[4] =  2.f - 1j*2.f;

  cxf_t res;
  volk_32fc_x2_conjugate_dot_prod_32fc_manual( &res, c1, c2, 5, "generic");
  printf("generic: %8.3f %8.3f\n", std::real(res), std::imag(res));
  //expect res = -10. - 1j*10.

  volk_32fc_x2_conjugate_dot_prod_32fc_manual( &res, c1, c2, 5, "a_generic");
  printf("a_generic: %8.3f %8.3f\n", std::real(res), std::imag(res));

  volk_32fc_x2_conjugate_dot_prod_32fc_manual( &res, c1, c2, 5, "a_avx");
  printf("a_avx: %8.3f %8.3f\n", std::real(res), std::imag(res));

  volk_32fc_x2_conjugate_dot_prod_32fc_manual( &res, c1, c2, 5, "u_avx");
  printf("u_avx: %8.3f %8.3f\n", std::real(res), std::imag(res));

  volk_32fc_x2_conjugate_dot_prod_32fc_manual( &res, c1, c2, 5, "a_sse");
  printf("a_sse: %8.3f %8.3f\n", std::real(res), std::imag(res));

  volk_32fc_x2_conjugate_dot_prod_32fc_manual( &res, c1, c2, 5, "u_sse3");
  printf("u_sse3: %8.3f %8.3f\n", std::real(res), std::imag(res));

  return 0;
}
```
Actual output:

```
generic:  -10.000  -10.000
a_generic:  -10.000  -10.000
a_avx:    2.000   -6.000
u_avx:    2.000   -6.000
a_sse:  -10.000  -10.000
u_sse3:  -10.000  -10.000
```

rear1019, 2021-03-15:
Which Volk version is used? I can not reproduce this with current master and strongly assume this is a duplicate of #448.

PS: In order to compile the provided code add the following header
```c
#include <stdio.h>
#include "volk/volk_complex.h"
#include "volk/volk_malloc.h"
#include "volk/volk.h"

constexpr std::complex<float> operator""j(unsigned long long d)
{
    return std::complex<float>{0.0f, static_cast<float>(d)};
}
constexpr std::complex<float> operator""j(long double d)
{
    return std::complex<float>{0.0f, static_cast<float>(d)};
}
```
and replace all instances of `cxf_t` by `lv_32fc_t`.

Ka-zam, 2021-03-17:
Version was 1.4 so this is #448 rearing it's head again!

closed 2021-03-17 by Ka-zam