gnuradio/volk #461: Rename default branch to `main`

jdemel, 2021-03-20:
You might have seen or heard about projects renaming their default branch to `main`. We want to be an inviting and inclusive community and thus, it would be appropriate to rename our default branch.

The [GitHub docs](https://github.com/github/renaming) describe how to rename the default branch. Further, it seems like there are already quite a few measures in place that help us avoid issues.

michaelld, 2021-03-21:
I am in favor of this renaming. It is "the right thing to do" as you note in the spirit of being an inviting and inclusive community. Moving to using "main" is an appropriate way to honor our own code of conduct.

In my reading, the origin of the term "master" in this context is not -clearly- linked to "master/slave"; other will disagree and that is their prerogative. Most words have multiple meanings, and the interpretation and feelings behind any word or phrase are individualistic in nature. Individuals are how projects succeed or fail, and we want to embrace everyone to as high a degree as reasonably possible in their complexities, interpretations, and feelings. Based on this interpretation of the situation, renaming the primary branch we use from "master" to "main" makes complete sense.

GIT and GitHub provide a means (1) move "master" to "main", and then (2) to "symlink" back from "main" to "master" to allow all current uses of those branch names to continue while we note the deprecation and eventual removal of the latter. I would approve of this method of renaming.

marcusmueller, 2021-06-05:
So, to quickly write down what just happened: 

* github's "rename master to main" *deleted* the `master` reference, and introduced a `main` reference to the current HEAD of said branch. The rest went beautifully (retargeted PRs etc)
* this broke tools/people trying to merge/pull/do things with the master branch
  * especially, pyBOMBS was affected
* @jdemel reintroduced a "master" reference by pushing the branch from his machine. This fixes it.

So, further methodology is to abandon `master`, but leave it there – there's really not much we can do, since enough external tools rely on its existence; we can't "half-have" it, but we can make clear (and that's the case) that `main` is where everything happens, and `master` is just there for legacy reasons. Maybe we'd want to replace `README.md` with "oh, you're using an unsupported branch", or so.

marcusmueller, 2023-11-06:
As [discussed in chat](https://matrix.to/#/!vEQwGiWsSqFVAdVgrn:gnuradio.org/$MWm86ybN1IBjW4rPlwvvTMm1yxCd_dwWyXy43glDbBo?via=gnuradio.org&via=t2bot.io&via=demel.tech): this was left open as a reminder, but with main being well-established now, can be closed.

closed 2023-11-06 by marcusmueller