gnuradio/volk #472: Unresolved external symbol for *_orc_impl when compiling against ORC with MSVC (Windows)

ryanvolz, 2021-05-07:
I just got an ORC package for a bunch of platforms over on conda-forge, and in trying to build against ORC with MSVC on Windows, I get a bunch of "unresolved external symbol" linker errors like:
```
2021-05-07T19:48:36.0242160Z LINK: command "C:\PROGRA~2\MICROS~1\2017\ENTERP~1\VC\Tools\MSVC\1416~1.270\bin\Hostx64\x64\link.exe /nologo lib\CMakeFiles\volk_obj.dir\volk_16ic_deinterleave_16i_x2_a_orc_impl.c.obj lib\CMakeFiles\volk_obj.dir\volk_16ic_deinterleave_real_8i_a_orc_impl.c.obj lib\CMakeFiles\volk_obj.dir\volk_16ic_magnitude_16i_a_orc_impl.c.obj lib\CMakeFiles\volk_obj.dir\volk_16ic_s32f_deinterleave_32f_x2_a_orc_impl.c.obj lib\CMakeFiles\volk_obj.dir\volk_16sc_magnitude_32f_aligned16_orc_impl.c.obj lib\CMakeFiles\volk_obj.dir\volk_16u_byteswap_a_orc_impl.c.obj lib\CMakeFiles\volk_obj.dir\volk_32f_s32f_add_32f_a_orc_impl.c.obj lib\CMakeFiles\volk_obj.dir\volk_32f_s32f_multiply_32f_a_orc_impl.c.obj lib\CMakeFiles\volk_obj.dir\volk_32f_s32f_normalize_a_orc_impl.c.obj lib\CMakeFiles\volk_obj.dir\volk_32f_sqrt_32f_a_orc_impl.c.obj lib\CMakeFiles\volk_obj.dir\volk_32f_x2_add_32f_a_orc_impl.c.obj lib\CMakeFiles\volk_obj.dir\volk_32f_x2_divide_32f_a_orc_impl.c.obj lib\CMakeFiles\volk_obj.dir\volk_32f_x2_dot_prod_32f_a_orc_impl.c.obj lib\CMakeFiles\volk_obj.dir\volk_32f_x2_max_32f_a_orc_impl.c.obj lib\CMakeFiles\volk_obj.dir\volk_32f_x2_min_32f_a_orc_impl.c.obj lib\CMakeFiles\volk_obj.dir\volk_32f_x2_multiply_32f_a_orc_impl.c.obj lib\CMakeFiles\volk_obj.dir\volk_32f_x2_subtract_32f_a_orc_impl.c.obj lib\CMakeFiles\volk_obj.dir\volk_32fc_32f_multiply_32fc_a_orc_impl.c.obj lib\CMakeFiles\volk_obj.dir\volk_32fc_magnitude_32f_a_orc_impl.c.obj lib\CMakeFiles\volk_obj.dir\volk_32fc_s32fc_multiply_32fc_a_orc_impl.c.obj lib\CMakeFiles\volk_obj.dir\volk_32fc_x2_multiply_32fc_a_orc_impl.c.obj lib\CMakeFiles\volk_obj.dir\volk_32i_x2_and_32i_a_orc_impl.c.obj lib\CMakeFiles\volk_obj.dir\volk_32i_x2_or_32i_a_orc_impl.c.obj lib\CMakeFiles\volk_obj.dir\volk_8i_convert_16i_a_orc_impl.c.obj lib\CMakeFiles\volk_obj.dir\volk_8i_s32f_convert_32f_a_orc_impl.c.obj lib\CMakeFiles\volk_obj.dir\constants.c.obj lib\CMakeFiles\volk_obj.dir\volk_prefs.c.obj lib\CMakeFiles\volk_obj.dir\volk_rank_archs.c.obj lib\CMakeFiles\volk_obj.dir\volk_malloc.c.obj lib\CMakeFiles\volk_obj.dir\__\include\volk\volk.h.obj lib\CMakeFiles\volk_obj.dir\volk.c.obj lib\CMakeFiles\volk_obj.dir\__\include\volk\volk_typedefs.h.obj lib\CMakeFiles\volk_obj.dir\__\include\volk\volk_cpu.h.obj lib\CMakeFiles\volk_obj.dir\volk_cpu.c.obj lib\CMakeFiles\volk_obj.dir\__\include\volk\volk_config_fixed.h.obj lib\CMakeFiles\volk_obj.dir\volk_machines.h.obj lib\CMakeFiles\volk_obj.dir\volk_machines.c.obj lib\CMakeFiles\volk_obj.dir\volk_machine_generic_orc.c.obj lib\CMakeFiles\volk_obj.dir\volk_machine_sse2_orc.c.obj lib\CMakeFiles\volk_obj.dir\volk_machine_sse3_orc.c.obj lib\CMakeFiles\volk_obj.dir\volk_machine_ssse3_orc.c.obj lib\CMakeFiles\volk_obj.dir\volk_machine_sse4_1_orc.c.obj lib\CMakeFiles\volk_obj.dir\volk_machine_sse4_2_orc.c.obj lib\CMakeFiles\volk_obj.dir\volk_machine_avx_orc.c.obj lib\CMakeFiles\volk_obj.dir\volk_machine_avx2_orc.c.obj lib\CMakeFiles\volk_obj.dir\volk_machine_avx512f_orc.c.obj /out:lib\volk.dll /implib:lib\volk.lib /pdb:lib\volk.pdb /dll /version:0.0 /machine:x64 /INCREMENTAL:NO -LIBPATH:%SRC_DIR%\lib\ORC_LIBRARY_DIR-NOTFOUND cpu_features\cpu_features.lib %PREFIX%\Library\lib\orc-0.4.lib kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib /MANIFEST /MANIFESTFILE:lib\volk.dll.manifest" failed (exit code 1120) with the following output:
2021-05-07T19:48:36.0401422Z volk_machine_avx512f_orc.c.obj : error LNK2001: unresolved external symbol "void __cdecl volk_16ic_deinterleave_16i_x2_a_orc_impl(short *,short *,class std::complex<short> const *,unsigned int)" (?volk_16ic_deinterleave_16i_x2_a_orc_impl@@YAXPEAF0PEBV?$complex@F@std@@I@Z
...
```
You can find the full log [here](https://dev.azure.com/conda-forge/84710dde-1620-425b-80d0-4cf5baca359d/_apis/build/builds/314593/logs/31).

I don't know why it can't resolve these symbols, because they should be in the compiled `*_orc_impl.c.obj` files that are included in the linking command. I'm hoping someone with more knowledge in this area can shed some light.

I've gone back to no ORC on Windows for the conda package for now, but getting this to work would be great.

jdemel, 2021-05-08:
It would be great to get ORC kernels compiled on Windows. I think we avoided that on Windows for some reason. This might require some deeper MSVC knowledge.

```
2021-05-07T19:47:27.6809030Z [[ RA-MD1LOVE ]] - [[         0001-Add-main-to-volk_modtool-script-usable-as-an-entry-p.patch ]]
2021-05-07T19:47:27.6810422Z [[ RA-MD1-OVE ]] - [[         0002-Fix-CMAKE_SYSTEM_PROCESSOR-arm64-should-AARCH64-not-.patch ]]
2021-05-07T19:47:27.6811762Z [[ RA-MD1LOVE ]] - [[                   0003-Add-support-for-Apple-M1-AArch64-processor.patch ]]
```
I'm curious about these patches.

ryanvolz, 2021-05-08:
Ah, the first patch is just a way to get `volk_modtool` to run as a command on Windows, where it refuses to recognize a Python script as something it can execute without some special setup. This patch lets me makes use of the "entry point" mechanism of setuptools that is exposed as part of the conda packaging, so it's probably not useful here upstream unless further changes were to be made to use a setuptools install.

The other two patches were adapted from the `cpu_features` PR linked in #428, and they succeed in getting volk to compile for osx-arm64 although I personally can't test the functionality. But the volk package on conda-forge should now install on the Apple M1!