gnuradio/volk #473: volk_8u_x4_conv_k7_r2_8u does not work on ARM platforms

Aang23, 2021-05-08:
Hi!
I recently ported one of my projects over to Android, building VOLK via the NDK (I had to patch a few things to make it build, I will open another issue on the topic later). About everything works as expected.

The only thing that did not work was volk_8u_x4_conv_k7_r2_8u (generic kernel). I initially thought this was an issue on my side with my unusual environment, so I pulled out a Raspberry Pi and was able to reproduce the issue reliably using volk_8u_x4_conv_k7_r2_8u as a streaming decoder (similarly to GNU Radio's CC Decoder). No issue is observed with another implementation such as libcorrect's.

The behavior observed is volk_8u_x4_conv_k7_r2_8u seemingly working normally, but outputting 0s instead.

odrisci, 2021-05-09:
I must confess that I know nothing about this code, but [this line](https://github.com/gnuradio/volk/blob/79a111ba600c86ae937830970b00ebe0c711df5a/kernels/volk/volk_8u_x4_conv_k7_r2_8u.h#L118) and the one following it look suspicious to me. It seems to me that the difference will always be non negative, so the decisions will always be 1 except when the differences are precisely 0.

Aang23, 2021-05-10:
> I must confess that I know nothing about this code, but [this line](https://github.com/gnuradio/volk/blob/79a111ba600c86ae937830970b00ebe0c711df5a/kernels/volk/volk_8u_x4_conv_k7_r2_8u.h#L118) and [...]

My understanding of this isn't the best either, but I do not see anything obviously wrong with this implementation compared to others that definitely do work. I'm honestly very surprised the generic kernel does not work, even more that it works perfectly fine on x86 and not arm. 

This kinda kills half of my usecase as I am using Volk's implementation the most. Guess I will have to debug on my end in the meantime.

Aang23, 2021-05-10:
Actually, I was partially wrong... It does not 0s but garbage data. A lower BER gets it down to more 0s. Weird behavior...
Here it is visualized in a bitstream viewer.
[image]

jdemel, 2021-05-11:
I'd suggest to test this kernel with a high SNR, or only very (or no) bit errors.
Also how exactly do you test this?

Aang23, 2021-05-12:
I tested it in several different ways : 
- Feeding the r=1/2, k=7, G1=-79, G2=-109 encoded CCSDS sync word (0xfca2b63db00d9794), converted to 0 to 255 uint8_t values. This was successful. After this, I modified a few bits (using 0xfca2b63db11d9794), which should be successfully decoded to 0x1acffc1d . Works fine on x86, but on my 3 ARM machines I tested on, results were very inconsistent with the same decode giving different results on each run (all variables in my test program were initialized to 0 or some other value).
- Directly iin one my actual programs (using METEOR-M 2 LRPT Data and JPSS-1 HRD data), and comparing each processing step on several ARM machines and x86. This allowed me to find a few mistakes on my side, to now end with an identical input to the viterbi decoder, but different outputs. (correction on lower SNRs being about none)
- Generating known good test data (generated from 10MB of /dev/random), fed into the decoder on once again several machines. This did work fine with minor differences in the output.

In all cases, VOLK was compiled with GCC 10.2.0 in debug mode (-O0), as well as the test program. Partab, metrics, etc were generated using GNU Radio's CC Decoder's code (which is as far as I know the only sample implementation I could find using VOLK) and the generic kernel.

In conclusion, in all cases, volk_8u_x4_conv_k7_r2_8u produces inferior results on all ARM platforms I could test. (Raspberry Pi 2, my Oneplus 6T (running a linux container) and a RK3318 board I had laying around)

Aang23, 2021-05-12:
Actually, I just realized I was using the SSE kernel on my x86 machine... My bad. Then, this means the generic kernel is in fact the issue here. Identical issues as seen on ARM are observed.

jdemel, 2021-05-18:
Thanks for debugging this issue further.
This particular kernel has several issues. It fails for AVX2, it produces inconsistent output on different platforms, and it is very hard to read and understand the code, thus it is terribly difficult to maintain it.
I'd really like to see this issue fixed though.

Aang23, 2021-05-19:
You're welcome!
Indeed. I initially thought only the AVX2 kernel had issues, but that turned out to be very much wrong once I was on a platform with only Generic available.

Definitely not a kernel I'm really able to help debugging more than that, I do not have in-depth knowledge of that algorithm so understanding and debugging this won't be very easy... Will give it a go whenever I can, but I'm not hoping for much if any success.

I would guess @trondeau would be the one to ask about this issue, as he wrote those kernels looking through the commit history.

michaelld, 2021-05-20:
@Aang23 do you have some test code that you can pass on that we can use to do debugging? Asking since I'm guessing you do have such code & that will be a lot faster than us trying to create our own! Thx!

Aang23, 2021-07-21:
> @Aang23 do you have some test code that you can pass on that we can use to do debugging? Asking since I'm guessing you do have such code & that will be a lot faster than us trying to create our own! Thx!

Oh sorry, I had entirely missed your reply. I will cleanup my test code and post it a bit later.

After discussing with @jdemel on Matrix, it seems like I wasn't clear enough about the exact issue.
The current generic kernel performs just fine given error-free data, but has very poor error-correction performances, hence why it probably did not show up in QA earlier.
Feeding the kernel some slightly corrupted data (like every real-world application would likely have) will result in no FEC gain at all, hence why I initially noticed the issue in my usecase.

https://github.com/gnuradio/volk/issues/473#issuecomment-839957073

Here with those images my intent was to show how the generic kernel only decoded valid data at peak SNR, when other FEC layers could take over, while in contract the spiral kernel performed much better.

noc0lour, 2021-12-21:
Running some CC Enc & Dec with error-free data in GNU Radio results in garbage output, so not sure if it even works at all. It worked with GNU Radio 3.7 & corresponding VOLK, but doesn't anymore..

PR #736 (2024-01-07): Fix bugs in convolutional decoder
Fixes #733.
Fixes #473.
Reverts #475.

VOLK's convolutional decoder has many bugs:

1. The test puppet has the input & output reversed, so it always tests the kernel with zeroes as input. Once this is corrected, many more bugs become apparent.
2. The `spiral` and `neonspiral` protokernels only renormalize the branch metrics if the first branch metric is greater than 210. This threshold is much too high, and results in constant overflows and incorrect decisions. The threshold was previously removed from the generic protokernel, so I've done the same for the oth [... cut]
closed 2022-02-06 by michaelld