gnuradio/volk #478: Segfault on FreeBSD

onno182, 2021-06-06:
I complied GNU Radio on FreeBSD succesfully, but get a segfault when running a simple .grc with a multiply block. 

The .grc works fine on Pentoo.

`Traceback (most recent call last):
  File "lesson1.py", line 165, in <module>
    main()
  File "lesson1.py", line 143, in main
    tb = top_block_cls()
  File "lesson1.py", line 112, in __init__
    self.connect((self.analog_sig_source_x_0_0, 0), (self.blocks_multiply_xx_0, 1))
  File "/usr/local/gnuradio/lib/python3.7/site-packages/gnuradio/gr/hier_block2.py", line 37, in wrapped
    func(self, src, src_port, dst, dst_port)
  File "/usr/local/gnuradio/lib/python3.7/site-packages/gnuradio/gr/hier_block2.py", line 100, in connect
    self.primitive_connect(*args)
ValueError: port number 1 exceeds max of 0
Segmentation fault (core dumped)
`

GDB output:

'Thread 1 received signal SIGSEGV, Segmentation fault.
0x000000080c7be663 in gr::block_registry::unregister_primitive(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >) ()
   from /usr/local/gnuradio/lib/libgnuradio-runtime.so.3.10.0git'

jdemel, 2021-06-07:
Is this issue still about the SegFault?
If so, is there a way to reproduce this issue with just VOLK without GNU Radio? Or is this not an issue at all

> [..] The error remains with cpu_features disabled. On the other hand with it activated, volk_profile did run successfully.

Or is this issue about CPU feature detection now? The topic seems to change.
In that case, does cpu_features need to be patched, or is it sufficient to rely on Linuxulator?

onno182, 2021-06-07:
@jdemel There is still a segfault in GNU Radio, also when disabling cpu_features.

I do not believe the segfault is in Volk, which compiles fine and also creates the profiles without issues. However, in order to get Volk to compile on FreeBSD some changes are necessary (see above). marcusmueller asked me to state which changes exactly were necessary to forward to Volk team. 

The issue of segfault remains. I have no idea what is the cause.  GNU Radio seems to work with some sketches but others not.  The fault seems to occur when dealing with a block with multiple inputs, for instance, when adding a multiply block which receives two inputs and outputs one.

marcusmueller, 2021-06-07:
Hi, to clarify: I've migrated this to VOLK because the code you've included makes VOLK work, and I needed @jdemel to be aware of it, and hope it's something that we can either find a workaround to in VOLK or find a way to upstream to cpu_features. But I don't know which is desirable here!

All in all,  might have been a bit fast with the migration button. If you think this is wrong here, Johannes can migrate back to GNU Radio.

vishwin, 2021-06-07:
I'll have a play once I have my copy of GNU Radio 3.9.1.0 to dogfood before sending out the call for testing in our ports tree. I will be testing with Python 3.9 compared to the OP's Python 3.7.

> In that case, does cpu_features need to be patched, or is it sufficient to rely on Linuxulator?

Don't even think about using the Linuxulator, unless we're dealing with an actual Linux binary.

onno182, 2021-06-07:
@vishwin just as a side note: My reason for compiling GNU Radio from source is because the binary package in ports seems to be broken as it uninstalls Gnome3. Also my version of GNU Radio is running with Python 3.7 as it is not possible to satisfy dependencies with binary packages from Python 3.9.

vishwin, 2021-06-07:
Yes, the dependency mess is a big reason why I'm updating the port. (I'm a committer)

PR #550 (2021-12-23): cpu_features: Update submodule pointer
We have 3 issues that should be fixed with this commit.

Fix #428
Should be fixed because cpu_features detects `arm64` now. Thus, VOLK builds with cpu_features on MacOS and reports M1 capabilities.

Fix #478
Fix #484
cpu_features received quite a bit of contributions for FreeBSD. All the issues we had should be fixed now. However, this might require further evaluation.
commit c8198263 (2021-12-23): cpu_features: Update submodule pointer

We have 3 issues that should be fixed with this commit.

Fix #428
Should be fixed because cpu_features detects `arm64` now. Thus, it
builds on MacOS and reports M1 capabilities.

Fix #478
Fix #484
cpu_features received quite a bit of contributions for FreeBSD. All the
issues we had should be fixed now. However, this might require further
evaluation.
closed 2021-12-23 by michaelld