gnuradio/volk #479: VREV for byte swap on Arm Neon?

Triang3l, 2021-06-07:
The Arm Neon versions of byte swaps (volk_*_byteswap.h) in VOLK use shifts/OR or lookup tables, somewhat similar to the x86 versions. However, Neon has a dedicated instruction for byte swaps — [VREV](https://developer.arm.com/documentation/den0018/a/NEON-and-VFP-Instruction-Summary/NEON-general-data-processing-instructions/VREV), usable as `vrev16q_u8` for 8-in-16, `vrev32q_u8` for 8-in-32, and `vrev64q_u8` for 8-in-64. Are there performance/compatibility reasons for not using it, or is that more of not knowing about the instruction when the code was written?

jdemel, 2021-06-07:
Could you point to the exact code that you have in mind?

Are you refering to this?

https://github.com/gnuradio/volk/blob/237a6fc9242ea8c48d2bbd417a6ea14feaf7314a/kernels/volk/volk_16u_byteswap.h#L223-L239

This implementation is essentially 7 years old. Are you interested in contributing an optimized version of this code?

Triang3l, 2021-06-07:
Yes, and the vtbl4/vqtbl1q implementations for 32-bit and 64-bit swaps. I can try setting up the environment on my phone and write direct vrev versions, and possibly run some speed comparisons, as well as tests, in the weekend.