gnuradio/volk #481: Add  volk_32f(c)_index_min_16u/32u

Zlika, 2021-06-09:
Hi all,
It would be nice to have volk_32f(c)_index_min_16u/32u functions in addition to the existing volk_32f(c)_index_max_16u/32u functions. Can I try to propose a PR to add these functions?
Regards,
Thomas

jdemel, 2021-06-09:
Yes, of course! Contributions are always welcome. An `index_min` kernel in addition to `index_max` kernels seems very reasonable as well.

Zlika, 2021-06-09:
Thanks @jdemel . I have a working version, based on the index_max kernels. I was able to test all the intel archs (avx, avx2, sse, sse3, sse4_1). As some of the functions have neon variants I wanted to test them on a raspberry pi 3 running the latest Raspberry pi OS lite (Debian 10.9, kernel 5.10.17-v7+), but the tests only run the generic versions.
cmake output is:
```
(...)
-- Performing Test have_mfpu_neon - Success
(...)
- Performing Test neon_compile_result - Failed
- Compiler doesn't support NEON, Overrules arch neon
```
Do you know how can I fix that to test the neon kernels on my raspberry pi 3?
Thanks

jdemel, 2021-06-10:
It is very hard to tell what actually happened here.
> - Performing Test neon_compile_result - Failed
This line indicates that a simple CMake compile test to detect NEON failed and thus, CMake drops NEON support. How exactly did you try to build VOLK?

Zlika, 2021-06-10:
Ok. Pull request created: https://github.com/gnuradio/volk/pull/482
I see that you have a ARM builder in your gitlab workflow, so I suppose we will be able to check the correctness of the NEON kernels with it.

Yash-dev1, 2021-07-26:
Sir, while running the file I am getting the issue of 'volk_32fc_index_max_32u_manual' was not declared in the scope. 
[image]
How to resolve this?

michaelld, 2021-07-26:
> Sir, while running the file I am getting the issue of 'volk_32fc_index_max_32u_manual' was not declared in the scope.
> How to resolve this?

@Yash-dev1 what does `make VERBOSE=ON` return? Look carefully at the header include arguments (from `-I...`) ... it could be that the Volk header directory is not being included ... or something else.

Yash-dev1, 2021-08-01:
@michaelld It's returning the same error (as attached)...Also, I've made sure that the directory is being included..still, I don't know how to deal with it.
[image]

jdemel, 2021-08-01:
@Yash-dev1 please don't use screenshots of text. Your Terminal supports coyping. Shortcuts are: CTRL+SHIFT+C, (and CTRL+SHIFT+V for pasting).

Besides, why do you need the `_manual` version of this kernel? This is certainly not the recommended way to do things.
I couldn't spot the `-I...` line but that might be due to the screenshot.

Yash-dev1, 2021-08-01:
@jdemel Sorry for the inconvenience sir. I am new to this community ad still learning how to deal with the issues. Anyways, I am attaching the result below:

`  -Dgnuradio_packetizer_EXPORTS -I"/home/dragon/check/gnuradioproject-master (2)/gr-packetizer/lib" -I"/home/dragon/check/gnuradioproject-master (2)/gr-packetizer/include" -I"/home/dragon/check/gnuradioproject-master (2)/gr-packetizer/build/lib" -I"/home/dragon/check/gnuradioproject-master (2)/gr-packetizer/build/include"  -O3 -DNDEBUG -fPIC   -fvisibility=hidden -o CMakeFiles/gnuradio-packetizer.dir/corr_est_cc_impl.cc.o -c "/home/dragon/check/gnuradioproject-master (2)/gr-packetizer/lib/corr_est_cc_impl.cc"`

Also, I removed the `_manual` version of this kernel too, but again I am getting the same error(ofcourse, without the `_manual` part) : `error: volk_32fc_index_max_32u’ was not declared in this scope`
Moreover, I am only following the report and the codes provided in this repository, so I dont think I am doing anything other than that, so if there's anything wrong I request your help.

jdemel, 2021-08-02:
It's alright. I just wanted to give you these hints. One more hint: For multiline verbatim, you want three ` at the beginning, newline and three of them at the end.

It's still difficult to tell what your problem might be. Let's start with some basics. What do the following commands return?
- `uname -a`
-  `volk-config-info -v`
- `volk-config-info --prefix`

Yash-dev1, 2021-08-02:
@jdemel The commands returned the following:
1. `uname -a` returned `Linux dragon 4.15.0-142-generic #146~16.04.1-Ubuntu SMP Tue Apr 13 09:27:15 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
`
2. `volk-config-info -v`  returned `1.2.1`

3.  `volk-config-info --prefix`  returned  `/usr`

jdemel, 2021-08-02:
Ah! This reveals a lot.

1. You still use Ubuntu 16.04. At this point, it has reached End-Of-Life. You should really upgrade to 18.04 or better 20.04.
2. VOLK v1.2.1 is ancient. It might be that we added this kernel in a later release. Due to some Ubuntu 16.04 specific bug, it'd be difficult to upgrade VOLK. VOLK v2.0 might work. But v2.5 is our latest release.
3. This indicates a system install via `apt ... ` or similar. If you'd try to install a newer VOLK version manually, you'd have to uninstall this old version.

I'd assume you try to use GNU Radio 3.7 in this case as well. I really recommend to move to a later release. Preferably GNU Radio 3.9. And please don't use a distribution that is EOL. After all, it is a security risk.

Yash-dev1, 2021-08-02:
@jdemel Sir, I'm using Ubuntu 18.04 now and also checked the same things in Ubuntu 20.04 and while running the commands, now I am getting these particular errors shown as:
While running the `cmake ../` command, it returned 
`-- Build type not specified: defaulting to release.`
`CMake Deprecation Warning at CMakeLists.txt:54 (cmake_policy):`
`  The OLD behavior for policy CMP0026 will be removed from a future version
  of CMake.`
`  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.`
`-- Boost version: 1.65.1`
`-- Found the following Boost libraries:`
`--   filesystem`
`--   system`
`CMake Error at CMakeLists.txt:143 (find_package)`
`  Could not find a configuration file for package "Gnuradio" that is
  compatible with requested version "3.7.2".`
`The following configuration files were considered but not accepted:`
`    /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GnuradioConfig.cmake, version: 3.8.2.0`
`-- Configuring incomplete, errors occurred!`
`See also "/home/merlin/chchc/gnuradioproject/gnuradioproject-master/gr-packetizer/build/CMakeFiles/CMakeOutput.log".`

So, I think here the problem now is with cmake version and I'm not getting how to resolve this because in the report it was mentioned that ubuntu 16.04 and gnuradio 3.7.10 was used, so I installed them only. 
And I ran the version commands(in the new PC) too that returned(just for the information purposes) :
1.  `cmake --version` returned `cmake version 3.10.2`
2.  `doxygen --version` returned `1.8.13`

jdemel, 2021-08-03:
Since you're in the process of upgrading your system, I'd really recommend to go with Ubuntu 20.04.

I assume, you installed GNU Radio via `sudo apt install gnuradio` or equivalent? 
On Ubuntu 20.04 that would get you [VOLK v2.2.1](https://packages.ubuntu.com/focal/libvolk2-dev).
Further installation instruction can be found here: https://wiki.gnuradio.org/index.php/UbuntuInstall
I recommend to stick with the binary install.

This is really a GNU Radio version issue.
```
CMake Error at CMakeLists.txt:143 (find_package)
Could not find a configuration file for package "Gnuradio" that is compatible with requested version "3.7.2".
The following configuration files were considered but not accepted:
/usr/lib/x86_64-linux-gnu/cmake/gnuradio/GnuradioConfig.cmake, version: 3.8.2.0
```
Your OOT module is still on GR 3.7 and needs to be converted to 3.8 or better 3.9. It seems like you just started with a new module. In this case, I recommend the [GNU Radio Tutorial](https://wiki.gnuradio.org/index.php/Tutorials). They'll explain a lot and in the end you know how to get started with the latest GNU Radio version.

Out of curiosity, where did you find these old (ancient) instructions?

Yash-dev1, 2021-08-03:
@jdemel Ok, sir, I will follow what you have advised.
Actually, I didn't find any of the instructions mentioned anywhere, not even in the report. So, if you are comfortable, I can send you the report via mail (my email:yasha18@iiserb.ac.in), where I can elaborate more on what I want to do with this project, and you can also comfortably help me out with the issues.

PR #482 (2021-09-22): Add volk_32f(c)_index_min_16/32u
Added functions volk_32f(c)_index_min_16/32u based on the already existing functions volk_32f(c)_index_max_16/32u.
Fixes #481.
closed 2021-09-22 by jdemel