gnuradio/volk #483: Cannot build neon kernels on Raspberry pi

Zlika, 2021-06-11:
Hi,
I tried to compile Volk on a raspberry pi 3 running the latest Raspberry pi OS lite (Debian 10.9, kernel 5.10.17-v7+).
However, the neon kernels are not compiled.
The cmake output is:
```
(...)
-- Performing Test have_mfpu_neon - Success
(...)
-- Performing Test neon_compile_result - Failed
-- Compiler doesn't support NEON, Overrules arch neon
```
The workaround is to add the following line in lib/CMakeLists.txt, just before the line "`check_c_source_compiles("#include <arm_neon.h>\nint main(){ uint8_t *dest; uint8x8_t res; vst1_u8(dest, res); }" neon_compile_result)`":
`set(CMAKE_REQUIRED_FLAGS="-mfpu=neon")`

Then cmake correctly detects and enables neon:
```
-- Performing Test neon_compile_result
-- Performing Test neon_compile_result - Success
-- Performing Test have_neonv7_result
-- Performing Test have_neonv7_result - Success
-- Performing Test have_neonv8_result
-- Performing Test have_neonv8_result - Failed
-- Compiler doesn't support neonv8n Overrules arch neonv8
-- Available architectures: generic;hardfp;neon;neonv7;norc
-- Available machines: generic;neon;neonv7_hardfp
```

But then there are errors during the build phase:
```
(...)
[ 35%] Building ASM object lib/CMakeFiles/volk_obj.dir/__/kernels/volk/asm/neon/volk_16i_max_star_horizontal_16i.s.o
volk_16i_max_star_horizontal_16i.s: Assembler messages:
volk_16i_max_star_horizontal_16i.s:11: Error: selected FPU does not support instruction -- 'vmov.i32 q12,#0'
volk_16i_max_star_horizontal_16i.s:18: Error: selected processor does not support 'vld2.16 {d16-d19},[r4]!' in ARM mode
(...)
```

Zlika, 2021-06-16:
My mistake, I didn't RTFM, the cmake toolchain file must be explicitly set for raspberry pi (`cmake -DCMAKE_TOOLCHAIN_FILE=../cmake/Toolchains/arm_cortex_a53_hardfp_native.cmake ..`)

closed 2021-06-16 by Zlika

jdemel, 2021-06-16:
I'm happy it works for you now!