gnuradio/volk #484: Disabling VOLK_CPU_FEATURES causes tests to fail on 2.5.0

diizzyy, 2021-06-11:
```
Test project /usr/ports/devel/volk/work/.build
        Start   1: qa_volk_16i_32fc_dot_prod_32fc
  1/135 Test   #1: qa_volk_16i_32fc_dot_prod_32fc .......................***Failed    0.03 sec
        Start   2: qa_volk_16i_branch_4_state_8
  2/135 Test   #2: qa_volk_16i_branch_4_state_8 .........................   Passed    0.00 sec
        Start   3: qa_volk_16i_convert_8i
  3/135 Test   #3: qa_volk_16i_convert_8i ...............................***Failed    0.01 sec
        Start   4: qa_volk_16i_max_star_16i
  4/135 Test   #4: qa_volk_16i_max_star_16i .............................   Passed    0.00 sec
        Start   5: qa_volk_16i_max_star_horizontal_16i
  5/135 Test   #5: qa_volk_16i_max_star_horizontal_16i ..................   Passed    0.00 sec
        Start   6: qa_volk_16i_permute_and_scalar_add
  6/135 Test   #6: qa_volk_16i_permute_and_scalar_add ...................   Passed    0.00 sec
        Start   7: qa_volk_16i_s32f_convert_32f
  7/135 Test   #7: qa_volk_16i_s32f_convert_32f .........................***Failed    0.01 sec
        Start   8: qa_volk_16i_x4_quad_max_star_16i
  8/135 Test   #8: qa_volk_16i_x4_quad_max_star_16i .....................   Passed    0.00 sec
        Start   9: qa_volk_16i_x5_add_quad_16i_x4
  9/135 Test   #9: qa_volk_16i_x5_add_quad_16i_x4 .......................   Passed    0.00 sec
        Start  10: qa_volk_16ic_convert_32fc
 10/135 Test  #10: qa_volk_16ic_convert_32fc ............................***Failed    0.02 sec
        Start  11: qa_volk_16ic_deinterleave_16i_x2
 11/135 Test  #11: qa_volk_16ic_deinterleave_16i_x2 .....................***Failed    0.01 sec
        Start  12: qa_volk_16ic_deinterleave_real_16i
 12/135 Test  #12: qa_volk_16ic_deinterleave_real_16i ...................***Failed    0.01 sec
        Start  13: qa_volk_16ic_deinterleave_real_8i
 13/135 Test  #13: qa_volk_16ic_deinterleave_real_8i ....................***Failed    0.01 sec
        Start  14: qa_volk_16ic_magnitude_16i
 14/135 Test  #14: qa_volk_16ic_magnitude_16i ...........................***Failed    0.01 sec
        Start  15: qa_volk_16ic_s32f_deinterleave_32f_x2
 15/135 Test  #15: qa_volk_16ic_s32f_deinterleave_32f_x2 ................***Failed    0.02 sec
        Start  16: qa_volk_16ic_s32f_deinterleave_real_32f
 16/135 Test  #16: qa_volk_16ic_s32f_deinterleave_real_32f ..............***Failed    0.01 sec
        Start  17: qa_volk_16ic_s32f_magnitude_32f
 17/135 Test  #17: qa_volk_16ic_s32f_magnitude_32f ......................***Failed    0.01 sec
        Start  18: qa_volk_16ic_x2_dot_prod_16ic
 18/135 Test  #18: qa_volk_16ic_x2_dot_prod_16ic ........................***Failed    0.02 sec
        Start  19: qa_volk_16ic_x2_multiply_16ic
 19/135 Test  #19: qa_volk_16ic_x2_multiply_16ic ........................***Failed    0.02 sec
        Start  20: qa_volk_16u_byteswap
 20/135 Test  #20: qa_volk_16u_byteswap .................................   Passed    0.00 sec
        Start  21: qa_volk_16u_byteswappuppet_16u
 21/135 Test  #21: qa_volk_16u_byteswappuppet_16u .......................***Failed    0.01 sec
        Start  22: qa_volk_32f_64f_add_64f
 22/135 Test  #22: qa_volk_32f_64f_add_64f ..............................***Failed    0.02 sec
        Start  23: qa_volk_32f_64f_multiply_64f
 23/135 Test  #23: qa_volk_32f_64f_multiply_64f .........................***Failed    0.02 sec
        Start  24: qa_volk_32f_8u_polarbutterfly_32f
 24/135 Test  #24: qa_volk_32f_8u_polarbutterfly_32f ....................   Passed    0.00 sec
        Start  25: qa_volk_32f_8u_polarbutterflypuppet_32f
 25/135 Test  #25: qa_volk_32f_8u_polarbutterflypuppet_32f ..............***Failed    0.01 sec
        Start  26: qa_volk_32f_accumulator_s32f
 26/135 Test  #26: qa_volk_32f_accumulator_s32f .........................   Passed    0.01 sec
        Start  27: qa_volk_32f_acos_32f
 27/135 Test  #27: qa_volk_32f_acos_32f .................................***Failed    0.01 sec
        Start  28: qa_volk_32f_asin_32f
 28/135 Test  #28: qa_volk_32f_asin_32f .................................***Failed    0.01 sec
        Start  29: qa_volk_32f_atan_32f
 29/135 Test  #29: qa_volk_32f_atan_32f .................................***Failed    0.01 sec
        Start  30: qa_volk_32f_binary_slicer_32i
 30/135 Test  #30: qa_volk_32f_binary_slicer_32i ........................***Failed    0.01 sec
        Start  31: qa_volk_32f_binary_slicer_8i
 31/135 Test  #31: qa_volk_32f_binary_slicer_8i .........................***Failed    0.01 sec
        Start  32: qa_volk_32f_convert_64f
 32/135 Test  #32: qa_volk_32f_convert_64f ..............................***Failed    0.02 sec
        Start  33: qa_volk_32f_cos_32f
 33/135 Test  #33: qa_volk_32f_cos_32f ..................................***Failed    0.02 sec
        Start  34: qa_volk_32f_exp_32f
 34/135 Test  #34: qa_volk_32f_exp_32f ..................................   Passed    0.00 sec
[... 293 lines cut ...]
        112 - qa_volk_32u_reverse_32u (Failed)
        113 - qa_volk_64f_convert_32f (Failed)
        114 - qa_volk_64f_x2_add_64f (Failed)
        115 - qa_volk_64f_x2_max_64f (Failed)
        116 - qa_volk_64f_x2_min_64f (Failed)
        117 - qa_volk_64f_x2_multiply_64f (Failed)
        119 - qa_volk_64u_byteswappuppet_64u (Failed)
        121 - qa_volk_64u_popcntpuppet_64u (Failed)
        122 - qa_volk_8i_convert_16i (Failed)
        123 - qa_volk_8i_s32f_convert_32f (Failed)
        124 - qa_volk_8ic_deinterleave_16i_x2 (Failed)
        125 - qa_volk_8ic_deinterleave_real_16i (Failed)
        126 - qa_volk_8ic_deinterleave_real_8i (Failed)
        127 - qa_volk_8ic_s32f_deinterleave_32f_x2 (Failed)
        128 - qa_volk_8ic_s32f_deinterleave_real_32f (Failed)
        129 - qa_volk_8ic_x2_multiply_conjugate_16ic (Failed)
        130 - qa_volk_8ic_x2_s32f_multiply_conjugate_32fc (Failed)
        131 - qa_volk_8u_conv_k7_r2puppet_8u (Failed)
        134 - qa_volk_8u_x3_encodepolarpuppet_8u (Failed)
Errors while running CTest
```

Applying https://github.com/google/cpu_features/pull/159 (patch needs more work) makes tests complete with VOLK_CPU_FEATURES enabled.

Tested on FreeBSD 13 (amd64)

jdemel, 2021-06-14:
Thanks for your report. I assume, we need to wait for this patch to be merged into cpu_features and then update our dependencies. This might also fix #478 .

diizzyy, 2021-06-14:
Thanks, are tests however expected to fail without cpu_features?

jdemel, 2021-06-14:
VOLK did always rely on CPU feature detection. Thus, I'd doubt we ever tested it without cpu_features. Besides, I can't tell why your tests failed specifically. `ctest -V` may reveal more info. If you'd share your CPU model that'd also be helpful to know.

diizzyy, 2021-06-14:
Hi,

=== configure log ===

```
===>  Configuring for volk-2.5.0
===>  Performing out-of-source build
/bin/mkdir -p /usr/ports/devel/volk/work/.build
-- The C compiler identification is Clang 11.0.1
-- The CXX compiler identification is Clang 11.0.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test HAVE_CX_LIMITED_RANGE
-- Performing Test HAVE_CX_LIMITED_RANGE - Failed
-- Performing Test HAVE_C_LIMITED_RANGE
-- Performing Test HAVE_C_LIMITED_RANGE - Failed
-- Build type set to Release.
-- Found Git: /usr/local/bin/git (found version "2.31.1")
-- Building Volk without cpu_features
-- Found PythonInterp: /usr/local/bin/python3.8 (found suitable version "3.8.10", minimum required is "3")
--
-- Python checking for distutils
-- Python checking for distutils - found
--
-- Python checking for python >= 3.4
-- Python checking for python >= 3.4 - found
--
-- Python checking for mako >= 0.4.2
-- Python checking for mako >= 0.4.2 - found
-- Looking for C++ include filesystem
-- Looking for C++ include filesystem - found
-- Performing Test CXX_FILESYSTEM_NO_LINK_NEEDED
-- Performing Test CXX_FILESYSTEM_NO_LINK_NEEDED - Success
-- Looking for aligned_alloc
-- Looking for aligned_alloc - found
-- Found PkgConfig: /usr/local/bin/pkg-config (found version "1.7.4")
-- Checking for module 'orc-0.4 > 0.4.11'
--   Found orc-0.4 , version 0.4.31
-- Found ORC: /usr/local/lib/liborc-0.4.so
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
-- QA Testing is enabled.
--   Modify using: -DENABLE_TESTING=ON/OFF
-- System profiling is disabled.
--   Modify using: -DENABLE_PROFILING=ON/OFF
-- Looking for cpuid.h
-- Looking for cpuid.h - found
-- Looking for intrin.h
-- Looking for intrin.h - not found
-- Looking for fenv.h
-- Looking for fenv.h - found
-- Looking for dlfcn.h
-- Looking for dlfcn.h - found
-- Compiler name: Clang
-- x86* CPU detected
-- Performing Test have_mfloat_abi_softfp
-- Performing Test have_mfloat_abi_softfp - Success
-- Performing Test have_mfloat_abi_hard
-- Performing Test have_mfloat_abi_hard - Success
-- Performing Test have_funsafe_math_optimizations
-- Performing Test have_funsafe_math_optimizations - Success
-- Performing Test have_mfpu_neon
-- Performing Test have_mfpu_neon - Success
[... 1316 lines cut ...]
        109 - qa_volk_32u_byteswappuppet_32u (Failed)
        111 - qa_volk_32u_popcntpuppet_32u (Failed)
        112 - qa_volk_32u_reverse_32u (Failed)
        113 - qa_volk_64f_convert_32f (Failed)
        114 - qa_volk_64f_x2_add_64f (Failed)
        115 - qa_volk_64f_x2_max_64f (Failed)
        116 - qa_volk_64f_x2_min_64f (Failed)
        117 - qa_volk_64f_x2_multiply_64f (Failed)
        119 - qa_volk_64u_byteswappuppet_64u (Failed)
        121 - qa_volk_64u_popcntpuppet_64u (Failed)
        122 - qa_volk_8i_convert_16i (Failed)
        123 - qa_volk_8i_s32f_convert_32f (Failed)
        124 - qa_volk_8ic_deinterleave_16i_x2 (Failed)
        125 - qa_volk_8ic_deinterleave_real_16i (Failed)
        126 - qa_volk_8ic_deinterleave_real_8i (Failed)
        127 - qa_volk_8ic_s32f_deinterleave_32f_x2 (Failed)
        128 - qa_volk_8ic_s32f_deinterleave_real_32f (Failed)
        129 - qa_volk_8ic_x2_multiply_conjugate_16ic (Failed)
        130 - qa_volk_8ic_x2_s32f_multiply_conjugate_32fc (Failed)
        131 - qa_volk_8u_conv_k7_r2puppet_8u (Failed)
        134 - qa_volk_8u_x3_encodepolarpuppet_8u (Failed)
Errors while running CTest
```

Hope it helps

dkozel, 2021-06-14:
Hi @diizzyy Which CPU do you have in your computer? An Intel Ivy Bridge Core processor?

diizzyy, 2021-06-15:
Ahh sorry, yes it's an Ivy Bridge one
https://ark.intel.com/content/www/us/en/ark/products/65719/intel-core-i7-3770-processor-8m-cache-up-to-3-90-ghz.html

jdemel, 2021-06-15:
Now, we can analyze this issue a bit further. Tests that fail show this line:
`Illegal instruction (core dumped)`
That's an error you receive if you e.g. want to execute an AVX512 command on a non AVX512 platform. With your specific CPU, the same would happen for AVX2 instructions.
We use cpu_features to detect which CPU extensions are available at runtime and call the appropriate specialized kernel. This is a one-time overhead at first function call. Since we compile all platform versions for x86_64, we need to do this check on runtime. CPU feature detection is quite difficult to do right. We had bugs in our own implementation that prevented us from cross platform support, thus we switched to a dependency where we hope all such knowledge is gathered in one place.

Without `cpu_features` compiled in, we can only rely on information acquired during compile time, i.e. a 64bit build will only offer SSE2. A 32bit build will only use generic kernels. Similar rules apply for ARM/NEON.
Currently, we don't provide such capabilities during compile time. We always relied on runtime feature detection.

diizzyy, 2021-06-15:
Perhaps a better approach would be to assume that lets instructions up to SSE4 and/or AVX are available if cpu_features are disabled or just remove that option?

vishwin, 2021-06-15:
The main issue for FreeBSD is how we do our sysctls for such detection, see google/cpu_features#159.

jdemel, 2021-06-16:
@diizzyy Unfortunately, there isn't a "quick fix". I pushed to adopt cpu_features because it is difficult and cumbersome to implement CPU feature detection (and we had long standing bugs that we could fix on the way).

@vishwin Thanks for this hint. I hope this PR will be merged soon. Afterwards we can update our cpu_feature dependency and we might even find a way to run CI for FreeBSD. At least I'd be happy to have it.

jdemel, 2021-12-22:
I assume we can fix this issue by moving to a newer `cpu_features` version.
https://github.com/google/cpu_features/pull/163

We should have a look at their CI system to run VOLK tests as well.

PR #550 (2021-12-23): cpu_features: Update submodule pointer
We have 3 issues that should be fixed with this commit.

Fix #428
Should be fixed because cpu_features detects `arm64` now. Thus, VOLK builds with cpu_features on MacOS and reports M1 capabilities.

Fix #478
Fix #484
cpu_features received quite a bit of contributions for FreeBSD. All the issues we had should be fixed now. However, this might require further evaluation.
commit c8198263 (2021-12-23): cpu_features: Update submodule pointer

We have 3 issues that should be fixed with this commit.

Fix #428
Should be fixed because cpu_features detects `arm64` now. Thus, it
builds on MacOS and reports M1 capabilities.

Fix #478
Fix #484
cpu_features received quite a bit of contributions for FreeBSD. All the
issues we had should be fixed now. However, this might require further
evaluation.
closed 2021-12-23 by michaelld