gnuradio/volk #492: Relicense VOLK under LGPL

jdemel, 2021-09-08:
VOLK is migrating from GNU General Public License v3.0 or later (GPL-3.0-or-later) to GNU Lesser General Public License v3.0 or later (LGPL-3.0-or-later).

**Please support us in this effort to re-license VOLK. This is really a long-standing community wish.**

We need to collect re-submissions from all contributors to do so. We might be forced to remove commits that are not resubmitted.

How to re-submit your contributions:
You don't need to start a PR with all your previous commits. You only need to clearly state that you want to an can re-submit your contributions.

We collect a list of all contributors that resubmitted their code in [AUTHORS_RESUBMITTING_UNDER_LGPL_LICENSE](https://github.com/gnuradio/volk/blob/main/AUTHORS_RESUBMITTING_UNDER_LGPL_LICENSE.md).

How do I add myself to this list?

1. Open a PR that adds yourself to the list. Use the suggested wording in your git commit message **Preferred**
2. Write an email to the maintainers, Michael Dickens or Johannes Demel, that clearly states your wish to be added to the list. We will add the email contents to the corresponding git commit that adds your name to the list.
3. Add a comment to this issue 

In order to make things easier for everyone, we prepared a statement that only needs your info
- your name that you used for your commits.
- your GitHub handle (helps us identify commits)
- your email address(es) that you used for your commits. Again to identify commits.

Suggested wording:
> I, {contributor name}, hereby resubmit all my contributions to the VOLK project and repository under the terms of the LGPLv3+. My GitHub handle is {github handle}, my email addresses used for contributions are: {email address(es)}, ... .

michaelld, 2021-09-08:
To folks reading this issue, and to clarify one point: Your relicensing information submitted in any form _must_ contain some variant on the `Suggested wording` noted above -- though it's easiest to use that language instead of creating your own. If submitting a PR, then the language can be in the commit or in the PR itself. Just changing the noted file (`AUTHORS_RESUBMITTING_UNDER_LGPL_LICENSE`) is _not_ enough. We need your _explicit_ relicensing approval; being explicit is _the_ major key here. Without your explicit approval in writing, we cannot accept your changes to the relicensing file. Thanks for you attention to this seemingly minor detail.

Teque5, 2021-09-29:
All code written by The Aerospace Corporation (including all @aero.org email addresses) was released to the GNU Radio project per our prior signed CLA. As such we have no issue with VOLK as a GNU Radio subproject re-licencing that code as LGPL. As a company we prefer to release as LGPL or GPL when possible.

michaelld, 2021-09-29:
@Teque5 Thanks for that note. I find these email addresses in the GIT log: `damian.miralles@aero.org` and `jessica.iwamoto@aero.org`. It's not clear that either of those folks work for The Aerospace Corporation any longer, so a blanket approval is an option so long as whoever is doing the approval has the authority to do so. It looks from your GitHub profile that you do work there currently, correct? If that is the case, then there are 3 options for proceeding that I can think of: (1) each individual noted does relicensing via a PR as noted above. (2) You create a PR that notes all of the people working for The Aerospace Corporation as listed along with your note above, a blanket statement covering all of The Aerospace Corporation. (3) I/we create the PR and you, representing The Aerospace Corporation, approve of it. Preferences?

Teque5, 2021-09-30:
> It looks from your GitHub profile that you do work there currently, correct?

Myself and another engineer essentially _own_ and manage the company's presence on GitHub.

> If that is the case, then there are 3 options for proceeding

I thought that my comment above satisfies your (2) without creating a PR. I can make a PR if you like but it will say the same thing as that comment.

jdemel, 2022-07-01:
With #577 we just added the last person to the list. Thus, we're ready to go ahead with a VOLK 3.0 release under LGPL.

jdemel, 2022-07-21:
Being technical: There are 3 people left (out of 74) who we haven't been able to get in contact with (at all), for a total of 4 (out of 1092) commits, 13 (of 282822) additions, and 7 (of 170421) deletions. We have reviewed these commits and all are simple changes (e.g., 1 line change) and most are no longer relevant (e.g., to a file that no longer exists). Volk maintainers (@michaelld  and @jdemel) are in agreement that the combination -- small numbers of changes per committer, simple changes per commit, commits no longer relevant -- means that we can proceed with relicensing without the approval of the folks. We will try reaching out periodically to these folks, but we believe it unlikely we will get a reply.

michaelld, 2022-07-21:
Totally agree with what @jdemel wrote, which is good ;)

PR #604 (2023-01-14): Prepare license switch
So far, this is a draft.

This PR does multiple things
- Removes GPL license files in favor of the new LGPL license files according to the new license that we use now.
- Temporarily adds a script that changes the SPDX license identifiers in all files were it currently says "GPL v3 ...".
- Makes sure that all reasonable files carry an SPDX license identifier. Restrictions apply to some file types, e.g. binary, auto-generated, orc and assembler files.

The intention of this PR is to make the switch as simple as possible. The only thing we would need to do is:
- run the scr [... cut]
closed 2023-01-14 by jdemel