gnuradio/volk #553: installation problem (-- Python checking for mako >= 0.4.2 - not found #594

nminpc, 2021-12-30:
-- Python checking for distutils
<string>:3: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
-- Python checking for distutils - found
<string>:2: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
<string>:2: DeprecationWarning: The distutils.sysconfig module is deprecated, use sysconfig instead
-- 
-- Python checking for python >= 3.4
-- Python checking for python >= 3.4 - found
-- 
-- Python checking for mako >= 0.4.2
-- Python checking for mako >= 0.4.2 - not found
**CMake Error at CMakeLists.txt:173 (message):
  Mako templates required to build VOLK**                                                                      
                                                                                                             
                                                                                                             
-- Configuring incomplete, errors occurred!
See also "/home/kali/build/volk/build/CMakeFiles/CMakeOutput.log".
make: *** No targets specified and no makefile found.  Stop.
make: *** No rule to make target 'test'.  Stop.
make: *** No rule to make target 'install'.  Stop.

pip install Mako==1.1.6 (does not work)

jdemel, 2021-12-30:
Hi @nminpc! Thanks for your report. There are a few things to unwrap here.

I assume your system lacks mako at the moment. Typically, I install it with [python3-mako](https://packages.ubuntu.com/focal/python3-mako) on Ubuntu.
I you want to install Mako via `pip`, you would probably do `pip install mako` and use [pip installed Mako](https://pypi.org/project/Mako/) from PyPI.

Besides, my comments involve a bit of guesswork. We'd need more info about your system.
Which OS/distro do you use? How did you install any other dependencies? i.e. how do you want to build and install VOLK?

The other issue is the ["Python distutils deprecation warning from PEP632"](https://www.python.org/dev/peps/pep-0632/). For now, we can ignore it. However, we need to update our build system to avoid problems with an upcoming Python 3.12.

We will probably follow the approach taken in https://github.com/gnuradio/gnuradio/pull/5371.
The most interesting bit for us is:
https://github.com/gnuradio/gnuradio/blob/2e77edb7136a3be73f7a7e8b047813494daa6bc9/cmake/Modules/GrPython.cmake#L145
We need to update this here:
https://github.com/gnuradio/volk/blob/b5b990b4afe2c49b87111ef52006b1efec6bd9bb/cmake/Modules/VolkPython.cmake#L102-L113
However, we could use some time to rework that file. GR seems to have received quite a few improvements.

nminpc, 2021-12-30:
sudo apt-get install git
git clone https://github.com/pybombs/pybombs.git
cd pybombs
sudo python setup.py install
sudo pybombs recipes add gr-recipes git+https://github.com/gnuradio/gr-recipes.git
sudo pybombs recipes add gr-etcetera git+https://github.com/gnuradio/gr-etcetera.git
sudo pybombs prefix init /usr/local/ -a gr-gsm
sudo pybombs -p gr-gsm install gr-gsm

jdemel, 2022-01-07:
The deprecation warning should be fixed with #554 .

OK, so you try to install `gr-gsm` via pybombs.
Still, it is very helpful if you provide more info about your system like I asked before:
- Which OS/distro do you use? 
- How did you install any other dependencies?

Did you try `sudo pybombs install libvolk`? That would help to localize the issue.

Besides:
> pip install Mako==1.1.6 (does not work)

Did you just try `pip install Mako`?
Probably even better: You might want to use the Mako version your distribution provides.
On Debianoids: `sudo apt install python3-mako`. (includes Ubuntu).

In most cases, PyBOMBS is run without privileges and asks for them if necessary. That might be an issue as well.

jdemel, 2022-04-12:
Closing this issue now. If there are more VOLK related issues, feel free to re-open.

closed 2022-04-12 by jdemel