gnuradio/volk #571: volk_cpu.tmpl.c on non-X86 cpus

michael-roe, 2022-05-22:
Looking at the code, it treats anything starting with "neon" as an ARM feature, and everything else as x86.

I haven't tried running it yet, but I think this will break when run on any architecture that isn't ARM or X86.

(I am going to upstream some MIPS fixes to cpu_features, and I have several different MIPS workstations to test on. If I have time and I'm feeling brave enough, I might try writing a RISC-V port of CPU features.)

I think we may need a more general mechanism to say which features are part of which architecture.

jdemel, 2022-07-23:
https://github.com/gnuradio/volk/blob/af69c60e908cd64b3bacd4ee6b4ce4a52b711865/tmpl/volk_cpu.tmpl.c#L101-L108

I had a closer look at your second comment. I assume you refer to the `volk_get_`lvarch` function. It is only used in `volk.c`, e.g.

https://github.com/gnuradio/volk/blob/af69c60e908cd64b3bacd4ee6b4ce4a52b711865/tmpl/volk.tmpl.c#L48

It is related to this field:
https://github.com/gnuradio/volk/blob/af69c60e908cd64b3bacd4ee6b4ce4a52b711865/tmpl/volk_machines.tmpl.h#L34
 
Currently, numbers up to `23` are used. This implies that 23 bit are used. It'd be better to start looking for a better solution here.

jdemel, 2023-10-03:
Since we use `cpu_features` with MIPS etc. now, and things still work, I suggest to close this issue.