gnuradio/volk #572: Test volk_32fc_s32fc_multiply_32fc fails on MIPS

michael-roe, 2022-05-26:
Steps to reproduce (on MIPS64 CPU, running Debian 11):

./volk_test_all volk_32fc_s32fc_multiply_32fc

Output:

RUN_VOLK_TESTS: volk_32fc_s32fc_multiply_32fc(131071,1)
[ 5072.839629] do_page_fault(): sending SIGSEGV to volk_test_all for invalid read access from 0000000000002291
[ 5072.840027] epc = 000000fff2dcfd94 in libc-2.31.so[fff2d1c000+1b5000]
[ 5072.840490] ra  = 000000fff326cc48 in libvolk.so.2.5.1[fff325c000+6a000]
Segmentation fault

=========

I haven't investigated what's going wrong here, but one possible cause of this type of bug is that MIPS is stricter about unaligned pointer accesses than other CPU architectures.

=====

The following tests are also failing, possibly for a similar reason:

volk_32fc_x2_s32fc_multiply_conjugate_add_32fc
volk_32fc_s32fc_rotatorpuppet_32fc

michael-roe, 2022-05-26:
See also this blog post:
https://trust-in-soft.com/blog/2020/04/06/gcc-always-assumes-aligned-pointer-accesses/

jdemel, 2022-05-27:
I think, we have a similar issue with PowerPC. At least the failing tests show the same pattern.

All failing tests include a scalar complex value as a parameter.

#442 might be helpful.

michael-roe, 2022-05-27:
@jdemel: Thanks! I agree it could be the same issue.

We ought to fix this, even if it needs an incompatible change to the API.

jdemel, 2022-05-27:
I'd really like to do this too. However, there are some pitfalls.

The library itself is written in C. Thus, we require support for complex values. And MSVC does not support C standard complex. The solution to this issue was that VOLK is compiled in C++ mode for MSVC. In order to make this work, we need this interface that basically assumes C++ `std::complex` is the same as `_Complex`. This assumption is incorrect and we're lucky it works on x86 and ARM.

I started to fix this in #488. However, this is still a draft. It all comes back to the issue that MSVC lacks features.

argilo, 2023-11-06:
Since this issue has the same root cause as #442, I'll close this off and we can continue the discussion there.

closed 2023-11-06 by argilo