gnuradio/volk #581: Prepare and release VOLK 3.0

jdemel, 2022-07-23:
We mostly finished #492, aka VOLK re-licensing.

This issue is intended to track work in preparation of a VOLK 3.0 release. VOLK 3.0 is intended to be the first LGPL-only release. Right before that we will see a final 2.x release.

This is a list of things, we need to do before the release.

1. Fixup SPDX license identifiers. In the end, we want to be able to run a simple find and replace to make sure that VOLK 3.0 has the correct LGPL header everywhere.
2. Re-organize the top-level folder. This was already started in #580 .
3. Move Doxygen related CMake code and files into the `docs/` folder.
4. Have a final VOLK 2.x release
5. Clean-up the license files.
6. Have a first VOLK 3.0 release that is LGPL-only.

I'm aware that there are long-standing issues that would require an API change. However, fixing these issues appropriately is a long process. I want to make sure, we do the license API change as soon as possible and the the "license-limbo".

michaelld, 2022-08-02:
What shall we do about copyright statements? Both the years as well as the copyright owner? Does FSF still own the copyright? I thought GR is now under SETI?

I like what GR has done for the years ... from the top level README:
```
## Legal Matters

Some files have been changed many times throughout the years. Copyright
notices at the top of source files list which years changes have been
made. For some files, changes have occurred in many consecutive years.
These files may often have the format of a year range (e.g., "2006 - 2011"),
which indicates that these files have had copyrightable changes made
during each year in the range, inclusive.
```

jdemel, 2022-08-02:
It'd be a huge effort to update all the copyright years in every file. Obviously, we'd need to have a script to automate that.
As far as I know, GR kept the old copyright notices and only new commits receive individual copyright entries. However, we re-license VOLK. Everyone re-submitted their code. How does that change things?

michaelld, 2022-08-02:
Yes redoing the (C) years would need to be automated. Not sure it's worth it. Was more wondering if we should, moving forward, go to either: (1) just use the initial creation year, since (C) is good for a long time & we will not be around when that time comes; (2) use ["creation year" - "latest change year"] to be inclusive of all possible changes for the given file; (3) just use "current year". IDK which is better / best / worst, but since we're going all this work we can determine a reasonable path forward, and then go with it.

michaelld, 2022-08-02:
(4) stick with what we have ...

PR #604 (2023-01-14): Prepare license switch
So far, this is a draft.

This PR does multiple things
- Removes GPL license files in favor of the new LGPL license files according to the new license that we use now.
- Temporarily adds a script that changes the SPDX license identifiers in all files were it currently says "GPL v3 ...".
- Makes sure that all reasonable files carry an SPDX license identifier. Restrictions apply to some file types, e.g. binary, auto-generated, orc and assembler files.

The intention of this PR is to make the switch as simple as possible. The only thing we would need to do is:
- run the scr [... cut]
closed 2023-01-14 by jdemel