gnuradio/volk #598: Doxyfile includes build paths

jdemel, 2022-09-04:
Doyxgen includes build paths quite excessively. First, these lines in `docs/Doxyfile.in` include absolute paths after the CMake `configure_file` command in `Doxyfile`.
```
PROJECT_LOGO           = @CMAKE_SOURCE_DIR@/docs/volk_logo_small.png
LAYOUT_FILE            = @CMAKE_CURRENT_SOURCE_DIR@/DoxygenLayout.xml
INPUT                  = @CMAKE_SOURCE_DIR@
EXCLUDE                = @CMAKE_BINARY_DIR@ @CMAKE_SOURCE_DIR@/cpu_features @CMAKE_SOURCE_DIR@/README.md @CMAKE_SOURCE_DIR@/docs/AUTHORS_RESUBMITTING_UNDER_LGPL_LICENSE.md
```
Also, doxygen includes the build path in a lot of titles, e.g.:
```
./_r_e_a_d_m_e_8md.html:<title>Vector Optimized Library of Kernels: {PATH_TO_MY_VOLK_REPO_CLONE}/kernels/README.md File Reference</title>
./_r_e_a_d_m_e_8md.html:<div class="title">{PATH_TO_MY_VOLK_REPO_CLONE}/kernels/README.md File Reference</div>  </div>
```

[Doxygen seems to be problematic with relative paths](https://www.doxygen.nl/manual/config.html#cfg_output_directory).

maitbot, 2022-09-05:
The generated Doxyfile should contain build paths - as it also should not be installed.

So far I have set
`STRIP_FROM_PATH        = @CMAKE_BINARY_DIR@ @CMAKE_SOURCE_DIR
STRIP_FROM_INC_PATH    = @CMAKE_SOURCE_DIR@/include @CMAKE_BINARY_DIR@ @CMAKE_SOURCE_DIR@/lib
SHOW_USED_FILES        = NO
EXCLUDE                = @CMAKE_BINARY_DIR@ @CMAKE_SOURCE_DIR@/cpu_features @CMAKE_SOURCE_DIR@/README.md @CMAKE_SOURCE_DIR@/cmake @CMAKE_SOURCE_DIR@/docs/AUTHORS_RESUBMITTING_UNDER_LGPL_LICENSE.md @CMAKE_SOURCE_DIR@/apps @CMAKE_SOURCE_DIR@/lib/*qa* @CMAKE_SOURCE_DIR@/tmpl
REFERENCES_LINK_SOURCE = NO
SOURCE_TOOLTIPS        = NO
HTML_FOOTER            = ""
`
which has greatly reduced the number of build paths in the generated HTML documentation.

But for whatever reason, I cannot seem to make
unionp__decision__t.html
structchar__split.html
uniondecision__t.html
structdword__split.html

put too much of the build path into the include line.

maitbot, 2022-09-05:
Work-in-progress at https://github.com/maitbot/volk/tree/mait/omit-doxygen-build-paths