gnuradio/volk #601: build problem 2.5.2 32 bit

cinaeds, 2022-09-11:
I can't bulid versiion 2.5.2 volk on a 32 bit armv7l cpu.

[build.log](https://github.com/gnuradio/volk/files/9543607/build.log)
[compile.log](https://github.com/gnuradio/volk/files/9543605/compile.log)

jdemel, 2022-09-12:
How did you run CMake etc?
I've read multiple reports that it is problematic to rely on CMake to detect the correct 32bit ARM CPU. Did you use a specific toolchain file?

balister, 2022-09-21:
I see an issue in my OE build which should have a good toolchain file. I'll investigate. Oddly enough, not sure it is the same one though .....

cinaeds, 2022-09-22:
Okay, thanks for the update.

It's 32 machine - I  use it primarily for viewing and testing flow 
charts since it has good frame buffer and a large monitor.

Would it be possible to just fool gnuradio it so I can build gnuradio 
3.10 on it?

-- Cinaed
-

balister, 2022-09-22:
Can you show the command line you use to build gnuradio? POking around, I think your issue is the the -mfpu=neon flag isn't making it into the build.

cinaeds, 2022-09-22:
Here are the cflags I passed cmake for volk (and gnuradio):

    armv7l)
      CFLAGS="-march=armv7ve -O2 -mtune=cortex-a15.cortex-a7 
-mfloat-abi=hard -mfpu=neon-vfpv4"

Here are the are the error messages - it bombs at about 44% completion:

/opt/gnuradio/src/gr-pre-libs/volk/kernels/volk/asm/neon/volk_16i_max_star_horizontal_16i.s: 
Assembler messages:
/opt/gnuradio/src/gr-pre-libs/volk/kernels/volk/asm/neon/volk_16i_max_star_horizontal_16i.s:11: 
Error: selected FPU does not support instruction -- `vmov.i32 q12,#0'
/opt/gnuradio/src/gr-pre-libs/volk/kernels/volk/asm/neon/volk_16i_max_star_horizontal_16i.s:22: 
Error: selected FPU does not support instruction -- `vsub.i16 q10,q8,q9'
/opt/gnuradio/src/gr-pre-libs/volk/kernels/volk/asm/neon/volk_16i_max_star_horizontal_16i.s:23: 
Error: selected processor does not support `vcge.s16 q11,q10,#0' in ARM mode
/opt/gnuradio/src/gr-pre-libs/volk/kernels/volk/asm/neon/volk_16i_max_star_horizontal_16i.s:24: 
Error: selected processor does not support `vcgt.s16 q10,q12,q10' in ARM 
mode
/opt/gnuradio/src/gr-pre-libs/volk/kernels/volk/asm/neon/volk_16i_max_star_horizontal_16i.s:25: 
Error: selected processor does not support `vand.i16 q11,q8,q11' in ARM mode
/opt/gnuradio/src/gr-pre-libs/volk/kernels/volk/asm/neon/volk_16i_max_star_horizontal_16i.s:26: 
Error: selected processor does not support `vand.i16 q10,q9,q10' in ARM mode
/opt/gnuradio/src/gr-pre-libs/volk/kernels/volk/asm/neon/volk_16i_max_star_horizontal_16i.s:27: 
Error: selected FPU does not support instruction -- `vadd.i16 q10,q11,q10'
make[2]: *** [lib/CMakeFiles/volk_obj.dir/build.make:1825: 
lib/CMakeFiles/volk_obj.dir/__/kernels/volk/asm/neon/volk_16i_max_star_horizontal_16i.s.o] 
Error 1

-- Cinaed

balister, 2022-09-23:
So the issue is you need to set the flags used by the assembler to the ones you are using for CFLAGS. Can you try setting ASFLAGS to the same as CFLAGS?

cinaeds, 2022-09-24:
Progress!

In instead of crashing at 44%, it now crashes at 83%.

-- Cinaed

[ 84%] Building C object lib/CMakeFiles/volk_obj.dir/volk_machine_neon_orc.c.o
In file included from /opt/gnuradio/src/gr-pre-libs/volk/kernels/volk/volk_8u_x4_conv_k7_r2_8u.h:624,
                 from /opt/gnuradio/src/gr-pre-libs/volk/kernels/volk/volk_8u_conv_k7_r2puppet_8u.h:15,
                 from /opt/gnuradio/src/gr-pre-libs/volk/build/lib/volk_machine_neon_orc.c:163:
/opt/gnuradio/src/gr-pre-libs/volk/include/volk/sse2neon.h: In function ‘_sse2neon_vld1q_u8_x4’:
/opt/gnuradio/src/gr-pre-libs/volk/include/volk/sse2neon.h:463:12: warning: implicit declaration of function ‘vld1q_u8_x4’; did you mean ‘vld1q_u64’? [-Wimplicit-function-declaration]
  463 |     return vld1q_u8_x4(p);
      |            ^~~~~~~~~~~
      |            vld1q_u64
/opt/gnuradio/src/gr-pre-libs/volk/include/volk/sse2neon.h:463:12: error: incompatible types when returning type ‘int’ but ‘uint8x16x4_t’ was expected
  463 |     return vld1q_u8_x4(p);
      |            ^~~~~~~~~~~~~~
/opt/gnuradio/src/gr-pre-libs/volk/include/volk/sse2neon.h: In function ‘_mm_cmpge_pd’:
/opt/gnuradio/src/gr-pre-libs/volk/include/volk/sse2neon.h:3330:14: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
 3330 |     d[0] = (*(double *) &a0) >= (*(double *) &b0) ? ~UINT64_C(0) : UINT64_C(0);
      |              ^~~~~~~~~~~~~~
/opt/gnuradio/src/gr-pre-libs/volk/include/volk/sse2neon.h:3330:35: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
 3330 |     d[0] = (*(double *) &a0) >= (*(double *) &b0) ? ~UINT64_C(0) : UINT64_C(0);
      |                                   ^~~~~~~~~~~~~~
/opt/gnuradio/src/gr-pre-libs/volk/include/volk/sse2neon.h:3331:14: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
 3331 |     d[1] = (*(double *) &a1) >= (*(double *) &b1) ? ~UINT64_C(0) : UINT64_C(0);
      |              ^~~~~~~~~~~~~~
/opt/gnuradio/src/gr-pre-libs/volk/include/volk/sse2neon.h:3331:35: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
 3331 |     d[1] = (*(double *) &a1) >= (*(double *) &b1) ? ~UINT64_C(0) : UINT64_C(0);
      |                                   ^~~~~~~~~~~~~~
/opt/gnuradio/src/gr-pre-libs/volk/include/volk/sse2neon.h: In function ‘_mm_cmpge_sd’:
/opt/gnuradio/src/gr-pre-libs/volk/include/volk/sse2neon.h:3351:14: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
 3351 |     d[0] = (*(double *) &a0) >= (*(double *) &b0) ? ~UINT64_C(0) : UINT64_C(0);
      |              ^~~~~~~~~~~~~~
/opt/gnuradio/src/gr-pre-libs/volk/include/volk/sse2neon.h:3351:35: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
 3351 |     d[0] = (*(double *) &a0) >= (*(double *) &b0) ? ~UINT64_C(0) : UINT64_C(0);
      |                                   ^~~~~~~~~~~~~~
/opt/gnuradio/src/gr-pre-libs/volk/include/volk/sse2neon.h: In function ‘_mm_cmpgt_pd’:
/opt/gnuradio/src/gr-pre-libs/volk/include/volk/sse2neon.h:3411:14: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
 3411 |     d[0] = (*(double *) &a0) > (*(double *) &b0) ? ~UINT64_C(0) : UINT64_C(0);
      |              ^~~~~~~~~~~~~~
/opt/gnuradio/src/gr-pre-libs/volk/include/volk/sse2neon.h:3411:34: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
 3411 |     d[0] = (*(double *) &a0) > (*(double *) &b0) ? ~UINT64_C(0) : UINT64_C(0);
      |                                  ^~~~~~~~~~~~~~
/opt/gnuradio/src/gr-pre-libs/volk/include/volk/sse2neon.h:3412:14: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
 3412 |     d[1] = (*(double *) &a1) > (*(double *) &b1) ? ~UINT64_C(0) : UINT64_C(0);
      |              ^~~~~~~~~~~~~~
/opt/gnuradio/src/gr-pre-libs/volk/include/volk/sse2neon.h:3412:34: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
 3412 |     d[1] = (*(double *) &a1) > (*(double *) &b1) ? ~UINT64_C(0) : UINT64_C(0);
      |                                  ^~~~~~~~~~~~~~
/opt/gnuradio/src/gr-pre-libs/volk/include/volk/sse2neon.h: In function ‘_mm_cmpgt_sd’:
/opt/gnuradio/src/gr-pre-libs/volk/include/volk/sse2neon.h:3432:14: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
 3432 |     d[0] = (*(double *) &a0) > (*(double *) &b0) ? ~UINT64_C(0) : UINT64_C(0);
      |              ^~~~~~~~~~~~~~
/opt/gnuradio/src/gr-pre-libs/volk/include/volk/sse2neon.h:3432:34: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
 3432 |     d[0] = (*(double *) &a0) > (*(double *) &b0) ? ~UINT64_C(0) : UINT64_C(0);
      |                                  ^~~~~~~~~~~~~~
/opt/gnuradio/src/gr-pre-libs/volk/include/volk/sse2neon.h: In function ‘_mm_cmple_pd’:
/opt/gnuradio/src/gr-pre-libs/volk/include/volk/sse2neon.h:3453:14: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
 3453 |     d[0] = (*(double *) &a0) <= (*(double *) &b0) ? ~UINT64_C(0) : UINT64_C(0);
      |              ^~~~~~~~~~~~~~
[... cut ...]
/opt/gnuradio/src/gr-pre-libs/volk/include/volk/sse2neon.h:7688:20: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
 7688 |     double sum = *((double *) &tmp) + *(((double *) &tmp) + 1);
      |                   ~^~~~~~~~~~~~~~~~
make[2]: *** [lib/CMakeFiles/volk_obj.dir/build.make:2457: lib/CMakeFiles/volk_obj.dir/volk_machine_neon_orc.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:342: lib/CMakeFiles/volk_obj.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
***@***.***:/opt/gnuradio/src/gr-pre-libs/volk/build#

balister, 2022-09-24:
I have a fix for that one in progress. Can you confirm what version of gcc you are using? I am guessing 12.

balister, 2022-09-24:
Try: https://github.com/gnuradio/volk/pull/602

cinaeds, 2022-09-25:
Great! Yes, the version of gcc is 12.1.0.

-- Cinaed

balister, 2022-09-26:
Can you confirm this fixes the issue and tell us a little about what  you are doing? I am guessing native compile on some 32 bit arm board. We need to document the use of CFLAGS and ASFLAGS for this case better.

cinaeds, 2022-09-26:
Sure, I used

   CFLAGS="-march=armv7ve -O2 -mtune=cortex-a15.cortex-a7 
-mfloat-abi=hard -mfpu=neon-vfpv4"
   ASMFLAGS="-march=armv7ve -O2 -mtune=cortex-a15.cortex-a7 
-mfloat-abi=hard -mfpu=neon-vfpv4"

noting  I ended up having to use ASMFLAGS instead of ASFLAGS.

The OS is Debian bookworm - or Debian 12 - and the gcc compiler was 
Debian 12.1.0-8.

-- Cinaed

balister, 2022-09-27:
OK this is weird ... see the definition here:
https://www.gnu.org/software/make/manual/html_node/Implicit-Variables.html

balister, 2022-09-27:
HMMM, Maybe ASFLAGS goes to the actual assembler and ASMFLAGS goes via gcc when called with -s to call the assembler. Quick google isn't returning anything I can point at though :(

cinaeds, 2022-10-03:
Yes, that was my assumption too.

But I did find a work around.

I was able to find the script used to build gr-3.7 and the source code 
for volk-2.4.1.

And I was able to configure and build volk and gr-3.10 without any 
issues - but it warned me at the end the of the configuration of gr to 
enable volk.

It may not be optimal - but for what I'm using the machine for - it's 
close enough.

Enclosed is a copy of the volk-2.4.1 profile.

-- CInaed

#this file is generated by volk_profile.
#the function name is followed by the preferred architecture.
  
  
  
  
  
  
  
  
  
volk_16ic_deinterleave_real_8i u_orc u_orc
volk_16ic_deinterleave_16i_x2 u_orc u_orc
volk_16ic_s32f_deinterleave_32f_x2 u_orc u_orc
  
  
  
  
  
  
volk_16i_s32f_convert_32f a_generic generic
volk_16i_convert_8i generic generic
  
  
volk_32f_x2_add_32f u_orc u_orc
  
  
  
  
volk_32fc_32f_multiply_32fc u_orc u_orc
  
volk_32f_log2_32f u_generic u_generic
  
  
  
volk_32f_cos_32f generic generic
  
  
  
  
  
  
  
volk_32fc_x2_conjugate_dot_prod_32fc block block
  
  
volk_32fc_deinterleave_64f_x2 a_generic generic
  
  
  
  
volk_32fc_x2_dot_prod_32fc generic generic
  
  
  
  
  
  
volk_32fc_magnitude_32f generic generic
volk_32fc_magnitude_squared_32f generic generic
  
volk_32fc_x2_multiply_32fc generic generic
volk_32fc_x2_multiply_conjugate_32fc generic generic
  
volk_32fc_conjugate_32fc a_generic generic
volk_32f_s32f_convert_16i a_generic generic
volk_32f_s32f_convert_32i a_generic generic
volk_32f_convert_64f generic generic
volk_32f_s32f_convert_8i a_generic generic
  
  
  
  
volk_32f_x2_divide_32f u_orc u_orc
volk_32f_x2_dot_prod_32f generic generic
  
  
volk_32f_x2_max_32f u_orc u_orc
volk_32f_x2_min_32f u_orc u_orc
volk_32f_x2_multiply_32f u_orc u_orc
  
  
volk_32f_s32f_normalize u_orc u_orc
  
volk_32f_sqrt_32f u_orc u_orc
  
  
volk_32f_x2_subtract_32f u_orc u_orc
  
volk_32i_x2_and_32i u_orc u_orc
volk_32i_s32f_convert_32f a_generic generic
volk_32i_x2_or_32i u_orc u_orc
  
volk_64f_convert_32f generic generic
  
  
  
  
  
  
  
  
  
  
  
volk_8i_convert_16i u_orc u_orc
volk_8i_s32f_convert_32f u_orc u_orc
volk_32fc_s32fc_multiply_32fc a_generic generic
volk_32f_s32f_multiply_32f u_orc u_orc
  
volk_32f_binary_slicer_32i generic_branchless generic_branchless
volk_32f_binary_slicer_8i generic_branchless generic_branchless
volk_32u_reverse_32u lut lut
volk_32f_tanh_32f series series

closed 2022-12-25 by jdemel