gnuradio/volk #608: volk_32fc_x2_divide_32fc gives nan in aarch64 for 0.0 / 1e-20

daniestevez, 2022-11-19:
Investigating a bug in gr-satellites (https://github.com/daniestevez/gr-satellites/issues/416), we have found that the `volk_32fc_x2_divide_32fc` kernel gives `nan + 1j nan` when calculating the division of `0.0 + 1j 0.0` by `1e-20 + 1j 0.0` in aarch64. This seems to happen both in the `generic` kernel and with the `neon` kernel (though replicating these tests on another aarch64 machine would be a good idea). The testing has been done in a Rasberry Pi 4 with the [volk 2.4.1 package in Debian Bullseye](https://packages.debian.org/bullseye/libvolk2.4).

I don't have an aarch64 machine at hand, but I've tested on an x86_64 and an armv7 machine and the result is `0.0 + 1j 0.0` as we would expect.

I'm using the following program to test:
```c
#include <volk/volk.h>
#include <complex.h>
#include <stdio.h>
#include <string.h>

int main(int argc, char **argv) {
    size_t alignment = volk_get_alignment();
    float complex *z = volk_malloc(512 * sizeof(float complex), alignment);
    float complex *w = volk_malloc(512 * sizeof(float complex), alignment);
    float complex *x = volk_malloc(512 * sizeof(float complex), alignment);
    memset(z, 0, 512 * sizeof(float complex));
    memset(w, 0, 512 * sizeof(float complex));
    z[0] = 0.0f;
    w[0] = 1e-20f;
    volk_32fc_x2_divide_32fc(x, z, w, 512);
    printf("%e + 1j %e\n", creal(x[0]), cimag(x[0]));
    return 0;
}
```
This is to be built as
```
gcc -O3 -Wall -o test_divide_volk test_divide_volk.c -lvolk
```

If we replace `1e-20` by `5.5e-20`, the test works in aarch64 (it gives zero rather than nan).

For gr-satellites this is not a huge problem, because the reason this popped up is because we have an AGC that does roughly `signal / (rms_amplitude(signal) + 1e-20))`. The `1e-20` in the denominator is there to prevent `0 / 0` when the input signal is zero. We can increase this small `1e-20` to whatever value is safe for all architectures.

However, I would like to chase this rabbit hole further and understand why this problem happens, whether something can be fixed or we have to live with this (for instance if setting up the kernel so that `0 / 1e-20` gives correct results would hurt performance), and what is a safe value for all architectures (apparently `5.5e-20` works on aarch64, but it still could be too small on other machines).

jdemel, 2022-11-19:
@daniestevez Thanks for the detailed issue report!

I tried to reproduce it. However, all I can say for now is: It works on my x86 AMD machine. I don't have an aarch64 machine available.

At the moment, I suppose we need to wait for someone who's able to re-produce the issue on an aarch64 machine and be willing to debug it further.

daniestevez, 2022-11-20:
I don't have access to an aarch64 machine at the moment either, but if I have enough time (not very likely) I might test it under qemu.

I have realised why computing `(0 + 1j 0) / (1e-20 + 1j 0)` might not be the best idea. The naïve way to compute the complex division $z/w$ is as $z\overline{w}/|w|^2$. If $w = 10^{-20}$, then $|w|^2 = 10^{-40}$. For a `float32`, [according to Wikipedia](https://en.wikipedia.org/wiki/Single-precision_floating-point_format#Exponent_encoding), the minimum positive normal value is $\sim 1.18 \cdot 10^{-38}$ and the minimum positive subnormal value is $~\sim 1.4 \cdot 10^{-45}$. I guess that an architecture that doesn't handle subnormal values too well would have trouble handling division by `1e-20 + 1j 0`. However, I don't think this is the whole story, because in aarch64 it seems that division by `5.5e-20 + 1j 0` already works well, and the modulus squared of this number is `3.025e-39`, which is still a subnormal value.

Aang23, 2022-12-25:
If anyone needs an aarch64 machine - I can lend ssh access no problem, got quite a few sitting around unused. I've also experienced this issue, but may not have time to debug it myself so soon.

argilo, 2023-11-05:
It looks like this is a consequence of using `-fcx-limited-range` and `-funsafe-math-optimizations`:

```c
#include <complex.h>
#include <stdio.h>
#include <stdlib.h>

int main(int argc, char *argv[]) {
  float complex a = atof(argv[1]) + atof(argv[2])*I;
  float complex b = atof(argv[3]) + atof(argv[4])*I;
  float complex c = a / b;
  printf("%f %f\n", crealf(c), cimagf(c));
  return 0;
}
```

```
$ gcc -fcx-limited-range -funsafe-math-optimizations -o test test.c && ./test 0 0 1e-20 0
nan nan
```

If either of those options are removed then the output is zero.

argilo, 2023-11-05:
Those compiler options probably make sense for a DSP environment where we want maximum performance, but correctness is sacrificed and we can't expect calculations involving extreme values to work.

jdemel, 2023-11-06:
So this is the thing, we actually want to fix
```bash
$ gcc -fcx-limited-range -funsafe-math-optimizations -o test test.c && ./test 0 0 1e-20 0
nan nan
```
One of two options:
- Remove `-fcx-limited-range`
- Remove `-funsafe-math-optimizations`
I read somewhere that GCC does not do vector optimization on ARM unless `-funsafe-math-optimizations` is present because NEON does not fully follow some math spec. This bites us in terms of correctness. I'd love to add references to my statements here but I've forgotten where in the GCC docs I read about this.

daniestevez, 2023-11-06:
@argilo your test gave me the clue to look at compiler explorer using `-fcx-limited-range` and `-funsafe-math-optimizations`. I agree that for DSP it is reasonable to use those flags, but I wanted to understand what is the fundamental difference that makes `(0.0 + 1j 0.0)/(1e-20 + 1j 0.0)` give `nan` on aarch64 but not on armv7 or x86_64.

[Here is my compiler explorer test](https://godbolt.org/z/z9z5b9oM5). Without `-fcx-limited-range` and `-funsafe-math-optimizations`, the function `__divsc3` is called in the three architectures. This function is supposed to do the division according to what C99 specifies.

With `-fcx-limited-range` and `-funsafe-math-optimizations`, code is generated to compute the division using some floating point instructions. In all cases they compute `(a + 1j b) / (c + 1j d)` as `(a*c + b*d)/(c*c + d*d) + 1j (b*c-a*d)/(c*c + d*d)`, as I suggested in one of my comments.

The interesting thing is that aarch64 is the only architecture that evaluates the inverse `1.0/(c*c + d*d)`. Then it multiplies the terms `a*c + b*d` and `b*c - a*d` by this inverse. The other two architectures instead perform the two divisions separately, without computing the inverse. What happens is that when `a = b = d = 0.0` and `c` is small, the calculation `1.0/(c*c + d*d)` gives `inf`, because the result is larger than the maximum `float`. This causes a `0/inf` division, yielding `nan`.

On the other architectures, the divisions such as `(a*c + b*d)/(c*c + d*d)` can be calculated correctly, because the numerator is `0.0` and the denominator is a very small number, but the result still gives zero.

Why the difference in the generated code? Probably because of some knowledge of the compiler about the cost of each instruction. In aarch64 it thinks it's cheaper to do one division and two multiplications, whereas in armv7 and x86_64 it thinks it's cheaper to do two divisions.

For me, in the context of volk, having `(0.0 + 1j 0.0)/(1e-20 + 1j 0.0)` give `nan` in some architectures and `0.0 + 1j 0.0` on others is acceptable. As Clayton said, we want maximum performance in DSP and are willing to sacrifice correctness in corner cases. The main reason I had for opening this issue was that I found this thing surprising and wanted to understand better why it happens. Having seen the compiler explorer output, now I think I have a good understanding, so I would be happy to close the issue.

daniestevez, 2023-11-06:
By the way, playing to see what happens if one remove one of the options `-fcx-limited-range` and `-funsafe-math-optimizations` on aarch64, I see that removing only `-fcx-limited-range` causes `__divsc3` to get called, whereas removing only `-funsafe-math-optimizations` generates NEON instructions but performs two divisions as in armv7 and x86_64. This is just a curiosity. I'm not suggesting that we should remove `-funsafe-math-optimizations` on aarch64.

argilo, 2023-11-06:
Thanks for the insight into the aarch64 result, @daniestevez!

I agree that the tradeoff is acceptable here, so I think it's OK for VOLK to keep using these compiler options.

jdemel, 2023-11-06:
@daniestevez would you be interested in documenting this behavior? I would suggest to add this info to the doc part of this kernel after the usage example.

daniestevez, 2023-11-06:
Yeah. I can do that. In fact I found that the exact compiler option that is behind what aarch64 does is `-freciprocal-math`.

daniestevez, 2023-11-06:
PR sent.

PR #694 (2023-11-07): volk_32fc_x2_divide_32fc: add documentation about numerical accuracy
This adds some documentation to the complex divide kernel about the potential pitfall in some architectures of obtaining inf or nan with a small denominator (even if the numerator is zero) due to reduced range because of -fcx-limited-range and -freciprocal-math.

This closes #608.
commit c3641104 (2023-11-07): volk_32fc_x2_divide_32fc: add documentation about numerical accuracy

This adds some documentation to the complex divide kernel about the
potential pitfall in some architectures of obtaining inf or nan
with a small denominator (even if the numerator is zero) due to
reduced range because of -fcx-limited-range and -freciprocal-math.

This closes #608.
closed 2023-11-07 by jdemel