gnuradio/volk #613: volk: GPL to LGPL changed not everywhere, e.g. in some *.cmake and CMakeLists.txt files

mobius217, 2023-01-14:
Run command:
grep -r -F " GPL-3" *
and see:
apps/CMakeLists.txt:# SPDX-License-Identifier: GPL-3.0-or-later
cmake/cmake_uninstall.cmake.in:# SPDX-License-Identifier: GPL-3.0-or-later
cmake/Modules/VolkConfig.cmake.in:# SPDX-License-Identifier: GPL-3.0-or-later
cmake/Modules/VolkConfigVersion.cmake.in:# SPDX-License-Identifier: GPL-3.0-or-later
CMakeLists.txt:# SPDX-License-Identifier: GPL-3.0-or-later
docs/CMakeLists.txt:# SPDX-License-Identifier: GPL-3.0-or-later
include/volk/volk_version.h.in: * SPDX-License-Identifier: GPL-3.0-or-later
lib/CMakeLists.txt:# SPDX-License-Identifier: GPL-3.0-or-later
lib/constants.c.in: * SPDX-License-Identifier: GPL-3.0-or-later
python/volk_modtool/CMakeLists.txt:# SPDX-License-Identifier: GPL-3.0-or-later

jdemel, 2023-01-15:
@mobius217 Thanks for the report!

Out of curiousity: How do you use VOLK?

closed 2023-01-15 by jdemel

mobius217, 2023-01-15:
> Out of curiousity: How do you use VOLK?

I am considering VOLK for one of my future personal projects.

PR #614 (2023-01-15): license: Fix SPDX identifiers
Some files still carried GPL identifiers. These were identified and updated. Thus, all files should carry the LGPL identifier now.

Files with incorrect identifiers were found with:
`grep -irnI --exclude-dir={build,.git,cpu_features} GPL . | grep -v LGPL`

Fixes #613
commit 43a5ecb7 (2023-01-15): license: Fix SPDX identifiers

Some files still carried GPL identifiers. These were identified and
updated. Thus, all files should carry the LGPL identifier now.

Files with incorrect identifiers were found with:
`grep -irnI --exclude-dir={build,.git,cpu_features} GPL . | grep -v LGPL`

Fixes #613