gnuradio/volk #624: Failure in cmake process

yourarchipelago, 2023-05-12:
Ubuntu 20.04
cmake 3.16.3
volk3.0.0

Failure in cmake process with the error below:

Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
-- c flags:  -fcx-limited-range -Wall -Werror=incompatible-pointer-types -Werror=pointer-sign;
-- asm flags: 
-- c flags:  -fcx-limited-range -Wall -Werror=incompatible-pointer-types -Werror=pointer-sign;
-- asm flags: 
-- Loading version 3.0.0 into constants...
  File "<string>", line 2
SyntaxError: bytes can only contain ASCII literal characters.
  File "<string>", line 2
SyntaxError: bytes can only contain ASCII literal characters.
CMake Error at cmake/Modules/VolkPython.cmake:147 (add_custom_target):
  add_custom_target cannot create target "ALL" because another target with
  the same name already exists.  The existing target is a custom target
  created in source directory
  "/home/icebag/桌面/volk3.0_git/volk/python/volk_modtool".  See
  documentation for policy CMP0002 for more details.
Call Stack (most recent call first):
  cmake/Modules/VolkPython.cmake:251 (VOLK_UNIQUE_TARGET)
  python/volk_modtool/CMakeLists.txt:23 (VOLK_PYTHON_INSTALL)

jdemel, 2023-05-31:
I had another look at your issue. It seems like something got messed up. Maybe you got another `git clone` inside your original `git clone ... volk`?

yourarchipelago, 2023-06-01:
I solved this problem.

yourarchipelago, 2023-06-01:
I used Ubuntu in Chinese Language, when I put the source code on Desktop, the absolute path contained some Chinese characters, which could be interpreted by cmake. 

I put the source at Home directory, everything was alright.