gnuradio/volk #626: volk installs different files depending on system state

ZeroChaos-, 2023-08-03:
I'm not really sure what's going on here, but the following files are installed *sometimes* by volk:
```
usr/bin/list_cpu_features
usr/include/cpu_features/cpu_features_cache_info.h
usr/include/cpu_features/cpu_features_macros.h
usr/include/cpu_features/cpuinfo_x86.h
usr/lib64/cmake/CpuFeatures/CpuFeaturesConfig.cmake
usr/lib64/cmake/CpuFeatures/CpuFeaturesConfigVersion.cmake
usr/lib64/cmake/CpuFeatures/CpuFeaturesTargets-none.cmake
usr/lib64/cmake/CpuFeatures/CpuFeaturesTargets.cmake
usr/lib64/libcpu_features.a
```

I am the gentoo package maintainer, and running this through the gentoo package manager.  In short that means that whatever is installed gets merged to the filesystem, and any files owned by the "old" version get wiped off the filesystem.  As I rebuild volk 3.0.0, it will install these files every other time.  If the files are on the filesystem it won't install them, if they aren't present then it will install them.  It doesn't seem that they are required for users, can you please modify cmake to not install them at all?

ZeroChaos-, 2023-08-03:
tests seem to fail if these aren't present, so maybe keep the "build when needed" as is and simply don't install these files?

ZeroChaos-, 2023-08-03:
If these are only required when `-DENABLE_TESTING=ON` then they should also be fixed to not build when `-DENABLE_TESTING=OFF`, right now it seems to build unconditionally (if not already installed on the filesystem)

jdemel, 2023-08-11:
This is a tricky one. We don't want to install any of these files because VOLK does not expose this interface. However, cpu_features doesn't offer a "do not install files" option. Thus, they get installed at the moment whenever someone runs make install on VOLK.

jdemel, 2023-10-03:
@ZeroChaos- does #631 fix this issue? I would assume that it does.

closed 2023-10-03 by ZeroChaos-