gnuradio/volk #628: Beginner cross compile question

w1ebr, 2023-09-11:
I want to create "aarch64" kernels for iPadOS on an Intel MacBook Pro. Specifying the toolchain as "aarch64-lionux-gnu" doesn't work (not surprising). What should I be doing?
Thanks!
Gene

w1ebr, 2023-09-11:
Thanks for starting this dialogue. 

I am trying to port an SDR radio program called SDR++ to the iPads with the M1 and M2 chips. The radio software uses the VOLK library, so I need to create a neon version of that library.

This is what I get when I run the tool path for "aarch64": 

> cmake -DCMAKE_TOOLCHAIN_FILE=../cmake/Toolchains/aarch64-linux-gnu.cmake -DCMAKE_MAKE_PROGRAM=make -DCMAKE_C_COMPILER=Clang -DCMAKE_CXX_COMPILER=Clang  ..
> -- Build type not specified: defaulting to release.
> -- Build type set to Release.
> -- 
> -- Python checking for python >= 3.4
> -- Python checking for python >= 3.4 - found
> -- 
> -- Python checking for mako >= 0.4.2
> -- Python checking for mako >= 0.4.2 - found
> -- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
> -- QA Testing is enabled.
> --   Modify using: -DENABLE_TESTING=ON/OFF
> -- System profiling is disabled.
> --   Modify using: -DENABLE_PROFILING=ON/OFF
> -- Compiler name: Clang
> -- Use `posix_memalign` for aligned malloc!
> -- x86* CPU detected
> -- Compiler doesn't support NEON, Overruled arch neon
> -- Compiler doesn't support NEON, Overruled arch neonv7
> -- Compiler doesn't support NEON, Overruled arch neonv8
> -- CPU width is 64 bits, Overruled arch 32
> -- machine is not riscv64, Overruled arch riscv64
> -- Available architectures: generic;softfp;hardfp;64;popcount;mmx;fma;sse;sse2;orc;norc;sse3;ssse3;sse4_a;sse4_1;sse4_2;avx;avx2;avx512f;avx512cd
> -- Available machines: generic_orc;sse2_64_mmx_orc;sse3_64_mmx_orc;ssse3_64_mmx_orc;sse4_a_64_mmx_orc;sse4_1_64_mmx_orc;sse4_2_64_mmx_orc;avx_64_mmx_orc;avx2_64_mmx_orc;avx512f_64_mmx_orc;avx512cd_64_mmx_orc
> -- BUILD TYPE = RELEASE
> -- Base cflags = -O3 -DNDEBUG  -Wall -Werror=incompatible-pointer-types -Werror=pointer-sign -target aarch64-apple-nonegnu
> -- BUILD INFO ::: generic_orc ::: Clang ::: -O3 -DNDEBUG  -Wall -Werror=incompatible-pointer-types -Werror=pointer-sign -target aarch64-apple-nonegnu 
> -- BUILD INFO ::: sse2_64_mmx_orc ::: Clang ::: -O3 -DNDEBUG  -Wall -Werror=incompatible-pointer-types -Werror=pointer-sign -target aarch64-apple-nonegnu -m64 -mmmx -msse -msse2
> -- BUILD INFO ::: sse3_64_mmx_orc ::: Clang ::: -O3 -DNDEBUG  -Wall -Werror=incompatible-pointer-types -Werror=pointer-sign -target aarch64-apple-nonegnu -m64 -mmmx -msse -msse2 -msse3
> -- BUILD INFO ::: ssse3_64_mmx_orc ::: Clang ::: -O3 -DNDEBUG  -Wall -Werror=incompatible-pointer-types -Werror=pointer-sign -target aarch64-apple-nonegnu -m64 -mmmx -msse -msse2 -msse3 -mssse3
> -- BUILD INFO ::: sse4_a_64_mmx_orc ::: Clang ::: -O3 -DNDEBUG  -Wall -Werror=incompatible-pointer-types -Werror=pointer-sign -target aarch64-apple-nonegnu -m64 -mmmx -msse -msse2 -msse3 -msse4a -mpopcnt
> -- BUILD INFO ::: sse4_1_64_mmx_orc ::: Clang ::: -O3 -DNDEBUG  -Wall -Werror=incompatible-pointer-types -Werror=pointer-sign -target aarch64-apple-nonegnu -m64 -mmmx -msse -msse2 -msse3 -mssse3 -msse4.1
> -- BUILD INFO ::: sse4_2_64_mmx_orc ::: Clang ::: -O3 -DNDEBUG  -Wall -Werror=incompatible-pointer-types -Werror=pointer-sign -target aarch64-apple-nonegnu -m64 -mmmx -msse -msse2 -msse3 -mssse3 -msse4.1 -msse4.2 -mpopcnt
> -- BUILD INFO ::: avx_64_mmx_orc ::: Clang ::: -O3 -DNDEBUG  -Wall -Werror=incompatible-pointer-types -Werror=pointer-sign -target aarch64-apple-nonegnu -m64 -mmmx -msse -msse2 -msse3 -mssse3 -msse4.1 -msse4.2 -mpopcnt -mavx
> -- BUILD INFO ::: avx2_64_mmx_orc ::: Clang ::: -O3 -DNDEBUG  -Wall -Werror=incompatible-pointer-types -Werror=pointer-sign -target aarch64-apple-nonegnu -m64 -mmmx -msse -msse2 -msse3 -mssse3 -msse4.1 -msse4.2 -mpopcnt -mavx -mfma -mavx2
> -- BUILD INFO ::: avx512f_64_mmx_orc ::: Clang ::: -O3 -DNDEBUG  -Wall -Werror=incompatible-pointer-types -Werror=pointer-sign -target aarch64-apple-nonegnu -m64 -mmmx -msse -msse2 -msse3 -mssse3 -msse4.1 -msse4.2 -mpopcnt -mavx -mfma -mavx2 -mavx512f
> -- BUILD INFO ::: avx512cd_64_mmx_orc ::: Clang ::: -O3 -DNDEBUG  -Wall -Werror=incompatible-pointer-types -Werror=pointer-sign -target aarch64-apple-nonegnu -m64 -mmmx -msse -msse2 -msse3 -mssse3 -msse4.1 -msse4.2 -mpopcnt -mavx -mfma -mavx2 -mavx512f -mavx512cd
> -- Compiler Version: Apple clang version 15.0.0 (clang-1500.0.40.1)
> Target: x86_64-apple-darwin23.0.0
> Thread model: posix
> InstalledDir: /Users/gene/Desktop/Xcode-beta.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
> -- c flags:  -Wall -Werror=incompatible-pointer-types -Werror=pointer-sign -target aarch64-apple-nonegnu;
> -- asm flags: 
> -- c flags:  -Wall -Werror=incompatible-pointer-types -Werror=pointer-sign -target aarch64-apple-nonegnu;
> -- asm flags: 
> -- Loading version 3.0.0 into constants...
> -- Use `posix_memalign` for aligned malloc!
> -- Using install prefix: /usr/local
> -- Configuring done (2.5s)
> CMake Warning (dev):
>   Policy CMP0068 is not set: RPATH settings on macOS do not affect
>   install_name.  Run "cmake --help-policy CMP0068" for policy details.  Use
>   the cmake_policy command to set the policy and suppress this warning.
> 
>   For compatibility with older versions of CMake, the install_name fields for
>   the following targets are still affected by RPATH settings:
> 
>    volk
> 
> This warning is for project developers.  Use -Wno-dev to suppress it.
> 
> -- Generating done (0.0s)
> -- Build files have been written to: /Users/gene/Dev/SDRPlusPlus-master-iPadOS/volk-main/build

I think the next step would be getting the compiler's ability to target NEON understood b ut I have no idea on how to proceed.

jdemel, 2023-09-13:
I know the architecture is called `arm64` on MacOS. I suppose iOS follows the same naming convention.

The [aarch64-linux-gnu toolchain file](https://github.com/gnuradio/volk/blob/main/cmake/Toolchains/aarch64-linux-gnu.cmake) is targeted at Linux on ARM.
I would hope that a similar file that targets iOS would do the trick. However, I don't have the required environment to test any of these cross-compile things.
I found a [SO question on this subject](https://stackoverflow.com/questions/65361672/build-apple-silicon-binary-on-intel-machine). Maybe it'll help you.
It'd be great to receive a PR with a toolchain file for your case too.

w1ebr, 2023-09-14:
Thanks, Johannes, for your help.
Gene

argilo, 2023-11-05:
It appears this question was answered, so I'll close off the issue. Let me know if it needs to be re-opened.

closed 2023-11-05 by argilo