gnuradio/volk #629: Build error

ashley-b, 2023-09-14:
```
CMake Error at lib/CMakeLists.txt:524 (target_include_directories):
  Error evaluating generator expression:

    $<TARGET_PROPERTY:CpuFeatures::cpu_features,INTERFACE_INCLUDE_DIRECTORIES>

  Target "CpuFeatures::cpu_features" not found.
```
This **CpuFeatures::cpu_features** should be **CpuFeatures::cpu_features**

Also this if/else block seem redundant to me, in both branches it is including cpu_features far as I can tell
https://github.com/gnuradio/volk/blob/efa2220e6f0b50724618b006457f97eb748327e7/lib/CMakeLists.txt#L523C1-L531C10

ashley-b, 2023-09-14:
As a back story I have been trying to link volk to an external version of cpu_features.

I have dug into cpu_feature and think I have found an issue with it.
When it is built separately it exports a target name CpuFeatures but when it is built as a sub-project it uses CpuFeature

For reference
https://github.com/google/cpu_features/issues/333

jdemel, 2023-09-14:
Ah! Thanks for reporting this. And also for opening the upstream issue.

We need to monitor this and update the git submodule pointer accordingly. I hope this will be fixed soon.

jdemel, 2023-09-15:
The upstream issue is fixed:
https://github.com/google/cpu_features/commit/982ae38da36a59c6b7629e35464c4be46359e1b9
This was the corresponding PR: https://github.com/google/cpu_features/pull/334

For us, this implies that we update the submodule pointer and fix the 2 lines that were updated in https://github.com/gnuradio/volk/pull/603.

Since none of this is part of a release yet, I assume we don't need to worry about any backwards compatibility.

closed 2023-09-27 by jdemel