gnuradio/volk #663: aarch64: sum_of_poly _32f, power_spectrum_32f fail

marcusmueller, 2023-10-19:
Found during a Fedora scratch build.
[build.log.txt](https://github.com/gnuradio/volk/files/13042347/build.log.txt)

Sum of poly is suspiciously fast in neonvert, and also wrong:
```
 72/139 Test  #72: qa_volk_32f_x3_sum_of_poly_32f .......................***Failed    0.01 sec
RUN_VOLK_TESTS: volk_32f_x3_sum_of_poly_32f(131071,1)
generic completed in 0.216802 ms
a_neon completed in 0.328402 ms
neonvert completed in 0.0616 ms
offset 0 in1: -8937.34 in2: -9080.31 tolerance was: 0.01
volk_32f_x3_sum_of_poly_32f: fail on arch a_neon
Best aligned arch: neonvert
Best unaligned arch: neonvert
```

and power spectrum's tolerance seems ambitious, but I haven't looked at that

```
 95/139 Test  #95: qa_volk_32fc_s32f_power_spectrum_32f .................***Failed    0.01 sec
RUN_VOLK_TESTS: volk_32fc_s32f_power_spectrum_32f(131071,1)
generic completed in 0.472402 ms
neon completed in 0.286401 ms
offset 114951 in1: -48.291 in2: -48.291 tolerance was: 1e-06
volk_32fc_s32f_power_spectrum_32f: fail on arch neon
Best aligned arch: neon
Best unaligned arch: neon

argilo, 2023-10-22:
This one fails in CI too: https://github.com/gnuradio/volk/actions/runs/6603892571/job/17937473189

```
  97: RUN_VOLK_TESTS: volk_32fc_s32f_power_spectrum_32f(131071,1)
  97: generic completed in 14.0574 ms
  97: neon completed in 11.6342 ms
  97: offset 108359 in1: -48.2963 in2: -48.2963 tolerance was: 1e-06
  97: volk_32fc_s32f_power_spectrum_32f: fail on arch neon
```

argilo, 2023-10-23:
So probably just a small amount of error from the approximate log calculation. I'm not sure how much error is acceptable here.

closed 2023-12-01 by jdemel