gnuradio/volk #667: Is arm64 assembly code used in Volk?

w1ebr, 2023-10-21:
What I have seen so far is that the NEON assembly code is written in armv7 syntax, not arm64 syntax. I am trying to assemble the NEON assembly routines on a Mac in Xcode and the language syntax has changed for ARM64 (vs arm 32 bit platforms), for example "PUSH {r4,r5,r6,r7} is not valid syntax for an arm64 platform. Any suggestions for how to build Volk for iOS for arm64 ?

jdemel, 2023-10-22:
In most cases, we use intrinsics, e.g.
https://github.com/gnuradio/volk/blob/39f0f8849cb0ecad273f3f3adfe009e75e6c1a5b/kernels/volk/volk_32f_s32f_multiply_32f.h#L232-L258

I'd recommend to stick to intrinsics and let the compiler do it's thing. Usually, assembly code is added when a target architecture is not supported otherwise. But NEON intrinsics are well supported nowadays. The NEON assembly functions are basically a remainder of this early approach.

w1ebr, 2023-10-22:
That’s great news. At the moment I am using Linux cross compile to generate the right files and then Xcode to do the compiles and linking. It’s not efficient but it’s a starting point.Sent from my iPad

https://github.com/gnuradio/volk/blob/39f0f8849cb0ecad273f3f3adfe009e75e6c1a5b/kernels/volk/volk_32f_s32f_multiply_32f.h#L232-L258

argilo, 2023-11-05:
I'll close this off, since the question was answered. Let me know if it should be re-opened.

closed 2023-11-05 by argilo