gnuradio/volk #669: qa_volk_32f_x2_dot_prod_16i is flaky

argilo, 2023-10-22:
Example of a failed CI run: https://github.com/gnuradio/volk/actions/runs/6601430584/job/17932450061

Failures can be reproduced locally with:

```
while ctest -R qa_volk_32f_x2_dot_prod_16i --output-on-failure; do :; done
```

Log:

```
Test project /home/argilo/git/volk/build
    Start 64: qa_volk_32f_x2_dot_prod_16i
1/1 Test #64: qa_volk_32f_x2_dot_prod_16i ......***Failed    0.07 sec
RUN_VOLK_TESTS: volk_32f_x2_dot_prod_16i(131071,1)
generic completed in 0.653972 ms
u_sse completed in 0.267041 ms
u_avx2_fma completed in 0.14574 ms
u_avx completed in 0.142483 ms
offset 0 in1: 168 in2: 167 tolerance was: 0
volk_32f_x2_dot_prod_16i: fail on arch u_sse
offset 0 in1: 168 in2: 167 tolerance was: 0
volk_32f_x2_dot_prod_16i: fail on arch u_avx2_fma
offset 0 in1: 168 in2: 167 tolerance was: 0
volk_32f_x2_dot_prod_16i: fail on arch u_avx
Best aligned arch: u_avx
Best unaligned arch: generic

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   0.07 sec

The following tests FAILED:
	 64 - qa_volk_32f_x2_dot_prod_16i (Failed)
```

It can be seen that the results are off by one. I would guess that the floating point result was very close to 167.5, resulting in a rounding difference, similar to #646.

marcusmueller, 2023-10-22:
uff.

jdemel, 2023-10-22:
I remember @noc0lour fought hard to fix this or make the CI tests more reliable at least. Or does my memory deceive me here?

Anyways, yes, these kernels might benefit from +-1 tolerance. I wouldn't worry about this kind of deviation in a communication system. In other areas this might be an issue though.

argilo, 2023-10-22:
I added a debug statement and observed that failure occurs when the result is very close to an integer value (e.g. -112.000359). That makes sense, because the kernels just cast from `float` to `int16_t`.

The cast seems like an undesirable thing to do in a signal processing environment though, since it truncates toward zero! I think the result should be rounded.

argilo, 2023-10-22:
`volk_32fc_s32f_deinterleave_real_16i` appears to have the same truncate-toward-zero defect. Other kernels use `rintf`.

jdemel, 2023-11-04:
Thanks for investigating this.

I agree we should round, not truncate towards zero.

closed 2023-12-01 by jdemel