gnuradio/volk #670: Consider to compile generic kernels into arch specific versions

jdemel, 2023-10-22:
Compilers are very smart nowadays. In a lot of cases, the compiler will produce more performant code than we can hand-optimize.

VOLK provides different important mechanisms.
1. hand-optimized kernels
2. an architecture agnostic interface
3. infrastructure to compile kernels for specific architectures cf. `-mavx2`
4. infrastructure to detect and load the best performing kernel on specific hardware.

We can use this interface and the infrastructure, keep all the kernels, and use a good benchmark.
Our benchmark would be "Are we better than the compiler optimizing for specific hardware"?

If we don't have an implementation for a specific architecture available, we can still leverage whatever the compiler can do.

marcusmueller, 2023-10-22:
would this be as simple as simply making our build infrastructure copy every `_generic` kernel to a new `_generic_${MACHINENAME}` kernel, and compile it for that target machine?

marcusmueller, 2023-10-22:
hm, I'm trying this, and interestingly, generic kernels and the same C code copied into …_avx2 kernels have at least in my sample (magintude_squared), very very similar runtimes.

I need to look at the disassembly, but chances are, GCC13 and clang16 are both not bad at optimizing code that leads to efficient microcode execution on a range of archs?

Need to check different intel gens.

marcusmueller, 2023-10-22:
scratch that. @jdemel  I think this is already happening. If you look through `nm -t libvolk.so`, you'll see the same `xyz_generic` symbol occur multiple times. Some are plain baseline code, others contain AVX512 instructions.

marcusmueller, 2023-10-22:
yeah, build/lib/CMakeFiles/volk_machine_${MACHINE_NAME}_.c.o already contains per-machine optimized generic kernels, so this is already implemented.

jdemel, 2023-10-23:
Yes, it should be as simple as: Let the build system copy the generic kernel into every architecture specific compilation unit. That would be the idea.

marcusmueller, 2023-10-23:
again, this is already happening.

jdemel, 2023-10-23:
> again, this is already happening.

The compiler could probably generate these version under the hood. But we don't do this explicitly.

marcusmueller, 2023-10-23:
Again, we do. We really really do.

jdemel, 2023-11-04:
Let me summarize this discussion.

1. We define a separate machine for each reasonable combination of SIMD extensions.
2. Every kernel that fits into a specific machine is compiled into it. e.g. an AVX2 machine includes all previous SSE versions and a generic version. This machine is compiled with `-msse -mavx2 ...`.
3. The generic kernel is included in every machine.
4. In effect we compile every generic kernel with every reasonable combination of SIMD extensions.
5. There's no need to generate `_generic_avx2` etc. kernels because we already do.

@marcusmueller thanks for digging this up.

Closing now as complete.

closed 2023-11-04 by jdemel