gnuradio/volk #673: Truncate-toward-zero distortion in some kernels

argilo, 2023-10-22:
As noted in https://github.com/gnuradio/volk/issues/669#issuecomment-1774158247 and https://github.com/gnuradio/volk/issues/669#issuecomment-1774160329, the following kernels cast directly from `float` to an integer type:

* `volk_32f_x2_dot_prod_16i`
* `volk_32fc_s32f_deinterleave_real_16i`

Truncate-toward-zero introduces distortion, as noted here: https://www.cs.cmu.edu/~rbd/papers/cmj-float-to-int.html

Most kernels avoid this problem by applying `rintf` to round the floating-point number before conversion.

I have not checked whether any other kernels have this defect.

jdemel, 2023-11-04:
The fix would be to introduce `rintf` in these two kernels as well.
Does this issue supersede #669 ?

argilo, 2023-11-04:
> Does this issue supersede #669 ?

I don't think so. #669 is about the test tolerance, which will still need to be increased even if we switch to `rintf` for rounding.

n-west, 2023-11-14:
Since you're working on round, you might be interested in the implications of rintf (especially compared to std::round) which *might* be unexpected but standards-compliant https://x.com/nathanewest/status/1715095455376560344?s=20

PR #709 (2023-12-01): Fix truncate-toward-zero distortion
Fixes #673.

Most kernels with integer outputs use `rintf` to round the output. But `volk_32f_x2_dot_prod_16i` and `volk_32fc_s32f_deinterleave_real_16i` cast directly to `int16_t`, which introduces truncate-toward-zero distortion. Here I've switched the affected kernels to use `rintf` instead.
closed 2023-12-01 by jdemel