gnuradio/volk #674: Undefined Behaviour in volk_8u_x4_conv_k7_r2_8u

argilo, 2023-10-22:
UBSAN reports undefined behaviour in this code:

https://github.com/gnuradio/volk/blob/42f57cd67506e7fb6a7795af3948b803de0085f4/kernels/volk/volk_8u_x4_conv_k7_r2_8u.h#L112-L113

Here `w` is an `unsigned int[2]`, and the `s * (sizeof(decision_t) / sizeof(unsigned int))` calculation intentionally reaches outside the bounds of the array to access the `s`th element of the `decision_t` array `d`.

It looks like this hack became necessary because `sizeof(decision_t)` is 8, and yet it is marked as being 16-byte aligned; because of this, accessing `d[1]` fails due to misalignment. I suspect that the 16-byte alignment of `decision_t` is incorrect.

jdemel, 2023-11-04:
Thanks for finding this issue. Bugs in the FEC decoder code are a nightmare because or test infrastructure is unsuitable for this kernel.

Not directly related to the bug at hand, but:
I had the idea to introduce [gtest](https://github.com/google/googletest) to VOLK. Writing more extensive tests for this kernel with specific input might be worthwhile.

argilo, 2023-12-14:
It looks like it will not be possible to avoid Undefined Behaviour in this kernel without changing the API. Decisions are returned as an array of `unsigned char` and yet they're actually packed as `unsigned int`. VOLK and GNU Radio both use a `decision_t` union to switch between the two in an unsafe way.

The decisions are packed bits. I think storing them in `unsigned char` would make more sense. Even though the function signature would remain the same, this would be an API change because the bytes would be in order, while currently each group of four bytes is reversed (on little-endian platforms, anyway). So we'd have to deprecate this kernel and create a new one.