gnuradio/volk #681: 64-bit integer testing is broken

argilo, 2023-10-23:
The `icompare` function casts all inputs to `int` before comparing & printing them. This will prevent errors in upper bits from being caught, and makes it difficult to diagnose failures.

https://github.com/gnuradio/volk/blob/e853e9bb5693c7800840b736e77db52d713d553e/lib/qa_utils.cc#L463-L480

jdemel, 2023-11-04:
We could use `long long int` [cf. Int types](https://en.cppreference.com/w/cpp/language/types).

However, the fundamental int types just feel broken.
I assume we should `#include <cstdint>` and use [fixed width integer types](https://en.cppreference.com/w/cpp/types/integer).

In this specific case `int64_t` would be correct. Though, this type is marked optional.

jdemel, 2023-11-04:
If we change this line:
https://github.com/gnuradio/volk/blob/d5b317c17efd23f6c6cd9a52584123e1a7545452/lib/qa_utils.cc#L493
to
```cpp
if (((unsigned int)abs(int64_t(((t*)(in1))[i]) - int64_t(((t*)(in2))[i]))) > tol) { 
```
we should be good. Technically, a `#include <cstdint>` line is required as well.

PR #722 (2023-12-12): Fix 64-bit integer testing
Fixes #681.

As suggested by @jdemel, I've switched to wider integer types in the `icompare` function.

After this change, errors in the upper 32 bits of 64-bit integers can be detected. For instance, I tried intentionally breaking the volk_64u_byteswap kernel like so:

```diff
diff --git a/kernels/volk/volk_64u_byteswap.h b/kernels/volk/volk_64u_byteswap.h
index 2fbf3cc..befde6d 100644
--- a/kernels/volk/volk_64u_byteswap.h
+++ b/kernels/volk/volk_64u_byteswap.h
@@ -128,7 +128,7 @@ static inline void volk_64u_byteswap_generic(uint64_t* intsToSwap,
                   [... cut]
closed 2023-12-12 by jdemel