gnuradio/volk #686: Undefined Behaviour in volk_32f_invsqrt_32f

argilo, 2023-10-24:
UBSAN shows the following Undefined Behaviour in volk_32f_invsqrt_32f:

`/home/argilo/git/volk/kernels/volk/volk_32f_invsqrt_32f.h:71:22: runtime error: signed integer overflow: 1597463007 - -569061536 cannot be represented in type 'int'`

This is the problematic line:

https://github.com/gnuradio/volk/blob/e853e9bb5693c7800840b736e77db52d713d553e/kernels/volk/volk_32f_invsqrt_32f.h#L71

jdemel, 2023-11-04:
This is: [Wikipedia fast inverse square root](https://en.wikipedia.org/wiki/Fast_inverse_square_root)
It is sometimes called John Carmack reverse because of its use in Q3A.

> Fast inverse square root, sometimes referred to as Fast InvSqrt() or by the [hexadecimal](https://en.wikipedia.org/wiki/Hexadecimal) constant 0x5F3759DF

It is an approximation. It works. It is a hack.

Since this is actually a floating point operation, I suggest to ignore this specific error.

argilo, 2023-11-04:
I suspect the signed integer overflow occurs because the input is negative. So this could just be a case where the test input (uniformly distributed floats in the range -1 .. +1) doesn't make sense.

jdemel, 2023-11-04:
As long as the output signature is `32f`, roots of negative values do not exist.

[The C reference for sqrt](https://en.cppreference.com/w/c/numeric/math/sqrt) says:
1. `+-0` should return the input value
2. Values smaller `+-0` are NaN. Since `NaN == NaN` is `false`, such input should make the test fail.

michael-roe, 2024-11-14:
According to
https://en.wikipedia.org/wiki/Fast_inverse_square_root

.. the way to avoid undefined behaviour is to use std::bit_cast

Modern compilers are much more likely thyan older ones to generate crazy code for undefined behavior; unclear if this is actually a problem here.

jdemel, 2024-11-14:
> According to https://en.wikipedia.org/wiki/Fast_inverse_square_root
> 
> .. the way to avoid undefined behaviour is to use std::bit_cast
> 
> Modern compilers are much more likely than older ones to generate crazy code for undefined behavior; unclear if this is actually a problem here.

Unfortunately, the kernels as well as the library itself are written in C, unlike all the test code. Thus, we can't use `std::bit_cast`. [There seem to be hacks according to SO](https://stackoverflow.com/questions/70464195/implementing-stdbit-cast-equivalent-in-c). If available, we probably want to use CPU FPUs for the task anyways, or their vector equivalent.

This function is probably a good candidate to implement a test with the new googletest based CI, where we implement a naive invsqrt and compare against this result. As long as the result is approximately equal, we know the compiler didn't start to mess with our hack.

Kiptoo-Deus, 2026-10-07:
This looks fixed by #811 (merged in January): the generic `Q_rsqrt` path now does the magic-number step on `uint32_t` (`0x5f3759df - (conv.u >> 1)`), so there is no signed overflow any more. Maybe this can be closed?