gnuradio/volk #717: Add i386 to CI

argilo, 2023-12-09:
Running VOLK's test suite on i386 revealed a broken protokernel and many broken tests:

* #712
* #713
* #714
* #715
* #716

Given this success, it would seem useful to add i386 to CI to catch more future bugs.

jdemel, 2023-12-10:
Looking at the fixes, I agree.

jdemel, 2023-12-10:
I'm looking for a way to run i386. A native GH Action would be preferable because they are much faster. I do have some hope that [MacOS13 arm64 images](https://github.com/actions/runner-images/blob/main/images/macos/macos-13-arm64-Readme.md) do that for us on MacOS.

For non-x86, we use [run-on-arch](https://github.com/uraimo/run-on-arch-action#supported-platforms). This action doesn't provide an i386 image.

argilo, 2023-12-10:
GNU Radio uses a 32-bit container: https://github.com/gnuradio/gnuradio/blob/bdaf5df2bd26325864d66bfb04caf8c8617296d1/.github/workflows/make-test.yml#L100-L102

argilo, 2023-12-10:
Another way of catching floating-point bugs might be to use a tool like https://github.com/verificarlo/verificarlo.

But an i386 build still would have been needed to catch #714.