gnuradio/volk #746: cannot built 3.1.1 from release tarball

ZeroChaos-, 2024-01-30:
Using the release tarball https://github.com/gnuradio/volk/releases/download/v3.1.1/volk-3.1.1.tar.xz I am unable to build due to missing due to cpu_features being missing.

```
-- cpu_features package not found. Requiring cpu_features submodule ...
CMake Error at CMakeLists.txt:127 (message):
  cpu_features/CMakeLists.txt not found.  Did you forget to git clone
  recursively?

  Fix with: git submodule update --init
```

It was suggested that part of cpu_features was moved to a sub module and may have cause this issue:
https://github.com/gnuradio/volk/pull/737

The tarballs that github automatically generates don't include submodules.  I know it's possible to manually generate tarballs to handle things like this, but I don't know how so cannot provide advice on that at this time.

ZeroChaos-, 2024-01-30:
There may be many more ways to solve this issue, but I have two that I can suggest:

1.) Document the dep and tell the user to handle it manually.  I don't prefer this solution, but it exists and needs to be considered.
2.) Use github actions to make a tarball and upload it to the release.  I have found several more and less complex examples which may be helpful

https://github.com/pytorch/pytorch/blob/main/.github/workflows/create_release.yml
https://github.com/Return-To-The-Roots/s25client/blob/master/.github/workflows/release.yml
https://github.com/pytorch/pytorch/pull/63022/files

argilo, 2024-01-30:
My understanding is that VOLK expects package maintainers to build against their packaged version of cpu_features, while end users building from source can obtain it through the git submodule.

argilo, 2024-01-30:
Hmm, I see that the 3.1.0 tarball had cpu_features but the 3.1.1 tarball does not.

argilo, 2024-01-30:
#737 looks to be unrelated, as that submodule was included in the tarball.

argilo, 2024-01-30:
As discussed on Matrix, it looks like the cause is that the release script is broken and failed to include the cpu_features submodule in the tarball.

Issue #748 was opened for the release script bug(s).

jdemel, 2024-02-24:
I re-created the release tarballs and uploaded them to the website as well as the github release page. I hope they are correct this time. Thanks for the updated script.

jdemel, 2024-02-29:
Closing this issue now. If the issue persists, feel free to re-open it.

closed 2024-02-29 by jdemel