gnuradio/volk #762: VOLK with Python

bigalnz, 2024-06-10:
I am looking at the possibility of using VOLK with my Python wildlife tracker project - and hope one of the contributing team might be able to help a little. 

The project involved counting CW beeps on frequencies starting at 160.120Mhz and going in 10khz channel spacings up to 161.110Mhz. 

My hardware is a  AirspyHF+ with python code runs on a RPi5 can only process about 6 channels at once due to my scipy.firwin and about 501 taps consuming a lot of processing power. 

I wonder if VOLK could be used with ctypes to be more efficient - despite the overhead of ctypes.

```
t = np.arange(len(x))/768000
h = signal.firwin(501, 0.02, pass_zero=True) <--- lots of taps
p = np.exp(2j*np.pi*t*161000)
y = y * p
y = signal.convolve(y, h, 'same') <--- ouch!
```

jdemel, 2024-06-14:
@bigalnz it is definitely possible to add a Python wrapper around VOLK. I suggest to use the same library that GR uses: [pybind11](https://pybind11.readthedocs.io/en/stable/).

I assume you refer to center frequencies from 160.12MHz to 161.11MHz, i.e. 1MHz of bandwidth with 100 channels.

Looking at your example code, I assume your implementation works as follows:
1. record a certain amount of samples for bulk processing
2. shift the current channel of interest to 0Hz.
3. Filter the current channel of interest.

First off, I assume this is only example code and all the taps generation etc. is done once during initialization and re-used. Thus, the only line of code that you need to execute repeatedly is:
```python
y = signal.convolve(y, h, 'same') <--- ouch!
```

You probably want to go the polyphase filterbank channelizer route here: https://wiki.gnuradio.org/index.php/Polyphase_Channelizer

Further, you probably want to apply some downsampling. This will reduce your computational burden significantly. Since you're using `scipy.signal`, smth like https://docs.scipy.org/doc/scipy/reference/generated/scipy.signal.resample_poly.html might come in handy.

fred harris is well known for his multirate work. Googling yielded https://s3.amazonaws.com/embeddedrelated/user/124841/fbmc_book_ch_6_text_5_61615.pdf
The theory on multirate systems is a bit broader than just the polyphase filter. You might stumble across name spelling oddities in the GR codebase, like `fred hARRIS`. This is intentional.
Another reference: https://www.dsprelated.com/thread/7758/understanding-the-concept-of-polyphase-filters

To summarize my comments:
- yes, it is possible to use VOLK in Python.
- in your 100 channel use case there are probably better ways to channelize.

jdemel, 2024-09-13:
Closing this issue. I suppose all the questions are resolved.

closed 2024-09-13 by jdemel