gnuradio/volk #773: qa_volk_8u_conv_k7_r2puppet_8u test failure: "free(): invalid pointer" for small inputs

camel-cdr, 2024-10-10:
Hi, I modified the `lib/testqa.cc` file as follows, to test my RVV port, but ran into a bug of an existing kernel, that also occurs on x86.

```diff
--- a/lib/testqa.cc
+++ b/lib/testqa.cc
@@ -27,2 +27,3 @@ int main(int argc, char* argv[])
     bool qa_ret_val = 0;
+    for (int def_vlen = 22; def_vlen < 25; ++def_vlen){

@@ -31,3 +32,2 @@ int main(int argc, char* argv[])
     int def_iter = 1;
-    int def_vlen = 131071;
     bool def_benchmark_mode = true;
@@ -51,4 +51,2 @@ int main(int argc, char* argv[])
                     return 1;
-                } else {
-                    return 0;
                 }
@@ -56,6 +54,2 @@ int main(int argc, char* argv[])
         }
-        std::cerr << "Did not run a test for kernel: " << std::string(argv[1]) << " !"
-                  << std::endl;
-        return 0;
-
     } else {
@@ -100,3 +94,3 @@ int main(int argc, char* argv[])
     }
-
+    }
     return qa_ret_val;
```

Running the test gave the following output:

```
RUN_VOLK_TESTS: volk_8u_conv_k7_r2puppet_8u(22,1)
spiral completed in 0.011711 ms
avx2 completed in 0.00194 ms
generic completed in 0.004461 ms
Best aligned arch: avx2
Best unaligned arch: avx2
RUN_VOLK_TESTS: volk_8u_conv_k7_r2puppet_8u(23,1)
spiral completed in 0.00034 ms
avx2 completed in 0.00035 ms
generic completed in 0.00241 ms
Best aligned arch: spiral
Best unaligned arch: spiral
RUN_VOLK_TESTS: volk_8u_conv_k7_r2puppet_8u(24,1)
spiral completed in 0.00031 ms
avx2 completed in 0.00032 ms
generic completed in 0.00238 ms
Best aligned arch: spiral
Best unaligned arch: spiral
free(): invalid pointer
Aborted
```

I'm not sure whats going on here, and how best to debug this, so I'm reporting this here.

argilo, 2024-10-10:
The issue is not with the kernel itself, but rather with the test wrapper (`volk_8u_conv_k7_r2puppet_8u`). It only performs memory allocations once (probably to avoid measuring the allocations during profiling), so it's not possible to pass in different vector lengths:

https://github.com/gnuradio/volk/blob/bd4694d0caa1ffd2f80b4c3780246a60c23a9977/kernels/volk/volk_8u_conv_k7_r2puppet_8u.h#L121-L152

Instead of modifying `testqa.cc`, I would suggest doing something like this:

```
for i in {22..24}; do apps/volk_profile -n -v $i -i 1; done
```

camel-cdr, 2024-10-11:
Ah, thanks.
Does volk_profile also validate?

closed 2024-10-11 by camel-cdr

argilo, 2024-10-11:
Yes, it verifies that the kernels produce correct output.