gnuradio/volk #780: Build of v2.5.0 broken on macOS Sequoia / clang 16.0.0

DanielO, 2024-11-28:
Hi,
I tried building the Volk [[https://github.com/macports/macports-ports/blob/master/science/volk/Portfile][Macport]] but it bombs out compiling `work/build/lib/volk_machine_avx512cd_64_mmx_orc.c` with:
```
error: invalid CFI advance_loc expression
error: invalid CFI advance_loc expression
```

It looks related to [this](https://reviews.llvm.org/D153167) commit, however finding exactly which code is triggering it is difficult since there is no location information associated with the error and the file in question pulls in a very large number of includes.

I've [reported](https://trac.macports.org/ticket/71416) it as a macports bug and someone suggested I report it here also.

I am not really sure where to start, any clues welcome :)

jdemel, 2025-01-22:
@DanielO Since there aren't any API changes apart from the license change, it is probably best to update to the latest VOLK version. Since this version is compiled and tested in our CI for MacOS (Intel and ARM), I'd assume the issue is fixed now.

DanielO, 2025-01-22:
> [@DanielO](https://github.com/DanielO) Since there aren't any API changes apart from the license change, it is probably best to update to the latest VOLK version. Since this version is compiled and tested in our CI for MacOS (Intel and ARM), I'd assume the issue is fixed now.

Yes, good idea - I tried that and it appears to work OK so I've created a PR for MacPorts.

closed 2025-01-22 by DanielO