gnuradio/volk #793: configuration file format: bad (broadly speaking)

marcusmueller, 2025-07-18:
[RFC 2119](https://github.com/user-attachments/assets/9c540091-b2a5-4e9d-ad0b-435de83fdc0c) applies.

- [ ] file format SHOULD document what version of VOLK and compiler information for each benchmark entry
- [ ] file format MUST include version number
- [ ] file format SHOULD allow for multiple, separate configurations based on matching either machine identifier (`/etc/machine-id`) or regex against a CPU identifier

jdemel, 2025-07-20:
A couple of entries:
- MUST include a `volk_config_version` field (at the top-level)
- MUST include a `configurations` entry that is a list of configurations
  - The `configurations` list SHOULD contain at least one `configuration`
- A configuration MUST include the following fields
  - `volk_version` that was used to create the `configuration`
  - `volk_machine` that was used to create the `configuration`
- A `configuration` SHOULD include the following fields
  - exact CPU identifier that was used to create the `configuration`
  - an `architecture` entry [`x86`, `aarch64`, `armv7`, `riscv`, `ppc64le`, ...]
  - A `compiler` entry with name and version, e.g., `gcc 13.1`
  - A list `kernels` that contains the following entries
    - `name`
    - `aligned`
    - `unaligned`

These are the entries that would improve the VOLK configuration. Also, it would make this file exchangeable and usable on multiple machines.

jdemel, 2026-01-15:
Happy new year! Sure.